z88dk/z88dk #294: sccz80: Out of memory error avoidable by adding code #bug #classclib

Fabrizio-Caruso, 2017-07-21:
I am trying to get my C game (only ANSI C + some conio.h) 
https://github.com/Fabrizio-Caruso/ASCII-CHASE
to work with Z88DK (sccz80) with the Spectrum target 

I am compiling it with
@REM 
zcc +zx -SO3 -pragma-define:ansicolumns=32 -DDEBUG_SPECTRUM_OUT_OF_MEMORY_HACK -DSPECTRUM_32COL  -vn -DAMALLOC -clib=ansi -lmalloc -lndos -create-app  -o %deliverables%\ZXSpectrum_32col_experimental.prg %mypath%\display_macros.c %mypath%\powerUps.c %mypath%\enemy.c %mypath%\invincible_enemy.c %mypath%\level.c %mypath%\character.c %mypath%\text.c %mypath%\missile.c %mypath%\strategy.c %mypath%\input.c %mypath%\main.c

where  -DDEBUG_SPECTRUM_OUT_OF_MEMORY_HACK  adds a WAIT_PRESS() command at the beginning
		#define WAIT_KEY_PRESS() \
		{ \
			while(!kbhit()) \
			{ \
			}; \
			getch(); \
		};	

If I remove this command (omit the compiler directive) I get an OUT OF MEMORY ERROR.

Otherwise the game (waits for a key press and) starts. It seems to run OK but very slow if compared to the versions produced by CC65 (not tested for more than a few seconds).

Compiling it with sdcc produces an immediate crash of the game independently of the hack used.

aralbrec, 2017-07-21:
The error report is just an accident after crashing.  The program has to be traced to find the errors.  I'm taking a look but may take a few days to find the problems.

Fabrizio-Caruso, 2017-09-14:
Issue solved.

closed 2017-09-14 by Fabrizio-Caruso