z88dk/z88dk #298: printf with CPC target

Fabrizio-Caruso, 2017-07-29:
I wonder whether stdio.h is correctly working for the CPC target or 
whether I am doing something really stupid...

I have not been able to correctly run a simple hello world such as:
  #include <stdio.h>
  int main()
  {
    printf("Hello world!\n");	
    return 0;
  }

My compile line is:
zcc +cpc -subtype=wav -vn -DAMALLOC -lmalloc -lndos -create-app -o %deliverables%\cpchello.prg cpc_hello.c
which seems to work.

Then I use WinApe and I type
|tape
run"
to load the game, which, sort of, loads and prints just a few letters but not the full the text.

What is wrong?

  Fabrizio

zx70, 2017-09-14:
Hello Fabrizio, is this problem still happening ?

Fabrizio-Caruso, 2017-09-15:
Problem no longer reproducible.

closed 2017-09-15 by Fabrizio-Caruso