z88dk/z88dk #299: Cannot run/load my game MSX even with trivial main()

Fabrizio-Caruso, 2017-07-29:
I am trying to port my game
https://github.com/Fabrizio-Caruso/ASCII-CHASE/
to the msx target but I am not able to get it to start/load.

My game is "successfully" compiled with
zcc +msx -vn -D__MSX__ -DAMALLOC -lmalloc -lndos -create-app -o %deliverables%\msx_experimental.prg  <c files>

The compiled executable is about 20k.

If I try to load it with
bload"cas:",r

the result is always that the emulator reboots while loading
no matter which  emulator (bluemsx, fmsx), no matter which msx model and no matter how much RAM used in the emulators.

I can even put a trivial main file (test working if I do not include the rest of the code) such as:
int main()
{
    puts("\x16\x0A\x0C" "Hello, world!");   

    return 0;
}
(or something equivalent that works correctly if compiled alone)
and in this case the emulators get stuck while loading the game.

So maybe the size of the executable (about 20k) is a problem 
(the emulators have been set with enough RAM).
Do I need to use some special options or configurations?

Remark: The game compiles and works fine with other targets (Spectrum and all CC65 targets).

  Fabrizio

zx70, 2017-07-31:
AMALLOC is the lazy way to use malloc, it leaves a fixed percentage of the remaining space for stack and the remaining slice for malloc.  Honestly I don't remember if the msx version is smart enough to tune correctly the pointers in both msxdos and basic mode.
The malloc heap can be manually tuned but it can be tricky for beginners.

Fabrizio-Caruso, 2017-07-31:
@zx70, could you please point me to a reference?
How should I configure the malloc heap to get it to work with the msx target?

zx70, 2017-08-01:
https://www.z88dk.org/wiki/doku.php?id=library:memory_allocation#multiple_named_heap_malloc_library_malloc.h

I am not sure that malloc is your problem, btw you should first look at an msxbasic memory map, and see if you have enough free memory.

Fabrizio-Caruso, 2017-08-02:
My game takes up 20k in total and I am only allocating a few bytes.
The emulator is set up with plenty of memory.

Fabrizio-Caruso, 2017-08-02:
My game takes up 20k in total and I am only allocating a few bytes.
The emulator is set up with plenty of memory.

No issue with other Z88DK targets (Spectrum and even VG 5000)
as well as all CC65 targets.

Fabrizio-Caruso, 2017-08-02:
My game takes up 20k in total and I am only allocating a few bytes.
The emulator is set up with plenty of memory.

No issue with other Z88DK targets (Spectrum, CPC and even VG 5000)
as well as all CC65 targets.

aralbrec, 2017-08-25:
I'm not too familiar with the msx target but the other thing you could do is make a cartridge.  That will give you 16k of code space and 16k of ram for variables.

zx70, 2017-08-31:
#323 is taking care of the same topic, this is redundant

closed 2017-08-31 by zx70