z88dk/z88dk #300: CPC target maybe totally broken

Fabrizio-Caruso, 2017-07-30:
I am trying to compile
https://www.z88dk.org/wiki/doku.php?id=examples:snippets:ansitest

using the very same code

For simplicity I am compile with the -subtype=wav option and I am using the WinApe emulator

My complete compile line is:
zcc +cpc -clib=ansi -subtype=wav -lndos -create-app -o %deliverables\cpc_ansitest ansitest.c

The program is loaded within WinApe with
|tape
run"

The result is a blue screen with nothing shown.

I have also tried to compile my game or just trivial snippets of code with similar results.

Am I doing something wrong?

  Fabrizio

aralbrec, 2017-07-30:
IIRC everything goes through the Amstrad firmware.

A little while ago, the amstrad port was made reliable after it was discovered that the firmware did not allow use of the exx registers.  However, recent changes to classic to accommodate sdcc may have broken this as there is another open issue https://github.com/z88dk/z88dk/issues/163 .

Maybe stef or dom can have a look.

Fabrizio-Caruso, 2017-07-31:
I managed to get the ansitest example to work by using 2ctd tool instead of Z88DK built wav option.

zx70, 2017-07-31:
This is a very interesting hint, thankyou

Fabrizio-Caruso, 2017-07-31:
@zx70 I have been able to run my game, as well.
So, Z88DK's built-in wav conversion corrupts the binaries in weird ways. They may be other problems but if I use 2cdt, I am getting what I expect.

zx70, 2017-07-31:
Which z88dk generated file did you use for the wav conversion?

Fabrizio-Caruso, 2017-07-31:
@zx70, I used the .cpc file with the following command line:
2cdt.exe -n -r cross_chase %deliverables%\cpc.cpc  %deliverables%\cpc.cdt
So I got a cdt file which I used as .wav file.

The .wav file generate through zcc does not work (for me).

zx70, 2017-08-01:
This means that for sure appmake is partially working, because it is able to generate the .cpc file.
Have you tried by changing the audio sampling rate ?  e.g. by adding:
-Cz--audio -Cz--rate=11025

zx70, 2017-08-03:
I just succeeded to load a compiled program with WinAPE, CPC PLUS features enabled.  Both the data rates are hooked.
EDIT:   also mame (cpc6128) works!   I'm now wondering which problem I had in my past attempt :S

Fabrizio-Caruso, 2017-09-15:
Problem no longer reproducible.

closed 2017-09-15 by Fabrizio-Caruso