z88dk/z88dk #305: z80asm: add directive to create floating point values

aralbrec, 2017-08-08:
In addition to defb for bytes, defw for ints, etc it would be useful to be able to declare floating point values.

There can be many different floating point formats.  The two main ones are genmath and math48 formats which are 48-bit floats as these are z88dk's floating point libraries.  However some targets have their own native float libraries with different formats, eg the zx spectrum has a 40-bit fp format.

I'm not sure what the best way is to deal with this but I do know that it's useful to be able to do something like this so I'll leave this feature request here for contemplation.

pauloscustodio, 2017-08-21:
I think this should be implemented with two commands:
1. One to define the float library to be used in the code as a command line option
2. One to declare a list of floats in the current format

An assembler macro should indicate the current format in use, i.e. __FLOAT_MATH48__, ...

This way, the same source code can be used for different float types by changing the command line options.

Suggestion:
```--float=genmath|math48|zx|zx81 ...``` and maybe others to define the float type for the source
```float expr, expr, ...``` to compute the compile-time expressions and store the float value; the expressions are calculated in the assembler as ```double``` entities and the result converted to the target float type

Any comments? What should be the default float type?

Can someone supply pointers to documentation describing the different float formats?

feilipu, 2017-08-21:
Just for reference, there is a good [summary of floating point formats](http://www.andreadrian.de/oldcpu/Z80_number_cruncher.html#mozTocId774334) by Andre Adrian, with some [references](http://www.andreadrian.de/oldcpu/Z80_number_cruncher.html#mozTocId485748).

I'm personally interested in the [Lawrence Livermore Laboratory floating point package](https://github.com/feilipu/LLL-Floating-Point) from 1975, [UCRL-51940, Floating-Point Package for Intel 8008 and 8080 Microprocessors](https://github.com/feilipu/LLL-Floating-Point/blob/master/LLL%20Floating%20Point%20Basic.pdf), because this was the same format used by the Am9511A Arithmetic Processing Unit. It predates IEEE754, by a year or so.

It is on my list to integrate a [driver for the Am9511](https://github.com/feilipu/LLL-Floating-Point/blob/master/am9511a.asm) into z88dk. It is done, but I'm working on some other things currently, so #64 is not a high priority.

So I'd like to see at least the following options
- Math48 - 48 Bits (Z88DK)
- GenMath - 48 Bits (Z88DK)
- Z80 - 40 Bits
- IEEE754 - 32 Bits (SDCC)
- LCRL-51940 - 32 Bits (LLL & Am9511A)

aralbrec, 2017-08-22:
The float type can vary inside the same file.

For example, sdcc only understands a 32-bit float type but that is being converted to math48's 48-bit type for the actual computation and then back again for return to sdcc.  So a single asm file may have to deal in both float types if it communicates with both a c function and the asm float library.

So maybe there's a default float type that can be changed on the command line and then a directive to change float type within a file?

The default float type should probably match what -lm is for the c compilers.  For classic that's genmath for new that's math48.  It doesn't really matter anyway as the default can be set via command line in the cfg files.

> `float expr, expr, ...` to compute the compile-time expressions and store the float value; the expressions are calculated in the assembler as double entities and the result converted to the target float type

Nice to have would be the ability to load into registers with masking.

deff pi = 3.141592653   ;; like defc for float, I don't know if there's better

ld de,+(pi >> 16) & 0xffff
ld hl,pi & 0xffff

pi may be in double form and then only converted to target type at the point of use.

The float type will have to be at least 64-bit.. the trs80 has a 64-bit float type.  These integer operations might be tricky since z80asm constants now are 32-bit and 64-bit means 32/64 issues to circumvent.

Or maybe to avoid that a special syntax to extract bit ranges from float constants.

aralbrec, 2017-08-22:
> Just for reference, there is a good summary of floating point formats by Andre Adrian, with some references.

The "z80 40-bit format" seems to be something he's invented.  I don't know of any existing code or anything that uses it.

I'll come back and document these a little later:

genmath (48 bit):

math48 (48 bit):

sdcc (32 bit):

LCRL-51940 (32 bit, maybe lll is a snazzier name):

feilipu, 2017-08-22:
I think it was discussed [elsewhere](https://github.com/z88dk/z88dk/issues/64#issuecomment-278848136), where it was mentioned that sdcc was almost (but not completely) IEEE754 compliant.

There seems to be ample reference to IEEE754 within their code, such as their [float.h](https://github.com/z88dk/z88dk/files/1240864/float.h.txt).
So I'd guess there is more innocent breakage than wilful non-compliance behind this.

Would it be a sensible path to suggest fixes to their code (preferred), or otherwise patch over their deviations from standard, to ensure that z88dk at least has one IEEE754 compliant math library?

This would at least make sure that math accuracy testing (of all math libraries) could baseline off a known standard, for example.

aralbrec, 2017-08-22:
> There seems to be ample reference to IEEE754 within their code, such as their float.h.
> So I'd guess there is more innocent breakage than wilful non-compliance behind this.

> Would it be a sensible path to suggest fixes to their code (preferred), or otherwise patch over their
> deviations from standard, to ensure that z88dk at least has one IEEE754 compliant math library?

It might be a good idea to be able to define iee754 floats in case there is communication with hardware that implements it or other computers on a network so I'll add that.

sdcc's format varies bit from ieee754 but it is missing all the format exceptions :- there's no nan or subnormal numbers for example.  So the code base is missing a fair bit.  It's also not a very good performer in terms of speed or space so I don't think anyone would use it except through sdcc.

feilipu, 2017-08-22:
Probably I should just move along, but for the record I've just found [Berkely SoftFloat](http://www.jhauser.us/arithmetic/SoftFloat.html), with version 3d released in August 2017.

>Berkeley SoftFloat is a free, high-quality software implementation of binary floating-point that conforms to the IEEE Standard for Floating-Point Arithmetic. SoftFloat is completely faithful to the IEEE Standard, while at the same time being relatively fast. All functions dictated by the original 1985 version of the standard are supported except for conversions to and from decimal. The latest release of SoftFloat implements five floating-point formats: 16-bit half-precision, 32-bit single-precision, 64-bit double-precision, 80-bit double-extended-precision, and 128-bit quadruple-precision. All required rounding modes, exception flags, and special values are supported. Fused multiply-add is also implemented for all formats except 80-bit double-extended-precision.

> SoftFloat is distributed in the form of ISO/ANSI C source code and should be compilable with almost any ISO-compliant C compiler. Using the GNU C Compiler (gcc), the package has been compiled and tested for several platforms.

[SoftFloat-3d.zip](https://github.com/z88dk/z88dk/files/1241005/SoftFloat-3d.zip)

aralbrec, 2017-08-22:
Well there it is.  It might be fun to try to extract and compile the 32-bit float part.  It looks like it's going to be big though.

feilipu, 2019-07-05:
Would be good if this to be developed directive could work with IEEE 754 4-Byte floats too. Would save manually converting `DEFQ`. Although for coefficient tables that is not too difficult, using on-line tools, [for example](https://www.h-schmidt.net/FloatConverter/IEEE754.html).

pauloscustodio, 2021-12-30:
@feilipu , are the floating point representations of genmath (zxx -lm) and math48 (zcc -lmath48) the same?

pauloscustodio, 2021-12-30:
Another issue: when running:
```
zcc +test -otest_t2_FLOAT.bin --math16 --list -m -create-app test_t2_FLOAT.c
```
I was expecting the floats to be encoded in 2 bytes, but they appear in 6. Am I missing something?

The C code:
```
float fdata[] = { 1,-1,255,-255,256,-256,65535,-65535,65536,-65536,5.5,-5.5,55,0.5 };
```
generates the assembly:
```
1     0000              ._fdata
1     0000  00 00 00 00 00 81 
                        	defb	0,0,0,0,0,129
1     0006  00 00 00 00 80 81 
                        	defb	0,0,0,0,128,129
```

suborb, 2021-12-30:
Genmath and math48 have the same format. 

Math16 is accessed via a separate type: _Float16 rather than float/double.

feilipu, 2021-12-30:
Just scrolling back through this discussion.

Lots learned over past 4 years, so the format needs are quite changed.

No mention of MBF versions is one omission that's obvious.

pauloscustodio, 2021-12-31:
Another question on zcc:
The option `-math-z88` should output z88 40-bit floats, but instead I am getting 48-bits with a leading byte of 0xe4. Am I missing something?

test.c:
```
float fdata[] = { 5.5, -5.5 };
```
output of: `zcc +test -math-z88 --list test.c`:
```
1     0000              ._fdata
1     0000  E4 00 00 00 30 82 
                        	defb	228,0,0,0,48,130
1     0006  E4 00 00 00 B0 82 
                        	defb	228,0,0,0,176,130
```

pauloscustodio, 2021-12-31:
Same problem with `-fp-mode=mbf40`, should output 40 bits, I'm getting 48.

pauloscustodio, 2021-12-31:
The FLOAT directive implementation is finished. The two remaining issues are the z88 and mbf40 that zcc outputs with 5 bytes instead of 4; the corresponding tests in z80asm are skipped.

pauloscustodio, 2021-12-31:
Documentation of the new FLOAT and SETFLOAT directives in https://github.com/z88dk/z88dk/wiki/Tool---z80asm---directives

commit bab6a7b0 (2021-12-31): Fix #305 add FLOAT directive
closed 2021-12-31 by pauloscustodio