z88dk/z88dk #307: Question: encoding of ALTD DJNZ

pauloscustodio, 2017-08-09:
From reading of the RabbitInstructions.pdf, page 46, it is not clear how the following should be assembled:
```ALTD DJNZ ASMPC``` - $76, $10, $FE as in the Z80, or $76, $10, $FD to repeat the same instruction?

quote: "This instruction’s mnemonic stands for Decrement and Jump if Not Zero. It decrements the data in B then, if the data in B does not equal 0, it adds the 8 bit signed constant e to PC.
2 is subtracted from the value e so the instruction jumps from the current instruction and not the following
instruction."

suborb, 2017-08-09:
It's a good question. I think other assemblers behave as per the z80. It certainly feels less surprising to do so

aralbrec, 2017-08-10:
I think it behaves like the z80.  The instruction is not repeating as ALTD means B' is used for the loop counter instead of B.

pauloscustodio, 2017-08-10:
Does anyone have access to a RCM2000/3000 assembler? After I finish the Rabbit implementation in z80asm I would like to test it with an independent assembler.

pauloscustodio, 2017-09-11:
Implemented as in Z80, i.e. do not repeat the ALTD. Closed.

closed 2017-09-11 by pauloscustodio