z88dk/z88dk #308: VG 5000 re-maps or inhibits video memory

Fabrizio-Caruso, 2017-08-09:
I am trying to port my game to the VG 5000 target.
It should be easy but conio is either slow or buggy.
So I have decided to write onto video memory. This is easy even though the VG 5000 has a weird way to display characters and their colors.

The simple program below should display a black screen with some 'A' characters but this
ONLY happens after exiting the program as if the code generated by the compiler were
re-mapping the video memory.

Is this a bug or a feature? How can I address video memory?

#include <stdio.h>
#include <stdlib.h>

#define POKE(addr,val) (*(unsigned char*) (addr) = (val))

int main(void)
{
	unsigned char i;
	
	printf("Hello world!\n");
	for(i=0;i<25;++i)
	{
		POKE(16384+80*i,32);
		POKE(16385+80*i,1);
	}
	for(i=0;i<10;++i)
	{
		POKE(16386+2*i,65);
		POKE(16387+2*i,7);
	}
	
	getchar();	
	return 0;
}

suborb, 2017-08-09:
As far as I can tell. The vg-5k driver does a look up using a couple of rom routines and then writes to memory with no paging.

What was your problem with the vg-5k conio driver?

Fabrizio-Caruso, 2017-08-09:
@suborb, conio is not usable for anything else that static text in any target in Z88DK, as far as I have seen.
In my humble opinion conio or some other portable solution should be just macros to the fastest display code for each target. So that a portable solution is made available for all Z88DK targets. 

On the other hand, I do not understand why the VG 5000 targets makes it impossible to display through direct writes onto video memory. The compiled code is doing something that stops video ram at $4000 from being used to display its content.

Fabrizio-Caruso, 2017-08-09:
@suborb, an additional problem of conio for VG 5000 is that the cursor is shown when you try to display anything.

I fear that the video memory problem I am having is not a bug. It is my lack of understanding how the video chip works. It is complicated.
The very same code above in BASIC, would work.

Fabrizio-Caruso, 2017-08-09:
VG 5000 has a very weird graphics chip...
Probably I will need to use a little assembly routine into my C code but I do not know how to do it.

Probably something like:
org $6000   

_debut:
      ld a,0
      ld ($47FD),a  ;poke 47FD c'est pour un curseur fixe
      ld a, 65
     
      ld h,20  ; cordonnée y
      ld l,20    ;coordonnée x
      ld d,65  ; caractere
      ld e,1    ; couleur
      call $0092    ;routine affichage
      jp _debut

Where parameters are passed through H,L,D,E registers.
How can I do this?

Fabrizio-Caruso, 2017-08-09:
A better way to implement character display (without cursor) would be:

	int _draw_ch(unsigned char x, unsigned char y, unsigned char ch, unsigned char col)
	{
		int xy = 0;
		int chCol = 0;
		xy = (y<<8) | x;
		chCol = (ch<<8) | col;
		return _draw_ch_aux(chCol,xy);
	}

	int _draw_ch_aux(int chCol, int xy)
	{
		#asm
		pop bc   ; bc = ret address
		pop hl   ; hl = int b
		pop de  ; de = int a

		push de    ; now restore stack
		push hl
		push bc
		
		call 0x0092	
		#endasm
	}

aralbrec, 2017-08-10:
@zx70 incorporated the VG5000.  It could that a specific video mode has to be enabled for it to behave as you expect.

zx70, 2017-08-10:
There is a recent forth implementation showing some good trick for an advanced use of the console. I don't remember the detail, but it was not an immediate task to borrow it.

zx70, 2017-08-10:
EDIT: I'm on holiday and I'm not fully operative.. pls forgive some evasive answer !

Fabrizio-Caruso, 2017-08-11:
The chip in VG 5000 is complex and the problem comes from my lack of understanding.
Please close this as a non-issue. 
Though, the code to print characters in Z88DK should avoid showing the cursor.

Fabrizio-Caruso, 2017-08-24:
@zx70, 
The cursor is shown if I use conio with no other modification.
I can only  fix the cursor with
			_ef9345:
				defb 0x04,0x20,0x82,0x29,0x00		
				pop bc
				pop hl
				pop de
				push de
				push hl
				push bc
				ld hl,_ef9345
				call 0x00AD		

The assembly code above is wrting $00 into EF9345's register $82. This will unset all bits in this special register and toggle off the cursor display.

zx70, 2017-08-24:
Thanks for the hint, I'll experiment a bit.  For your information,  an alternate  text driver for vt emulation exists.

Fabrizio-Caruso, 2017-08-24:
VG5000 prints on screen are very slow (maybe 100 times slower) than:

	int _draw_ch_aux(int chCol, int xy)
	{
		#asm
		pop bc   ; bc = ret address
		pop hl   ; hl = int b
		pop de  ; de = int a

		push de    ; now restore stack
		push hl
		push bc
		
		call 0x0092		
		#endasm	
	}

	int _draw_ch(unsigned char x, unsigned char y, unsigned char ch, unsigned char col)
	{
		int xy = 0;
		int chCol = 0;
		xy = ((y+8+Y_OFFSET)<<8) | (x+X_OFFSET);
		chCol = (ch<<8) | col;
		return _draw_ch_aux(chCol,xy);
	}

zx70, 2017-08-25:
The ansi text driver is calling also A7h, to keep a copy of the video memory to scroll, but I wouldn't expect such impact on speed.  The ansi engine itself is well optimized (a buffer is even supported, but not used on z88dk), the slowness I mentioned for the Spectrum is on the way the single character is being printed.

zx70, 2017-08-25:
Native driver (non ansi) was never completed, sorry for confusion

Fabrizio-Caruso, 2017-08-25:
I fear scrolling is triggered if the last character of the last column is displayed.
Text scrolling is not what you want in character graphics. Can I print all characters on the screen without risking scrolling the whole text?

Here the problems for the VG5000 are: cursor + slowness. My assembly above fixes both problems but I am not fluent in  assembly and I fear instability. Could you please check whether I am not screwing the stack or registers?

The slowness of Spectrum may come from the fact that it only has a bitmap mode but conio is super slow compared to any native directives (classic or new)

zx70, 2017-08-25:
I could add your code and disable the cursor at startup, did you discover if we have conditions where it gets restored ?

The slowness on Spectrum.   Main reason for it is the dynamic resolution I wanted to have: any character width between 3 to 8 (9, in some case) pixels is allowed at run time (no need to recompile).
This is based on pixel shifting, obviously slow.

zx70, 2017-08-25:
..correcting your function

void no_cursor(void)
{
	#asm
               jr clean_cursor
		_ef9345:
			defb 0x04,0x20,0x82,0x29,0x00		
               clean_cursor:
			ld hl,_ef9345
			ld ix,$47FA
			call 0x00AD		
	#endasm		
}

Fabrizio-Caruso, 2017-08-25:
Is scrolling trigger by conio if the last character of the lowest row is displayed?
This is wrong and makes conio not usable for character graphics (unless you avoid the last row).

On the VG 5000 I think the cursor is restored by something but I do not know what.
I am not familiar with assembly. 
In my code I am calling no_cursor whenever I print a character. This is not necessary but if I call it only once, it will not work. So something is resetting the cursor display. I need to further investigate this.

Fabrizio-Caruso, 2017-08-25:
My personal wish:
I hope Z88DK will provide a portable solution for all or most targets for simple fixed standard size character graphics and keyboard input.
If only fixed-size character graphics (with no scrolling) and simple keyboard input is used, Z88DK
should ideally provide a common API (e.g., conio) that should use the most efficient available code.

Fabrizio-Caruso, 2017-08-25:
@zx70, Why ld ix,$47FA?

zx70, 2017-08-25:
Looks like the rom code needs to have  ix pointing to the first byte of the "screen communication flags", but the z88dk may have used ix.   The cursor disable trick works, I'm adding a custom cls file doing also this.

Fabrizio-Caruso, 2017-08-25:
Thanks a lot Stefano!
Should I also use ld ix,$47FA on the print code below?
int _draw_ch_aux(int chCol, int xy)
{
	#asm
	pop bc   ; bc = ret address
	pop hl   ; hl = int b
	pop de  ; de = int a

	push de    ; now restore stack
	push hl
	push bc
	
	call 0x0092		
	#endasm	
}

int _draw_ch(unsigned char x, unsigned char y, unsigned char ch, unsigned char col)
{
	int xy = 0;
	int chCol = 0;
	xy = ((y+8+Y_OFFSET)<<8) | (x+X_OFFSET);
	chCol = (ch<<8) | col;
	return _draw_ch_aux(chCol,xy);
}

Will you also provide a function to print characters based on call 0x0092?
It is faster than conio by a huge factor.
My assembly skills are inexistent. Could you please also check the print routine?

zx70, 2017-08-25:
you can add the ix initialization before the call, then you're set.  The current  ansi vt is already based on that call but it calls also $A7.  I don't remember why, perhaps to feed a memory buffer for the scroll or to make the video chip picking the char from the mirror memory.

zx70, 2017-08-26:
I applied your suggestion to hide the cursor.   The bad news is that it works only if repeated after every single character output.   What the current driver does now is:
- display the character
- ask the rom to find the display buffer in memory (perhaps this makes also the cursor visible)
- use the found location to update the buffer with the character and attribute codes
- hide the cursor

I'm afraid it is slowing rather than speeding-up.   Direct controller access is possible, but dealing with the text scrolling comes at some cost (development time and possibly code size too).

Fabrizio-Caruso, 2017-08-26:
Hi!
Thanks a lot for the nice support! 
Thanks a lot for introduction the no_cursor code!
That is what I suspected. My code is so fast I need to slow it down.
I am using the no_cursor code + call 0x0092.

CONIO should not produce scrolling.
It does not so on DOS/Windows, nor on CC65 (6502), nor on CMOC (6809).
CONIO is not supposed to behave like stdio.h.

Unfortunately Z88DK still lacks any portable common API to do usable character graphics.
I hope you guys will provide it in the future even if it is not CONIO.

Writing a portable game for Z80 is fun but currently much harder than for other 8 bit platforms:
I have to patch each single target and find out how to print a character on the screen and how to read the keyboard (getk seems to rely on ROM code and does not behave equally across the different targets and as a consequence on some targets, e.g., msx, repeated key-presses have a lag that makes it useless for games).

You can the VG-5000 version of my game at:
https://github.com/Fabrizio-Caruso/ASCII-CHASE/releases/tag/VG-5000

Without your support, I would not have progressed so much! Thanks!

zx70, 2017-08-26:
My intention was to have a similari text behavior (cursor addressing, color code) via terminal emulation.  I'm mightly impressed to discover that other 8 bit CPU deve kits could reach a better text supporto portabilità considering that eventi the CP/M failed to do it.  By the way it it great to seee you were abile to port your game between different kits, it is a relevant goal for all !  The issue related to the vg5k display limits can be closed, because the scroll should ne prevented on the generic part of the vt emulation engine and will change the actual behavior on all the targets...   Il chech it, btw

closed 2017-08-26 by zx70

Fabrizio-Caruso, 2017-08-26:
Z88DK is the largest project of such kind and provides tons of great tools.
It is better in many respects than other projects.
Other projects such as CC65 are smaller in scope and aim at portability: if I remove specific frills,  the very same code works efficiently on all 6502 targets. I get 14 or more versions of my game with no effort.
They provide almost no specific libraries. They do provide portable bitmap routines as well.

They implement just whatever is necessary to do gotoxy, cegtc, kbhit  and cputc and not the full ANSI VT.
You can do the same by having macros point to the most efficient implementation of gotoxy, knit, cgetc, cputc with no scrolling for the standard and most common resolution on each target.

@zx70 , would it be possible to have a chat with you?