z88dk/z88dk #309: make at the top directory fails - testsuite

pauloscustodio, 2017-08-09:
Calling make at the top directory of the master branch fails in testsuite. I'm running Ubuntu in Linux Subsystem for Windows 10.
```
~/git/z88dk$ make
[....]
make -C testsuite
make[1]: Entering directory '/home/paulo/git/z88dk/testsuite'
zcc +test -vn -a Issue_254_KR_Specifiers.c -o Issue_254_KR_Specifiers.opt
Doing something 0x215c190
diff -w tmp.opt results/Issue_254_KR_Specifiers.opt
zcc +test -vn -a 02_addr_ptr.c -o 02_addr_ptr.opt
diff -w tmp.opt results/02_addr_ptr.opt
15,16c15,16
<       push    hl
<       ld      hl,100  ;const
---
>       ld      (hl),#(100 % 256 % 256)
>       ld      l,(hl)
18,20d17
<       pop     de
<       ld      a,l
<       ld      (de),a
Makefile:11: recipe for target '02_addr_ptr.opt' failed
make[1]: *** [02_addr_ptr.opt] Error 1
make[1]: Leaving directory '/home/paulo/git/z88dk/testsuite'
Makefile:118: recipe for target 'testsuite' failed
make: *** [testsuite] Error 2
```

suborb, 2017-08-10:
I think that's old code or at least an old sccz80 running. There's a debug line which I got rid of being printed.

pauloscustodio, 2017-08-10:
Problem found: wrong git core.autocrlf configuration. It must be 'input' on a Unix system, was set as 'auto'

Solved by running:
```git config --global --replace-all core.autocrlf input```

closed 2017-08-10 by pauloscustodio