z88dk/z88dk #312: z80asm: z80 variants from fpga implementations of z80 core (zx next)

aralbrec, 2017-08-10:
Paulo do you think there can be an easy way to define additional opcodes for the z80?  I am thinking in general here as z80 implementations exist for fpgas and we now have a specific case where new opcodes are being added for a machine.

The zx next, which I am working on now, is adding these instructions:
```
   swapnib    ED 23    A bits 7-4 swap with A bits 3-0
   mul        ED 30    multiply HL*DE = HLDE (no flags set)
   add  hl,a  ED 31    Add A to HL (no flags set)
   add  de,a  ED 32    Add A to DE (no flags set)
   add  bc,a  ED 33    Add A to BC (no flags set)
   outinb     ED 90    out (c),(hl), hl++
   ldix       ED A4    As LDI,  but if byte==A does not copy
   ldirx      ED B4    As LDIR, but if byte==A does not copy
   lddx       ED AC    As LDD,  but if byte==A does not copy, and DE is incremented
   lddrx      ED BC    As LDDR,  but if byte==A does not copy
```
That's a partial list as more is coming.

I have also been thinking about adding via m4 but that is not ideal because all asm would have to be in an m4 file and this would require changes in zcc.  I could also do it in pre-processing with copt.

pauloscustodio, 2017-08-10:
Yes. I'm currently working on the full Rabbit implementation, and it would be easy to add a new cpu-type to the input files.

aralbrec, 2017-08-11:
So it's possible to do in mainline z80asm now?

Then the assembler invoke would be "z80asm --cpu=cpuname".  Have you thought of any naming to accommodate these devices?  Maybe "z80asm --cpu=z80-zxn" (z80 variant of zx next).

pauloscustodio, 2017-08-11:
Yes. Give me a couple of days.

I agree with your suggestion for the naming. I assume this z80-zxn cup does not have any of the z180 and rabbit opcodes and it had all the undocumented z80 opcodes. Correct?

aralbrec, 2017-08-11:
Yes everything z80 is in there, it's just the additional instructions and ones yet to come.

pauloscustodio, 2017-08-11:
Just a note: ```MUL``` will be confusing:
- in zxn it is coded as ED 30 and means HLDE := HL*DE
- in r2k it is coded as F7 and means HLBC := BC*DE

feilipu, 2017-08-12:
And, we're clear to use the `MLT` instruction with the z180 too?

The `MLT` performs unsigned multiplication on two 8-bit numbers yielding a 16-bit result. `MLT` may specify `BC`, `DE`, `HL`, or `SP` registers. The 8-bit operands are loaded into each half of the 16-bit register and the 16-bit result is returned in that register.

aralbrec, 2017-08-12:
Yes z80asm has separate tables for z80, z80-zxn, z180, r2k and r3k.

aralbrec, 2017-08-16:
Another update... sorry this may be a frequent activity now but people find it helpful to get it in.

```
New Z80 opcodes on the NEXT (more to come)
======================================================================================
   swapnib           ED 23           A bits 7-4 swap with A bits 3-0
   mul               ED 30           multiply HL*DE = HLDE (no flags set)
   add  hl,a         ED 31           Add A to HL (no flags set)
   add  de,a         ED 32           Add A to DE (no flags set)
   add  bc,a         ED 33           Add A to BC (no flags set)
   add  hl,NNNN    ED 34 LO HI     Add NNNN to HL (no flags set)
   add  de,NNNN     ED 35 LO HI     Add NNNN to DE (no flags set)
   add  bc,NNNN     ED 36 LO HI     Add NNNN to BC (no flags set)
   outinb            ED 90           out (c),(hl), hl++
   ldix              ED A4           As LDI,  but if byte==A does not copy
   ldirx             ED B4           As LDIR, but if byte==A does not copy
   lddx              ED AC           As LDD,  but if byte==A does not copy, and DE is incremented
   lddrx             ED BC           As LDDR,  but if byte==A does not copy
   fillde            ED B5           Using A fill from DE for BC bytes
   ld  hl,sp         ED 25           transfer SP to HL
   ld  acc32,hlde    ED 20           transfer HLDE into ACC32
   ld  hlde,acc32    ED 21           transfer ACC32 into HLDE
   ex  acc32,hlde    ED 22           swap ACC32 with HLDE
   inc hlde          ED 37           increment 32bit HLDE
   dec hlde          ED 38           increment 32bit HLDE
   add hlde,a        ED 39           Add A to 32bit HLDE
   add hlde,bc       ED 3A           Add BC to 32bit HLDE
   add hlde,NNNN    ED 3B LO HI     Add NNNN to 32bit HLDE
   sub hlde,a        ED 3C           Subtract A from 32bit HLDE
   sub hlde,bc       ED 3D           Subtract BC from 32bit HLDE
   mirror a          ED 24           mirror the bits in A     
   mirror de         ED 26           mirror the bits in DE     
   push NNNN        ED 8A LO HI     push 16bit immidiate value
   popx              ED 8B           pop value and disguard

   nextreg reg,val   ED 91 reg,val   Set a NEXT register (like doing out($243b),reg then out($253b),val )
   nextreg reg,a     ED 92 reg       Set a NEXT register using A (like doing out($243b),reg then out($253b),A )
** reg,val are both 8-bit numbers

   pixeldn           ED 93           Move down a line on the ULA screen
   pixelad           ED 94           using D,E (as Y,X) calculate the ULA screen address and store in HL
   setae             ED 95           Using the lower 3 bits of E (X coordinate), set the correct bit value in A
```

suborb, 2017-08-16:
Any chance of an ld hl,(sp+dd) - that would be really useful. 

And yes, swapping round to use dehl would allow us to use it without an ex.

aralbrec, 2017-08-16:
They've changed the order to DEHL for us, very nice :)  I'll update the instruction list above.

> Any chance of an ld hl,(sp+dd) - that would be really useful.

It sounds like they are open to that too.

What is missing really?

ld r,(sp+d)
ld rp,(sp+d)
ld (sp+d),r
ld (sp+d),rp

ld ix,sp   ;; mirrors ld hl,sp already added
ld iy,sp

add sp,nn

suborb, 2017-08-16:
That's brilliant. Creating a zxn_rules.1 along the lines of the rabbit version should be easy and allow us to use some of those opcodes in a trivial way.

From the rabbit file there's these extra addressing modes that are useful for C code:

ld hl,(sp+n) - n unsigned byte
ld ix,(sp +n)
ld iy,(sp +n)
ld (sp+n),hl 
ld (sp+n),ix
ld (sp+n),iy
ld hl,(hl + d) - d is signed byte
ld hl,(ix + d)
ld hl,(iy + d)
add sp,d  

Supporting the other pairs instead of hl would be useful, but not hugely critical really.

From the rabbit file, it looks like I made quite a lot of use of bool hl, this basically turns hl into a boolean and sets flags. Thus a comparison to zero is easy and a true boolean value is yielded. In the rabbit world, this is a single opcode which makes it particularly efficient.

These are useful:

and hl, de
or hl, de
xor hl,de
neg hl
rr hl
rl hl

suborb, 2017-08-16:
I've checked in a zxn_rules.1 file, but not hooked it in as of yet, you'll need to add:

    COPTRULESCPU    DESTDIR/lib/zxn_rules.1

to the zxn.cfg.

Test file attached to show it working (probably only with sccz80 at the moment)

[Issue_312_zxn_optimisations.txt](https://github.com/z88dk/z88dk/files/1229312/Issue_312_zxn_optimisations.txt)

aralbrec, 2017-08-17:
There is only one emulator currently that is accepting the opcodes and it's only a partial emulation so I think we should wait a bit before enabling the opcodes outside the assembler.

aralbrec, 2017-08-19:
@pauloscustodio  Any chance the new list up there can be included soon?

A complete emulator (ZEsarUX) is going to implement the instructions so once that's there, I think it's safe to enable them in the entire toolchain.

pauloscustodio, 2017-08-19:
I'm on it...
Shouldn't ```fillde```` be ```fill de``` (as in verb-object)?

pauloscustodio, 2017-08-19:
I've committed the change to implement the current opcodes.
I've changed ```test``` to be the same as the Z180, ```tst N```.

aralbrec, 2017-08-20:
We don't have any control of the opcode names... they should probably conform to existing instructions on other z80 derivatives where they are the same but all we can do is suggest.

test should be tst (z180)
swapnib should be swap (gameboy z80)

I'm not sure if there are any others in there.

pauloscustodio, 2017-08-20:
I've committed the following variants of z80-zxn opcodes:
- ```swap``` as synonym to ```swapnib```
- ```fill de``` as synonym to ```fillde```
- ```test``` as synonym to ```tst```

aralbrec, 2017-08-24:
New instructions that would help most directly would include stack relative addressing.

I looked at two sources for inspiration:

1. The Rabbit family of processors which are based on the z80 and z180.  They were
also constrained by opcode space and added some instructions for their compiler.

https://github.com/z88dk/techdocs/blob/master/rabbit/RabbitInstructions.pdf

2. The Z380 family by Zilog.  They expanded the instruction set and have a full
set of stack relative instructions as well as 16-bit logical, multiplication, 
division and accessing the exx set in a finer grain.  I highly recommend that Victor
look at how Zilog added the instructions while remaining compatible (they defined
escape opcodes in the instruction listing, eg) and borrow some of their names.
Their degree of compatibility won't be as high because the z80 in the next must
also implement crazy undocumented opcodes.

See Appendix B page 228 for an alphabetical listing:
https://github.com/z88dk/techdocs/blob/master/zilog/z380_cpu_um.pdf

Instructions that would help most for the c compilers, given constraints:

* ld hl,(sp+n)
   ld rp,(sp+n)  \
   ld r,(sp+n)   / Take up opcode space, in z380

* ld (sp+n),hl
   ld (sp+n),rp  \
   ld (sp+n),r   / Take up opcode space, in z380

* add sp,d

ld ix,sp

more important if stack relative addressing isn't possible:

* ld hl,(ix+d)
   ld rp,(ix+d)

* ld (ix+d),hl
   ld (ix+d),rp

In the above "d" is a two's complement 8-bit number but "n"
is an unsigned 8-bit number.  The z380 treats "n" as signed
for stack relative addressing too but it doesn't make too
much sense to index below the stack pointer.  If you're sharing
the ix+d logic with sp+n then two's complement will happen.

the rabbit constrained itself to these but the z380 goes whole hog
including constant NNNN operand:

andw hl,de
orw hl,de
xorw hl,de

negw hl
rlw de
rrw hl
rrw de

The rabbit added this:

bool hl (or rp)

to convert non-zero value to 1.

If you can wait another week I will be rewriting the integer math so maybe
something will come out of that.

BTW, the "TEST" instruction seems to be the same as "TST" for z180 and later.
"SWAPNIB" has similar function to the gameboy's swap and the z380's SWAP.
It may be a good idea to use the same mnemonics where it makes sense.

aralbrec, 2017-08-27:
@pauloscustodio @suborb 

A couple of new instructions also added to the main list above:

```
21T*   ldirscale         ED B6           As LDIRX,  if(hl)!=A then (de)=(hl); HL_E'+=BC'; DE+=DE'; dec BC; Loop.
14T*   ldpirx            ED B7           (de) = ( (hl&$fff8)+(E&7) ) when != A

* = guessing
```

ldirscale is going to scale a source graphics up or down in size.  In my mind DE should be increased by one for each step so I'm not clear yet on what the intention is.

ldpirx I'm not clear on its intention either.

aralbrec, 2017-08-28:
I fixed up "ldirscale" in the posts above to match the correct register use:

`21T*   ldirscale         ED B6           As LDIRX,  if(hl)!=A then (de)=(hl); HL_A'+=BC'; DE+=DE'; dec BC; Loop.`

suborb, 2017-08-28:
What's hl_a'? How should it be treated?

aralbrec, 2017-08-28:
h,l,a' is 16.8 fixed point
b',c'  is 8.8 fixed point

You're doing a 24-bit addition:   h,l,a' += 0,b',c'
a' = a' + c' then l = l + b' + carry then h = h + carry.  But no actual flags are affected AFAIK.

suborb, 2017-08-29:
I think we're missing one, latest email suggests 38 opcodes implemented, but we've only got 37 listed above.

aralbrec, 2017-08-29:
There's only 37 now in public.

suborb, 2017-08-29:
Phew. Let's hope there's some stack relative addressing soon.

aralbrec, 2017-08-29:
I guess that's among the 80 instructions under consideration.  I've asked Jim to doublecheck our list and he'll do that tomorrow.

suborb, 2017-09-05:
Playing around with stuff, it would be really good if add hl,nnnn etc set the flags (well carry). This way we can use them as part of a long + constant expression. 

Having said that, if they were adc rather than just add then we could use them wholly.

aralbrec, 2017-09-05:
> I've checked in a zxn_rules.1 file, but not hooked it in as of yet, you'll need to add:
> 
> `COPTRULESCPU    DESTDIR/lib/zxn_rules.1`
> 
> to the zxn.cfg.

I've hooked it in now but for file "zxn_rules.2" for now.  ZEsarUX still does not emulate the new opcodes so waiting for that.  This other rules file only contains a rule to inline the nextreg instruction but it should be added to rules.1 once ZEsarUX is ready.

suborb, 2017-09-05:
More evidence for the need for stack relative addressing. The following is the prologue for getting a variable from the stack:

4 bytes, 21T
```
     ld hl, nnnn
     add hl,sp
```

Using ZXN instructions we get:

6 bytes, 16T:

```
   ld hl,sp
   add hl,nnnn
```

Then there's the l_gint call which is a constant 51T or 24T if inlined. So to access a local variable, sccz80 is at least 7T slower.

We use it a lot in sccz80, so in the md5sum test it makes about a 2% improvement in speed, but a 10% larger binary.

aralbrec, 2017-09-23:
A few more memory management instructions are about to be added:

```
8T*  MMU0 n  ED 80
8T*  MMU1 n  ED 81
8T*  MMU2 n  ED 82
8T*  MMU3 n  ED 83
8T*  MMU4 n  ED 84
8T*  MMU5 n  ED 85
8T*  MMU6 n  ED 86
8T*  MMU7 n  ED 87
```

These ones specify which 8k memory page is banked into an 8k slot of the z80's 64k memory space.
So, eg, `MMU6 100` would put 8k ram page 100 into the 48k-56k memory region.

I'll update the main table above https://github.com/z88dk/z88dk/issues/312#issuecomment-322659205

suborb, 2017-09-23:
d7f998a634246983d0c93d9456e70c88e6976c50 added mmu opcodes to the disassembler.

pauloscustodio, 2017-09-23:
I think we should keep them - its easy to remember. Just give me the new opcodes when they are frozen.

aralbrec, 2017-09-24:
The mmu instructions have been demoted to nextreg and the equivalent byte sequence is listed above https://github.com/z88dk/z88dk/issues/312#issuecomment-322659205

aralbrec, 2017-09-25:
If we're going to do these mmus:

```
12T*  mmu0 NN           ED 91 50 NN      macro: Ram page in slot 0-8k
12T*  mmu1 NN           ED 91 51 NN      macro: Ram page in slot 8k-16k
12T*  mmu2 NN           ED 91 52 NN      macro: Ram page in slot 16k-24k
12T*  mmu3 NN           ED 91 53 NN      macro: Ram page in slot 24k-32k
12T*  mmu4 NN           ED 91 54 NN      macro: Ram page in slot 32k-40k
12T*  mmu5 NN           ED 91 55 NN      macro: Ram page in slot 40k-48k
12T*  mmu6 NN           ED 91 56 NN      macro: Ram page in slot 48k-56k
12T*  mmu7 NN           ED 91 57 NN      macro: Ram page in slot 56k-64k
```

We should probably consider doing these too which can set mmu with register `a`:

```
12T*  mmu0 a            ED 92 50         macro: Ram page in slot 0-8k
12T*  mmu1 a            ED 92 51         macro: Ram page in slot 8k-16k
12T*  mmu2 a            ED 92 52         macro: Ram page in slot 16k-24k
12T*  mmu3 a            ED 92 53         macro: Ram page in slot 24k-32k
12T*  mmu4 a            ED 92 54         macro: Ram page in slot 32k-40k
12T*  mmu5 a            ED 92 55         macro: Ram page in slot 40k-48k
12T*  mmu6 a            ED 92 56         macro: Ram page in slot 48k-56k
12T*  mmu7 a            ED 92 57         macro: Ram page in slot 56k-64k
```

These I had use for tonight.

I don't know if we should go so far as to read the memory state:

```
ld a,mmu0
ld a,mmu1
ld a,mmu2
ld a,mmu3
ld a,mmu4
ld a,mmu5
ld a,mmu6
ld a,mmu7
```

These are actually complex, have side effects, and probably shouldn't be done.

```
ld a,mmu0
=
ld	bc,0x243b
ld	a,0x50
out	(c),a
inc	b
in	a,(c)
```

pauloscustodio, 2017-09-25:
I'll implement the ```mmuN a``` and ```mmu N,a```.

As for the ```ld a,mmuN```, I think the ease of use outweights the disadvantages, although the ```ld``` seems inappropriate, as the instruction is doing much more than moving data to A.

Proposal, that we can extend later for other similar uses:
```eval a = mmuN``` - push BC, sequence above, pop BC;
```eval(bc) a = mmuN``` - the same without saving BC; ```eval``` means compute something while using the registers in parentheses, or saving all registers in none are given

suborb, 2017-09-25:
The `ld a,mmuN` pseudo instructions are, I think, better suited to user-defined macros rather than something baked into the assembler.

aralbrec, 2017-09-25:
Yeah I think `ld a,mmuN` equivalents are better suited outside the assembler too.  I don't think they would even be useful for the compilers - I just committed a c inlined macro version of this but I'm looking at making the macro substitution earlier in the compile process so that the peephole optimizers have a go at it and can improve the code.

aralbrec, 2017-10-09:
Some instructions are on the bubble due to fpga space:

`fillde`, `mul dehl` (to be replaced by 8-bit multiply), `ld hl,sp`, `ld a32,dehl`, `ld dehl,a32`, `ex a32,dehl`, `popx`, `ex hl,ix`, `ex hl,iy`, `inc dehl`, `dec dehl`, `add dehl,a`, `add dehl,bc`, `add dehl,nnnn`, `sub dehl,a`, `sub dehl,bc`

Still waiting to hear what the final list is.

suborb, 2017-10-09:
Never heard of some of them. :)

Given this news I take it to mean there’s no hope of stack relative. So combined with the possible removal of the opcodes we do actually use I’ll make a note to strip out custom support for zxn from sccz80 and the cpu specific opt file when this is confirmed.

aralbrec, 2017-10-11:
Yes stack relative is very unlikely.

A copper unit has been added like the amiga copper.  It understands two instruction - wait and move - that is triggered on raster line and horizontal position.  I'm not sure yet if its instruction queue is stored in ram or on the fpga (probably the latter).  The queue size is 1k or 512 instructions.

The point is should z80-zxn assemble these instructions as if they were z80 so that they can be inlined in regular asm?  Or be done by macro (currently that's m4 which would require parameters in parentheses which is different from how they want the mnemonics defined like normal `wait a,b`).

aralbrec, 2017-10-12:
Maybe namespacing the instructions would ensure they are easy to see in source and won't be mixed up with regular opcodes:

```
copperlist:

   cu.wait 193,0   ;; wait for border
   cu.move 7,3     ;; set cpu to 14 MHz
   cu.wait 311,0   ;; wait for line before start of screen
   cu.move 7,2     ;; set cpu to 7 MHz
   cu.move 0,0     ;; end program
```

aralbrec, 2017-10-25:
I've updated the z80-zxn instruction list with how it is now.  It is likely the instructions marked REMOVED are permanently gone due to fpga space but if you'd like to wait to make changes until the list is final, that's ok too.

However, I would like to suggest the copper instructions be added to z80asm.  I am writing some copper code now and it is a royal pain to figure out the hex opcodes by hand.  The copper is an independently running unit that understands two instructions and can make changes to the computer's configuration in sync with the video display (like the amiga).

The choice is macro or z80asm doing them but if I go macro route, that's m4 involved and that means the instructions will be in slightly less nice syntax `WAIT(line,horz)` instead of `cu.write lin,hor` but more importantly the asm file will have to be changed to .asm.m4 which still scares some people for such a simple application.

The idea is to namespace them with leading "cu." so they are not confused with z80 instructions:

```
cu.write LINE,HOR   ->  16-bit encoding 0x8000 + (HOR << 9) + LINE
(0<=LINE<=311, 0<=HOR<=55)

cu.move REG,VAL  -> 16-bit encoding (REG << 8) + VAL
(0<= REG <= 127, 0 <= VAL <= 255)
```

I'm hoping a few more will be added but we'll see.

This one is not defined as a separate copper instruction (it is equivalent to `cu.move 0,0`) but it might be nice to have anyway:

```
cu.stop   -> 16-bit encoding 0x0000
```

z80asm already uses dots for degroup members I think but if the opcodes are parsed first I don't think there is a problem?

Thoughts?

pauloscustodio, 2017-10-25:
Parsing the cu.xxx opcodes is no problem and can be added easily to z80asm. I will make them appear only with --cpu=z80-zxn

Just out of curiosity: how is the copper processor started on a list of instructions?

aralbrec, 2017-10-25:
The copper's instruction memory is 1k (512 copper instructions max) and is on the fpga chip.  You write to the copper's instruction memory via a couple of NEXTREG registers:

```
(R) 0x60 (96) => Copper data
  bits 7-0 = Byte to write at "Copper list"
  Note that each copper instruction is composed by two bytes (16 bits).

(R) 0x61 (97) => Copper control LO bit
  bits 7-0 = Copper list index address LSB.
  After the write, the index is auto-incremented to the next memory position.
  (Index is set to 0 after a reset)

(R) 0x62 (98) => Copper control HI bit
   bits 7-6 = Start control
       00 = Copper fully stoped
       01 = Copper start, execute the list, then stop at last adress
       10 = Copper start, execute the list, then loop the list from start
       11 = Copper start, execute the list and restart the list at each frame
   bits 1-0 = Copper list index address MSB
```

So you would send the program by setting the index to 0 in registers 0x61 / 0x62 (0x62 would also have the copper at fully stopped).  And then you would write the instructions by outing to nextreg 0x60 which auto-increments the internal index.  To start it, you would write the appropriate bits to 0x62 and you would normally choose the mode that restarts with vbi (start control=11).

aralbrec, 2017-10-25:
I've corrected `cu.write` to `cu.wait` - it's a wait for a particular beam position on screen.

feilipu, 2017-10-25:
Are those ZXN folk are aware the the `MUL` instruction is now a capability subset of the z180 `MLT` ?

`MUL` should at least be equivalent to z180 version, to prevent years of confusion, imho.

Is there a forum link to point this out?

aralbrec, 2017-10-25:
`mul d,e`, `test NN` and `swapnib` have equivalents in the z180 and other processors.  Yes the mnemonics should be changed and I will point that out.  Are there any others anyone has spotted?

feilipu, 2017-10-25:
Thanks.

I was more wondering why / whether the intention was that `MUL` would only work with `D` and `E` ?  Z180 `MLT` can act on any 16 bit register. 

And `TST` should be the same name too, yes. Thanks.

aralbrec, 2017-10-25:
I'm not sure why it's only `d` and `e`.  The z80 is not in fact being changed - it's more like logic beside the z80 and most instructions are aimed at helping with games on layer 2 which involves a lot more memory pushing than the original machine (8x).   The z180 and later derivatives simply replaced the undocumented opcode space with their new instructions but the z80-zxn can't do that because software used the undocumented instructions.  Replacing them leads to incompatibility so your more constrained over what undocumented opcodes you can replace.

Here's a first copper program if you're interested:

```
;; COPPER LIST

;; This is an asm file that is passed through m4 before assembling.
;; M4 is a sophisticated macro processing facility common on linux/unix
;; machines and is incorporated into z88dk as a macro pre-processor.
;; Its syntax can look odd for people not familiar with the tool.

;; To see what m4 does to this file, ask zcc to process it
;;    zcc +zxn -vn -m4 copper.asm.m4

include(`z88dk.m4')                ; useful m4 macros
dnl#include(`config_private.inc')      ; symbol values for zxn target

define(`__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM', 0x15)

define(`CU_WAIT', `defw 0x`'eval((0x8000 + (($1)&(0x1ff)) + ((($2)&(0x3f))<<9)),16,4)')
define(`CU_MOVE', `defw 0x`'eval(((($1)&(0x7f))<<8) + (($2)&(0xff)),16,4)')
define(`CU_STOP', `defw 0x0000')

SECTION rodata_user

PUBLIC _copper
PUBLIC _copper_end

_copper:

   ; There are two hi-res windows (x,width,y,height) = (2,28,1,19) and (32,28,3,19)
   
   ; Further these are 64x24 character coordinates but the copper operates in 32x24
   ; character coordinates so we want even sizes for the 64-col (x,width) to divide
   ; by two to get exact 32x24 x coordinates.
   
   ; To make the timex hi-res display appear on top of the scrolling layer 2
   ; we'll make use of the nextreg layer priorities register 0x15 (__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM)
   ; and have layer 2 on top by default and then ula on top for the hi-res terminals.
   
   ; layer 2 is on top by default
   ; (this part is not actually necessary because layer 2 on top set later)
   
   CU_WAIT(193, 0)                                     ; wait just past visible screen area
   CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0)   ; layer 2 on top
   
   ; the first window is on its own for vertical y char coord in [1,3)
   ; change to ula on top while beam is over first window
   
   Z88DK_FOR(`LOOP', eval(1*8), eval(3*8-1),
   `
      CU_WAIT(LOOP, eval(2/2))                               ; wait for x = 2/2 start first window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0x08)   ; ula on top
      
      CU_WAIT(LOOP, eval((2+28)/2))                          ; wait for x=(2+28)/2 end first window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0)      ; layer 2 on top
   ')
   
   ; two windows appear on the same scan line for y char coord in [3,20)
   ; change to ula on top while beam is over either window

   Z88DK_FOR(`LOOP', eval(1*8), eval(20*8-1),
   `
      CU_WAIT(LOOP, eval(2/2))                               ; wait for x = 2/2 start first window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0x08)   ; ula on top
      
      CU_WAIT(LOOP, eval((2+28)/2))                          ; wait for x=(2+28)/2 end first window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0)      ; layer 2 on top
      
      CU_WAIT(LOOP, eval(32/2))                              ; wait for x = 32/2 start first window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0x08)   ; ula on top
      
      CU_WAIT(LOOP, eval((32+28)/2))                         ; wait for x=(32+28)/2 end first window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0)      ; layer 2 on top
   ')

   ; the second window is on its own for vertical y char coord in [20,22)
   ; change to ula on top while beam is over second window
   
   Z88DK_FOR(`LOOP', eval(20*8), eval(22*8-1),
   `
      CU_WAIT(LOOP, eval(32/2))                              ; wait for x = 32/2 start second window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0x08)   ; ula on top
      
      CU_WAIT(LOOP, eval((32+28)/2))                         ; wait for x=(32+28)/2 end second window
      CU_MOVE(__IO_REGISTRY_REG_SPRITE_LAYER_SYSTEM, 0)      ; layer 2 on top
   ')

   CU_STOP

_copper_end:
```

Still working out how to do it nicely and it should be better with the pseudo-instructions in z80asm.

suborb, 2017-10-25:
It makes the 16x16 multiply a fair bit slower (I think the following is correct?), from 4T to 74T, though still less than then 5-600T of the vanilla z80.

```
int mult(int v1, int v2)
{
#asm
        ld      hl,2
        add     hl,sp
        ld      e,(hl)
        inc     hl
        ld      d,(hl)
        inc     hl
        ld      a,(hl)
        inc     hl
        ld      h,(hl)
        ld      l,a

l_mult:

        ld      a,d             ;a= v2:msb
        ld      d,h             ;v1:msb
        ld      c,e             ;c = v2:lsb
        ld      b,l             ;b = v1:lsb
        mul     de
        ex      de,hl           ;hl = v1:msb * v2:lsb
        ld      d,a             ;
        mul     de              ;de = v2:msb * v1:lsb
        add     hl,de
        ld      e,c             ;e = v2:lsb
        ld      d,b             ;d = v1:lsb
        mul     de              ;de = v1:lsb * v2:lsb
        ld      h,l
        ld      l,0
        adc     hl,de
        ret
}
```

feilipu, 2017-10-26:
> The z180 and later derivatives simply replaced the undocumented opcode space with their new instructions but the z80-zxn can't do that because software used the undocumented instructions.

That is fair enough, and the last thing they want would be to break any z80 compatibility...

> Replacing them leads to incompatibility so you're more constrained over what undocumented opcodes you can replace. I'm not sure if that has anything to do with being picky about how many `mlt` there are.

Except that the opcodes for the z180 `MLT` are already in the `ED` space (`ED4C`, `ED5C`, `ED6C`, and `ED7C`, for `BC`, `DE`, `HL`, and `SP` respectively) and, if I read correctly, they are only an undocumented alias for the __real__ opcode `NEG` (`ED44`).

With respect, it sounds to me as if they are just being a tiny bit lazy.

It would take quite a few cycles off the above 16x16 multiply routine not to have to juggle the MSB and LSB bytes around. But more importantly, `MLT` compatibility would save having to maintain two separate math solutions, forever...

aralbrec, 2017-10-26:
I've corrected above - copper is big endian so the 16-bit value should be written that way when generating code.

aralbrec, 2017-10-26:
@suborb I think there is an error in your multiply:

```
hl = de * hl
de = x, hl = y

z80-zxn 

4    ld  a,d     ; a = xh
4    ld  d,h     ; d = yh
4    ld  h,a     ; h = xh
4    ld  c,e     ; c = xl
4    ld  b,l     ; b = yl
8    mul de      ; yh * xl
4    ex de,hl
8    mul de      ; xh * yl
11   add hl,de   ; add cross products
4    ld e,c
4    ld d,b
9    mul de      ; yl * xl
4    ld a,l      ; cross products LSB
4    add a,d     ; add to MSB final      
4    ld h,a
4    ld l,e      ; hl = final
     ret

84 cycles, 19 bytes
```

```
hl = de * hl
de = x, hl = y

z180-like

4    ld  a,d     ; a = xh
4    ld  d,h     ; d = yh
4    ld  h,a     ; h = xh
4    ld  c,e     ; c = xl
4    ld  b,l     ; b = yl
8    mlt de      ; yh * xl
8    mlt hl      ; xh * yl
11   add hl,de   ; add cross products
8    mlt bc      ; yl * xl
4    ld a,l      ; cross products LSB
4    add a,b     ; add to MSB final
4    ld h,a
4    ld l,c      ; hl = final
     ret

71 cycles, 16 bytes
```

suborb, 2017-10-26:
I'm not sure why I included the carry either. Think my head must've been in 32 bit land.

feilipu, 2017-10-28:
Having a go at a full 16x16 multiply, I get this.
I can't see a way to do it without using the stack or index registers.
```
l_mult:
    ld  a,d     ; a = xh        /  4
    ld  d,h     ; d = yh        /  4
    ld  h,a     ; h = xh        /  4
    ld  c,d     ; c = xh        /  4
    ld  b,h     ; b = yh        /  4
    push bc     ;               / 11
    ld  c,e     ; c = xl        /  4
    ld  b,l     ; b = yl        /  4

    mlt de      ; yh * xl       / 17
    mlt hl      ; xh * yl       / 17
    add hl,de   ; add cross products            / 11

    ld  e,h     ; MSB yh * xl + xh * yl         /  4
    ld  h,l     ; LSB yh * xl + xh * yl         /  4
    ld  l,0     ; 0 LSB                         /  7
    ld  d,l     ; 0 MSB                         /  4

    mlt bc      ; LSB product yl * xl           / 17
    add hl,bc   ; add LSB X to LSB product      / 11
    ex de, hl   ; move LSW to DE, MSB X to HL   /  4

    pop bc      ;                               / 10    
    mlt bc      ; MSB product yh * xh           / 17    

    adc hl,bc   ; add MSB X to MSB product + C  / 15
    ex de, hl   ; move MSW to DE, LSW to HL     /  4
    ret
```
Thoughts?

aralbrec, 2017-10-29:
For the newlib, the complete set of math functions that have to be replaced are in:
https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/integer

Those entry points are what the compilers and library call into and they are mainly stubs that select between the fast integer functions (they do loop unrolling and leading zero elimination) and the small integer functions in the small subdirectory.  The z180 / z80-zxn won't likely benefit from unrolling for multiplication at least so the small subdir is probably what has to be replaced for the multiply.   zsdcc does distinguish between 8bit and 16bit multiply so in fact some of the library entry points should probably be turned into inlined mlt instructions.

aralbrec, 2017-11-06:
I think there are some issues in that 16x16->32 multiply to do with carries not being added at the right places.

I tried this one which I think is correct:

```
l_small_mul_32_16x16:

   ; multiplication of two 16-bit numbers into a 32-bit product
   ;
   ; enter : de = 16-bit multiplicand = y
   ;         hl = 16-bit multiplicand = x
   ;
   ; exit  : dehl = 32-bit product
   ;         carry reset
   ;
   ; uses  : af, bc, de, hl

4   ld b,l                      ; xl
4   ld c,d                      ; yh
4   ld d,l                      ; xl
4   ld l,c
11  push hl                     ; xh yh
4   ld l,e                      ; yl

   ; bc = xl yh
   ; de = xl yl
   ; hl = xh yl
   ; stack = xh yh

17  mlt de                      ; xl * yl

17  mlt bc                      ; xl * yh
17  mlt hl                      ; xh * yl

7   add hl,bc                   ; sum cross products

4   sbc a,a
6   and $01
4   ld b,a                      ; carry from cross products
4   ld c,h                      ; LSB of MSW from cross products
   
4   ld a,d
4   add a,l
4   ld d,a                      ; de = final LSW

9   pop hl
17   mlt hl                      ; xh * yh

10  adc hl,bc                   ; hl = final MSW
3   ex de,hl

; 158 cycles, 27 bytes

   ret
```

z180 timing is different so it's a little strange.

aralbrec, 2017-11-14:
There's a bug in z80asm's copper instructions:

"test.asm"
```
SECTION rodata_user

PUBLIC _copper
PUBLIC _copper_end

EXTERN __REG_SPRITE_LAYER_SYSTEM
EXTERN __RSLS_LAYER_PRIORITY_SUL, __RSLS_LAYER_PRIORITY_SLU

_copper:

   cu.wait 100,1
   cu.move 0x15, 0x08
   cu.move __REG_SPRITE_LAYER_SYSTEM, __RSLS_LAYER_PRIORITY_SUL
   cu.wait 200,1

_copper_end:
```

zcc +zxn -c -vn --list -clib=sdcc_iy test.asm

Produces this list file:

```
1     0000              MODULE test_asm
2     0000              LINE -1, "test.asm"
-1    0000              
0     0000              
1     0000              SECTION rodata_user
2     0000              
3     0000              PUBLIC _copper
4     0000              PUBLIC _copper_end
5     0000              
6     0000              EXTERN __REG_SPRITE_LAYER_SYSTEM
7     0000              EXTERN __RSLS_LAYER_PRIORITY_SUL, __RSLS_LAYER_PRIORITY_SLU
8     0000              
9     0000              _copper:
10    0000              
11    0000  82 64          cu.wait 100,1
12    0002  15 08          cu.move 0x15, 0x08
13    0004                 cu.move __REG_SPRITE_LAYER_SYSTEM, __RSLS_LAYER_PRIORITY_SUL
14    0004  82 C8          cu.wait 200,1
15    0006              
16    0006              _copper_end:
17    0006              
```

Here you can see that if the copper instruction contains extern symbols, nothing is generated.

pauloscustodio, 2017-11-14:
Sorry, my mistake. I assumed constant expressions, I.e. known at compile time.
I will change them to link time expressions.

[44 more replies]

commit 444e6506 (2017-08-11): Fix #312 - z80 variant for the zx next

Assemble opcodes added by the ZX Next Z80 core with the new --cpu=z80-zxn.

swapnib    ED 23    A bits 7-4 swap with A bits 3-0
mul        ED 30    multiply HL*DE = HLDE (no flags set)
add  hl,a  ED 31    Add A to HL (no flags set)
add  de,a  ED 32    Add A to DE (no flags set)
add  bc,a  ED 33    Add A to BC (no flags set)
outinb     ED 90    out (c),(hl), hl++
ldix       ED A4    As LDI,  but if byte==A does not copy
ldirx      ED B4    As LDIR, but if byte==A does not copy
lddx       ED AC    As LDD,  but if byte==A does not copy, and DE is incremente [... cut]
commit 6edf0113 (2017-08-19): Fix #312 - implement more opcodes for z80-zxn

Impemented test N as ```tst N``` to use the same opcode as the Z180.
commit ecb35fb0 (2017-08-28): Fix #312: add ldirscale and ldpirx to z80-zxn
commit 596bda07 (2017-09-23): Fix #312: add new MMU opcodes to z80-zxn
closed 2018-12-24 by pauloscustodio