z88dk/z88dk #313: Possible bug in bit_fx code

Fabrizio-Caruso, 2017-08-11:
(I have posted this message in the forum. Sorry for re-posting it. Where should I post messages about possible bugs?)

If I use bit_fx in my game
(https://github.com/Fabrizio-Caruso/ASCII-CHASE)
I get reproducible crashes at a given place
zcc +zx -startup=1 -clib=sdcc_iy -vn -DSPECTRUM_32COL -D__SPECTRUM__ -create-app -o ...

Somewhere in my code:
...
CLEAR_SCREEN();
sleep(1);
PING_SOUND(); /*= bit_fx(bitfx_19); */
printExtraLife();
sleep(1);
++lives;
...

Remark 1: No crashes if bit_fx is avoided (remove PING_SOUND above solves the crashes)

Remark 2: No crashes in the rest of the code with neither of the two compilers in Z88DK nor CC65

Remark 3: My game is no longer using malloc. The crash is independent of this because the crash occurs even in the version of my game where I still used malloc.

aralbrec, 2017-08-11:
I extracted the relevant bit of code:

```
#include <stdio.h>
#include <string.h>
#include <sound/bit.h>

#define PING_SOUND() bit_fx(bitfx_19)
#define CLEAR_SCREEN() {printf("\xc");};

unsigned char lives;

void sleep(unsigned int sec) 
{ 
	unsigned long i; 
	for(i=0;i<sec*400UL;++i){}; 
} 

#define X_OFFSET 0
#define Y_OFFSET 3

#define XSize 32
#define YSize 24

#define COLOR_WHITE 2
#define TEXT_COLOR COLOR_WHITE
#define SET_TEXT_COLOR(c) {printf("\020%c",c);};

#define gotoxy(x,y) printf("\x16%c%c",x+1,y+1); 
#define PRINTF(x,y,str,val) {gotoxy(x+X_OFFSET,y+Y_OFFSET); printf(str,val); };

void printCenteredMessage(char *Text)
{
	SET_TEXT_COLOR(TEXT_COLOR);
	PRINTF((XSize - strlen (Text)) / 2, YSize / 2,"%s", Text);
}

void printExtraLife(void)
{
	printCenteredMessage("E X T R A   L I F E ! ! !"); 
	sleep(1);
}

void main(void)
{
	CLEAR_SCREEN();
	sleep(1);
	PING_SOUND();
	printExtraLife();
	sleep(1);
   ++lives;
}
```

zcc +zx -vn -startup=1 -clib=sdcc_iy test.c -o test -create-app

This works fine.

I will have to take a look at the code generation around this bit in the actual setting.

aralbrec, 2017-08-11:
> (I have posted this message in the forum. Sorry for re-posting it. Where should I post messages about possible bugs?)

Either place is ok.  Here is better for a short bug report and the forum is probably better for in depth discussions.

Fabrizio-Caruso, 2017-08-11:
Thanks!

I cannot exclude my code from doing something stupid even though my code seems fine and has no issues in other targets.

The crash only occurs if a specific action in the game is performed before (you beat the skull). Beating the guy involves the following innocent-looking code:
"
	unsigned char i;
	invincibleGhost._status = 0;
	DELETE_INVINCIBLE_GHOST(invincibleGhost._x,invincibleGhost._y, invincibleGhost._imagePtr);
	invincibleGhost._x=XSize-2; invincibleGhost._y=YSize-2;
	invincibleGhostAlive = 0;
	for(i=0;i<4;++i)
		EXPLOSION_SOUND();
	points+=INVINCIBLE_GHOST_POINTS;
	if(missileLevel())
	        points+=INVINCIBLE_GHOST_POINTS;
	displayStats();
"
and this may also imply that some other specific portion of my code is executed...
I will try to reproduce the smallest possible piece of code that produces a crash.

  Fabrizio

aralbrec, 2017-08-11:
Another common cause for mysterious crashing is the stack.  Make sure there is enough room for it (256 bytes is usually plenty).

When using zsdcc on the spectrum, having the rom interrupt routine running will cause crashing too.  But since you are doing nothing with interrupts you should be ok because the crt disables them on startup.

aralbrec, 2017-08-11:
> Another common cause for mysterious crashing is the stack. Make sure there is enough room for it (256 bytes is usually plenty).

With newlib compiles the default stack location is 65368:
https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/zx/crt_config.inc#L28

You can change that via pragma or the compile line.  If you have several pragmas, it's usually best to consolidate them in a separate file.

On the compile line:

zcc .... -pragma-define:REGISTER_SP=-1 ...

You may want to try that first to see if problems go away.  The -1 tells the crt not to move SP so that the program will use the stack location set by the basic loader in the tap file, which will place it below your program.

Fabrizio-Caruso, 2017-08-11:
With the pragmatic I get a basic loader that does not load the game.

What should I do? 

No mallocs in my code.

aralbrec, 2017-08-11:
Both pragmas have to be in:

zcc ... -pragma-define:REGISTER_SP=-1 -pragma-define:CLIB_MALLOC_HEAP_SIZE=0 ...

You should see them in the zcc_opt.def file that is generated.  If the heap size is not set to zero, the crt will determine a negative heap size and exit without running the program.

Fabrizio-Caruso, 2017-08-11:
Tested and it still crashes at the very same moment.
Only way to avoid it is to remove the bit_fx call...

It could be my code but I have no idea.
And it is sort of weird that it works fine with all other targets.

Is there a simple way to increase the size of the stack?

aralbrec, 2017-08-11:
You have a lot of stack space when REGISTER_SP=-1 so that will not be it.  It was worth a try.
I'll have a look at the generated code this evening.

Fabrizio-Caruso, 2017-08-11:
I really don't know what is going on...
If I use int instead of long inside sleep... I get no crash...

 void sleep(unsigned int sec)
 {
-       unsigned long i;
-       for(i=0;i<sec*400UL;++i){};
+       unsigned int i;
+       for(i=0;i<sec*400;++i){};
 }

Maybe my code is wrongly writing somewhere in some memory location
and depending on the exact code size, this generates a crash, random behavior or no observable behavior.

aralbrec, 2017-08-11:
Another thing to try is compiling with clib=new instead of clib=sdcc_iy to rule out compiler.

Fabrizio-Caruso, 2017-08-12:
@aralbrec, clib=new won't compile with lots of errors:
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1525: symbol '_in_wait_key' n
ot defined
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1526: symbol '_in_wait_nokey'
 not defined
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1538: symbol '_in_wait_key' n
ot defined
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1539: symbol '_in_wait_nokey'
 not defined
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1616: symbol '_in_wait_key' n
ot defined
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1617: symbol '_in_wait_nokey'
 not defined
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 2486: symbol '_in_inkey' not
defined
7 errors occurred during assembly
Errors in source file c:/z88dk/\libsrc\_DEVELOPMENT\target\zx\zx_crt.asm:
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1525: symbol '_in_wait_key' n
ot defined
                   ^ ----       call    _in_wait_key
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1526: symbol '_in_wait_nokey'
 not defined
                   ^ ----       call    _in_wait_nokey
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1538: symbol '_in_wait_key' n
ot defined
                   ^ ----       call    _in_wait_key
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1539: symbol '_in_wait_nokey'
 not defined
                   ^ ----       call    _in_wait_nokey
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1616: symbol '_in_wait_key' n
ot defined
                   ^ ----       call    _in_wait_key
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc8D0813.asm' line 1617: symbol '_in_wait_nokey'
 not defined

aralbrec, 2017-08-12:
How can I reproduce the bug?  I changed the sleep function back to long and am trying to reproduce the bug with sdcc_iy compile:

zcc +zx -vn -startup=1 -clib=sdcc_iy -DSPECTRUM_32COL -create-app ... -o ACM

The problems with the sccz80 compile are coming because you're redefining the input functions, eg in file "input.h" you have `unsigned int in_inkey(void);`

This definition is masking what is defined in <input.h>.  The same thing is happening with the other symbols in file "input_macros.h".  There are macros defined in the system header files that ensure the most efficient calling method is used for those functions and it just so happens that the zsdcc macros are still working (I think or else you are getting less efficient calls).

These function declarations should be removed in favour of including <input.h>

aralbrec, 2017-08-12:
I can see the same thing in spectrum_sounds.h:

```
#include <sound.h>
void bit_fx(void (*bfx)(void));
```

<sound.h> defines that function prototype already and then you are repeating it.  Because of the use of macros in the system headers and the naming used for library functions, this can interfere with calling into the library.  It would be better to comment these out if you want to document them.

Fabrizio-Caruso, 2017-08-12:
Thanks A LOT!
Thanks for all the nice support. This is encouraging me to continue on my project!

In order to reproduce the problem, you need to take my code excluding the very latest commits (i.e., all commits from Aug 11th, e.g., stop at revision dfeeceb).

Then you need to kill the skull in the game in the 5th level by taking the gun and shooting at him (the bullet may go through him and that does not count as a hit)
This can be greatly sped up by setting the following constants in settings.h:
INITIAL_LEVEL 5
MIN_INVINCIBLE_GHOST_HITS 1
(Invert their respective values)

The game then crashes after the level bonus message.

I can avoid the crash by removing the call to the sound routine or changing the code for the sleep function. All of this does not make much sense and needs to be understood.

  Fabrizio

Fabrizio-Caruso, 2017-08-12:
@aralbrec, the prototypes are there because I could not get it to compile otherwise. I used a lazy solution.

  Fabrizio

Fabrizio-Caruso, 2017-08-16:
@aralbrec, I have made it more simple to see the crash.
(1) Just take the latest version from my git repo and use -DDEBUG_CRASH.
(2) Now you start straight at the 5th level where you can immediately fetch the gun
(3) Shoot at the yellow skull that appears a few seconds after level start
(4) After killing the skull press any button and the game crashes.

-DDEBUG_CRASH is only changing the definition of sleep with long type instead of int.
it is also changing the initial level and number of hits for the skull just to speed up the crash.

Fabrizio-Caruso, 2017-08-16:
How can I avoid prototyping 
unsigned int in_inkey(void) in my code, which is using a macro,  (in my own input.h file):
		unsigned int in_inkey(void);
		#define MOVE_PLAYER() {kbInput = in_inkey(); movePlayerByKeyboard(kbInput)} ;

My code is including input.h in multiple files including my own input.h and my main.c.

If I remove the prototype I get in_inkey is not found and the compiler "suggests" to prototype it.

aralbrec, 2017-08-17:
> could an unnecessary prototype declaration cause the crash?

I am wondering if that's the case.  It may screw up the stack but I will have to look at the generated code to be sure.

Thanks for simplifying, I will have a look now.

aralbrec, 2017-08-17:
> could an unnecessary prototype declaration cause the crash?
On the latest version, I can create the bug by inserting an unnecessary prototypes:
including void bit_fx(void (*bfx)(void));

Yes this is causing incorrect code to be generated.  I can see the compiler pushing params on stack and calling the fastcall version of bit_fx when the param should be passed via register.

So I think the crashes are due to the extra prototypes.  Can you try again after eliminating those?

Fabrizio-Caruso, 2017-08-17:
THANKS! Thanks a lot!
We now have a culprit!
As I wrote I did not include prototypes just because I like them. I had to insert them to get the compiler to compile. The compiler itself suggests using prototypes. I have got rid of the bit_fx prototype in the latest version.

On the other hand I have not been able to remove the input.h prototypes (e.g., in_inkey). I am including input.h in multiple files including the ones where the compiler complains about not finding the function. Are they equally bad?

I am going to further simplify my input code which is way too complicated and try again.

aralbrec, 2017-08-17:
> On the other hand I have not been able to remove the input.h prototypes (e.g., in_inkey). I am including input.h in multiple files including the ones where the compiler complains about not finding the function. Are these other prototypes as bad as the ones for bit_fx?

Yes the same thing may happen.  The header is telling the preprocessor to change function name to use one calling method and the compiler is generating code for another calling method according to your protoypes.

The library's input.h uses "_INPUT_H" as ifdef guard.  I don't think your code defines that anywhere?  I can see though that we need to mangle up these if-guard names a bit better.

You can see what the preprocessor does with a file by running something like:

`zcc +zx -vn -E -clib=sdcc_iy file.c`

The output will be in "file.c.i".  That file is going to be pretty big but what you're looking for is whether input.h was included.  Do a search for one of the functions that is apparently missing to see if it's there, or do a search for the guard "_INPUT_H" to see if it's there.

I'll have a look later today.

aralbrec, 2017-08-17:
The last problem with input.h is happening because you have another input.h in the working directory.  So `#include <input.h>` is actually including your local input.h instead of the system input.h and none of the prototypes are being defined.  #316 has been opened to solve the issue.

I guess the only thing to do at the moment is rename input.h to something else or temporarily copy the proper prototypes needed from the system [input.h](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sdcc/input/input_zx.h).

aralbrec, 2017-08-19:
A fix for the header issue is currently committed for zsdcc + zx target, so I think that deals with all the issues you've found.

The fix was committed after the nightly build unfortunately so if you use that, you'll have to wait for the [next one](https://www.z88dk.org/forum/viewforum.php?id=19) on Aug 20.

aralbrec, 2017-08-25:
@Fabrizio-Caruso  
I think all the issues are resolved in this thread?  I'm just waiting to close :)

Fabrizio-Caruso, 2017-08-25:
@aralbrec, I think so myself. I will test further but you can close this thread. Thanks a lot for the support!

closed 2017-08-25 by aralbrec