z88dk/z88dk #316: cpp, zsdcpp:  pre-processors do not treat system include path differently from user include path

aralbrec, 2017-08-17:
The search path is no different for files included within angle brackets and quotes:

[code]
#include <stdio.h>

void main(void)
{
   return;
}
[/code]

If you create a stdio.h in the current directory and compile, the local file will be included instead of the system file.

If you don't have a local stdio.h but instead do a `#include "stdio.h"` the system include file will be found.

This bug is occurring in #313 where a user-supplied input.h is present which is being included in preference to the system input.h when `#include <input.h>` is used.

suborb, 2017-08-17:
I suspect sdcpp supports -isystem. Zcpp certainly won't. 

So if you change the appropriate -I over hopefully the problem will go away for sdcc

aralbrec, 2017-08-18:
Hmm, I've put -isystem in but zsdcpp is still preferring the local header.

suborb, 2017-08-18:
Maybe a space after the option?

I looked at zcpp earlier and it goes through the directories in order, presumably the order on the command line. 

I'll have a play later though.

aralbrec, 2017-08-18:
I tried space too but it's still searching the local dir first.  But I did find this:
```
The most commonly-used option is -Idir, which causes dir to be searched after the current directory (for the quote form of the directive) and ahead of the standard system directories.
```

which means -I trumps the system search path.

It looks like -iquote and -isystem is what is needed instead of -I.  I'll put those in as equivalent to -I for zcpp.

suborb, 2017-08-18:
It looks like there's a few issues coming into play in my tests:

* The .cfg file OPTIONS line has -I. in it
* zcc pushes user specified -I ahead of the system path
* zcpp doesn't have the distinction between quote and system path

aralbrec, 2017-08-18:
Yes the cfg files have to be changed to use -iquote and -isystem, zcc has to quote the filenames when it sees those options and zcpp has to have iquote and isystem undestandable as -I.

zcpp is doomed to get this wrong until it is updated to meaningfully use iquote and isystem.  The -I behaviour is correct though - it does get applied ahead of isystem.

suborb, 2017-08-18:
I think the above implements the feature for zcpp, it's a bit ugly, but lets see if it works.

I've found a few #include problems in the classic library as result as well.

aralbrec, 2017-08-19:
I've made the necessary changes to zcc and I made -iquote and -isystem synonyms for -I in zcpp so things would continue to work as normal.  The changes are simple so any conflict should be easy to fix with your zcpp additions.

I made changes to zx.cfg too to use -iquote and -isystem and it seems to work for zsdcpp/zsdcc.

The rest of the cfg files will have to be changed similarly but let's see if it also works with zcpp when you've committed that.

aralbrec, 2017-08-19:
Before I forget :- I didn't make any change to how classic compiles add the system library in zcc.   -isystem will have to be added there too.

suborb, 2017-08-19:
I've updated the iquote_isystem_316 branch to include your changes and updated all the config files.

commit 01dbfb1d (2017-08-20): Fix #316: zcpp, zsdcpp now honour quoted and system include paths
closed 2017-08-20 by aralbrec