z88dk/z88dk #323: MSX reboots or freezes while loading my game (no heap used in the code)

Fabrizio-Caruso, 2017-08-20:
I am having an issue with loading my game with any MSX emulator emulating an MSX 1, 64k or 128K RAM (fMSX freezes and blueMSX reboots).
Remark: The boot screen on the emulatated MSX shows only 23432 bytes available to the BASIC.

I compile with
zcc +msx  -vn  -D__MSX__ -lndos -create-app -o %deliverables%\msx_experimental.prg <c files>
which generates msx_experimental.cas

My game uses NO heap at all (no malloc nor any other kind of explicit dynamic memory allocation).

Remark: My game works with about 16 different targets and multiple compilers.

Fabrizio-Caruso, 2017-08-23:
If I use the light version of my game (a sub-set of the same code), I manage to get the game to load and run. So the problem comes from the size.

Could this be a problem related to the way I load it?
I use BLOAD"cas:",r
Maybe BLOAD uses BASIC-available memory only and this causes the game not to load.

zx70, 2017-08-31:
The msx basic+bios takes 32k of the linear 64k address space, I'm quite sure an msxdos version could benefit of more available space .   Also try to optimize, eg the -O3 option and the -lndos foo file library could help in this.   Sdcc is much more powerful and createz much faster and smaller code, if you manage how to run it with msx (can be tricky).

Fabrizio-Caruso, 2017-09-13:
This problem is no longer reproducible. Please close this issue.

aralbrec, 2017-09-13:
Was it program size that was causing the issue?

Fabrizio-Caruso, 2017-09-13:
The problem existed only for the full version of my game (>16k) and not for the light version (<16k).

With the latest version of Z88DK and my code I can no longer reproduce the problem.

zx70, 2017-09-13:
"The problem existed only for the full version of my game (>16k) and not for the light version (<16k).
With the latest version of Z88DK and my code I can no longer reproduce the problem."
Not just related to the code size, but seems to be solved.

closed 2017-09-13 by zx70