z88dk/z88dk #324: nightly build git update fail

aralbrec, 2017-08-21:
> Please commit your changes or stash them before you merge.

Probably won't go away without interceding.

suborb, 2017-08-21:
Yeah. It's a line feed issue. Will clear the workspace and that should sort it out.

suborb, 2017-08-21:
Looks like these are the remaining files with line feed issues:

```
#	modified:   examples/ticalc/spritest.c
#	modified:   lib/lambda_altint.def
#	modified:   lib/mtx_crt0.asm
#	modified:   lib/oz_customisr.def
#	modified:   lib/shellapi.def
#	modified:   lib/x1_crt0.asm
#	modified:   lib/zx81_altint.def
#	modified:   lib/zxr_packages.def
#	modified:   libsrc/fcntl/osca/flos/close.c
#	modified:   libsrc/fcntl/osca/flos/mkdir.c
#	modified:   libsrc/fcntl/osca/flos/open.c
#	modified:   libsrc/fcntl/osca/flos/rename.c
#	modified:   libsrc/fcntl/osca/flosmulti/close.c
#	modified:   libsrc/fcntl/osca/flosmulti/mkdir.c
#	modified:   libsrc/fcntl/osca/flosmulti/open.c
#	modified:   libsrc/fcntl/osca/flosmulti/rename.c
#	modified:   libsrc/fcntl/spectrum/microdrive/close.c
#	modified:   libsrc/fcntl/spectrum/microdrive/open.c
#	modified:   libsrc/fcntl/spectrum/microdrive/rename.c
#	modified:   libsrc/oz/ozgfx/ozputs_direct.asm
#	modified:   libsrc/oz/ozinput/ozeditline.c
#	modified:   libsrc/oz/oztime/oztime.c
#	modified:   libsrc/rs232/cpc/booster/rs232_close.asm
#	modified:   libsrc/rs232/cpc/booster/rs232_get.asm
#	modified:   libsrc/rs232/cpc/booster/rs232_init.asm
#	modified:   libsrc/rs232/cpc/booster/rs232_put.asm
#	modified:   libsrc/rs232/osca/rs232_params.asm
```

I'll sort them out this evening.

suborb, 2017-08-22:
I pushed through another one this morning and it worked

closed 2017-08-22 by suborb

suborb, 2017-08-23:
I fixed the old build machine, I've now fixed the current one. So hopefully that'll be it until line feeds break again.

pauloscustodio, 2017-08-23:
What was the line-feed issue? Anything we can avoid on our side?

suborb, 2017-08-23:
It was some files in CVS that were checked in with windows line endings. I was getting a lot of warnings so I normalised them.

The build machine had also normalised them as part of its checkout, so there was a conflict - even though they were the same.

As long as I've zapped them all we should be okay since .gitattributes defines the endings. However, I'm not sure whether the GitHub web submit respects .gitattributes though - I think a couple of files came from there.