z88dk/z88dk #329: ZX81 conio. clib=ansi won't compile with conio...

Fabrizio-Caruso, 2017-08-24:
Hi!
I have managed to compile my game for ZX81 with
zcc +zx81  -lndos -vn -D__ZX81__  -create-app -o
The code uses conio.h which produces wrong results: it prints the coordinates and other codes.

Could someone explain to me what ansi vt means for Z88DK?
Does it mean that conio is supposed to work?
Das clib=ansi means ansi vt = conio support.

How should I compile to get conio support?

if I add clib=ansi the code won't even compile:
Error at file 'stdio/ansi/f_ansi.asm' line 630: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 633: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 706: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 710: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 779: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi_lf.asm' line 28: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi_lf.asm' line 31: symbol 'text_rows' not defined
Error at file 'graphics/zx81/hr/_clg_hr.asm' line 32: symbol 'hr_rows' not defined
Error at file 'stdio/ansi/zx81/f_ansi_scrollup.asm' line 38: symbol 'text_rows' not defined
9 errors occurred during assembly
Errors in source file c:/z88dk/\lib\zx81_crt0.asm:
Error at file 'stdio/ansi/f_ansi.asm' line 630: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 633: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 706: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 710: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi.asm' line 779: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi_lf.asm' line 28: symbol 'text_rows' not defined
Error at file 'stdio/ansi/f_ansi_lf.asm' line 31: symbol 'text_rows' not defined
Error at file 'graphics/zx81/hr/_clg_hr.asm' line 32: symbol 'hr_rows' not defined
Error at file 'stdio/ansi/zx81/f_ansi_scrollup.asm' line 38: symbol 'text_rows' not defined

zx70, 2017-08-25:
The conio mode requires to be in ansi vt mode.  On the zx81 the ansi vt is available only with a graphics mod or expansion.   You can still use the 32x24 standard text and use zx_setcursorpos(b-1,a-1) to position your cursor, it should work with zx81, spectrum and ts2068 in non-vt mode.

zx70, 2017-08-31:
The conio IS supported, but only when hr graphics is ednabled

closed 2017-08-31 by zx70