z88dk/z88dk #330: sccz80:  variable arguments is bugged

aralbrec, 2017-08-24:
zzz.c
```
#include <stdio.h>
#include <string.h>
#include <stdarg.h>

int error(char *fmt, ...)
{
   static unsigned char ebuf[65];
   
   va_list v;
   va_start(v, fmt);
   
   vsnprintf(ebuf, sizeof(ebuf), fmt, v);
   // ebuf[strlen(ebuf)-1] += 0x80;
   
   return (int)ebuf;
}

void main(void)
{
   unsigned char *p;
   
   p = (void*)error("Hello");
   printf("%s\n", p);
   
   p = (void*)error("Hello %s", "There");
   printf("%s\n", p);
   
   p = (void*)error("Hello %s %s", "There", "Man");
   printf("%s\n", p);
}
```

This is an excerpt from an esxdos dot command.

With sdcc & newlib, it works fine:
`zcc +zx -vn -startup=30 -clib=sdcc_iy -SO3 --max-allocs-per-node200000 zzz.c -o zzz -create-app`

With sccz80 & newlib, it doesn't:
`zcc +zx -vn -startup=30 -clib=new zzz.c -o zzz -create-app`

With sccz80 & classic, also not ok:
`zcc +zx -vn -lndos zzz.c -o zzz -create-app`

With sdcc & classic, crashes:
`zcc +zx -vn -compiler=sdcc -SO3 --max-allocs-per-node200000 --reserve-regs-iy -lndos zzz.c -o zzz -create-app`

suborb, 2017-08-24:
It's the silly left to right call again, sccz80 thinks that fat is at sp+4 on entry to the function.

 With sccz80 you have to bodge it a little, here's the old C printf to show you how:

```
int printf(char *fmt,...)
{
        int  *ct;
        ct= (getarg()*2)+&fmt-4;

        return (vfprintf(stdout,(unsigned char *)(*ct),ct-1));
}
```

I think it might be best for sccz80 to generate variadic functions with r2l semantics.

aralbrec, 2017-08-24:
It's the va_start macro that is wrong then:
`#define va_start(ap,last)       ap=(getarg()*2)+&last-2`

I'm just wondering if there is a reason the macro is kept like that?

suborb, 2017-08-24:
I think I must've given up trying to get the macro to work in a sensible way. (Not a great reason I do admit).

aralbrec, 2017-08-25:
I came up with something that works:

```
#include <stdio.h>
#include <string.h>
#include <stdarg.h>

int error(char *fmt, ...)
{
   static unsigned char ebuf[65];
   
   va_list v;
   va_start(v, fmt);
	
   // vsnprintf(ebuf, sizeof(ebuf), fmt, v);
   vsnprintf(ebuf, sizeof(ebuf), va_ptr(v,char *), v);
   // ebuf[strlen(ebuf)-1] += 0x80;
	
   return (int)ebuf;
}

void main(void)
{
   unsigned char *p;
   
   p = (void*)error("Hello");
   printf("%s\n", p);
   
   p = (void*)error("Hello %s", "There");
   printf("%s\n", p);
   
   p = (void*)error("Hello %lu %x", 123456UL, 0xabcd);
   printf("%s\n", p);
}
```

The only catch with l2r is that "fmt" will not point at fmt in the argument list.  Instead, you have to use a non-standard macro va_ptr to get its address.  va_ptr returns the argument currently being pointed at, as opposed to va_arg which will decrement the pointer by the sizeof of the type and then return the argument.

I'll commit that now for both classic and new.  For classic and sdcc, you have to define __Z88DK_R2L_CALLING_CONVENTION on the compile line to get sdcc's stdarg.

commit b85dbe20 (2017-08-25): Fix #330: bring sccz80 closer to standard variable argument handling
closed 2017-08-25 by aralbrec