z88dk/z88dk #333: z80asm: map file (symbol info) needs some more information on memory banks and symbol types

aralbrec, 2017-08-27:
The z88dk disassembler that was added today is able to disassemble binaries using symbol information from the map file.  From a thread in the pull requests:

> If I put the spaces in:
> 
> ```
> z88dk-dis -o 32768 -s 32768 -mz80-zxn -x sudoku.map sudoku_CODE.bin > out.txt
> Cannot load file '32768'
> Cannot load file '32768'
> ```
> 
> but the label thing is fixed.
> 
> It's nice to see the symbols in there:
> 
> ```
> asm_zx_scroll_wc_up_noexx:	exx     		;d9
> l_824b:             	push    bc		;c5
> l_824c:             	push    de		;d5
> l_824d:             	push    hl		;e5
> l_824e:             	exx     		;d9
> l_824f:             	call    asm_zx_scroll_wc_up		;cd 57 82
> l_8252:             	pop     hl		;e1
> l_8253:             	pop     de		;d1
> l_8254:             	pop     bc		;c1
> l_8255:             	exx     		;d9
> l_8256:             	ret     		;c9
> asm_zx_scroll_wc_up:	inc     d		;14
> l_8258:             	dec     d		;15
> l_8259:             	jp      nz,asm_zx_cls_wc		;c2 aa 81
> asm0_zx_scroll_wc_up:	inc     e		;1c
> l_825d:             	dec     e		;1d
> l_825e:             	ret     z		;c8
> l_825f:             	ld      a,(iy+$03)		;fd 7e 03
> l_8262:             	dec     a		;3d
> l_8263:             	sub     a,e		;93
> l_8264:             	jp      c,asm_zx_cls_wc		;da aa 81
> l_8267:             	inc     a		;3c
> l_8268:             	ld      c,a		;4f
> l_8269:             	push    hl		;e5
> l_826a:             	push    de		;d5
> l_826b:             	ld      h,(iy+$02)		;fd 66 02
> l_826e:             	ld      l,(iy+$00)		;fd 6e 00
> l_8271:             	call    asm_zx_cxy2saddr		;cd 19 82
> l_8274:             	ex      de,hl		;eb
> l_8275:             	ld      a,(iy+$02)		;fd 7e 02
> l_8278:             	add     a,l		;85
> l_8279:             	ld      h,a		;67
> l_827a:             	ld      l,(iy+$00)		;fd 6e 00
> l_827d:             	call    asm_zx_cxy2saddr		;cd 19 82
> l_8280:             	push    hl		;e5
> l_8281:             	push    de		;d5
> l_8282:             	exx     		;d9
> l_8283:             	pop     hl		;e1
> l_8284:             	pop     de		;d1
> l_8285:             	call    asm_zx_saddr2aaddr		;cd 40 82
> l_8288:             	ex      de,hl		;eb
> l_8289:             	call    asm_zx_saddr2aaddr		;cd 40 82
> l_828c:             	ld      b,$00		;06 00
> l_828e:             	exx     		;d9
> copy_up_loop_0:     	exx     		;d9
> ```
> 
> It would be nice if no end address is specified that is pursues all branches rather than stopping after a fixed line count.  yazd does something like that.
> 
> > I think the map file probably needs to distinguish between labels and defc to get the symbolic compilation to truly work.
> 
> I also use defc to define address aliases.  I don't know if z80asm would be able to distinguish between a link-time defc (an address) and an assemble time constant in the map file.  We also have to look at putting bank information into the map file (and have it sorted by bank) and it would be nice if z88dk-dis could handle multiple banks eventually.
>

aralbrec, 2017-08-27:
For different memory banks, I put a separation into `appmake +glue` that will look at section names and if they contain "BANK_NN" then the section is assumed to belong to a separate 64k bank.  Maybe a revisit and rethink can allow the tools to deal in an extended memory space.

aralbrec, 2017-08-28:
Maybe adding section name and symbol type (link time or asm time) to the map file is sufficient.  And then sort it according to memory bank so we don't have symbols from different banks mixed together.  If the section name contains BANK_NN with NN in hex, then it belongs to a 64k bank BANK_NN.  Otherwise it's in the main bank of 64k.

pauloscustodio, 2017-08-28:
For the short-time solution, adding a section name to each symbol in the map file is easy. Sorting according to section should be easy but I have to check. Adding the symbol type is more difficult, as z80asm treats all expressions to compute at link time as equal.

For a longer term solution, I would like to settle on a design that solves the shortcomings of the existing sections in handling non-linear address space, and at the same time find an alternative to the ORG -1 / ORG defined-or-not-defined kludge.

Proposal for discussion:
- Sections are organized within section groups. 
- One section group has a name, may have a start address defined at link time, or may follow another section group, may be squashed in the output together with the previous section, and produces one binary file
- The default section group with empty name is the owner of all sections if no group is referenced.

Example syntax:
```
group text at 0x100
group data squash text
group bss after data
section code_1 in text
...
```
The org address of group text can be overridden by a link-time option.
The code above generates two output files, file_text.bin (containing the concatenation of groups text and data) and file_bss.bin.

Any comments are wellcome.

pauloscustodio, 2017-08-28:
See also #19

pauloscustodio, 2017-08-28:
Maybe we need to extend the address range to be able to flow around holes in the address space, e.g.
```
group text at 0x100 to 0x200, 0x400 to 0x8000
```
Where the space from 0x200 to 0x3FF is not usable, and code will flow around the hole. The linker will produce two binary files, one for each region of memory, and will raise an error if the address space moves beyond 0x8000.

pauloscustodio, 2017-08-28:
See also #42