z88dk/z88dk #337: Need an Easy Way to Install Third Party Libraries

aralbrec, 2017-08-29:
I've set aside directories within the newlib to hold 3rd party libraries and include files so that 3rd party libraries can be linked against by adding `-ltarget/libname` to the compile line and include files can be found in the system directories with `#include <arch/target/libname.h>`

The same could be done for classic.

I'd like to add an `install` tool that will copy the 3rd party library and include files to the write places within the z88dk install.

aralbrec, 2017-09-02:
It's not quite working yet but it is there.

z88dk-install
* gives help

z88dk-install +zx
* lists installed 3rd party libraries for zx target

z88dk-install +zx mylib
* installs mylib library for zx target
* classic header:  mylib.h copied to include/lib/zx
* classic library: mylib.lib copied to lib/clibs/lib/zx
* newlib header:  mylib_newlib.h copied to include/_DEVELOPMENT/proto/lib/zx
* newlib header:  "Winmake" or "make" is run from include/_DEVELOPMENT to build headers
* newlib library: mylib_sccz80.lib is copied to libsrc/_DEVELOPMENT/lib/sccz80/lib/zx
* newlib library: mylib_sdcc_ix.lib is copied to libsrc/_DEVELOPMENT/lib/sdcc_ix/lib/zx
* newlib library: mylib_sdcc_iy.lib is copied to libsrc/_DEVELOPMENT/lib/sdcc_iy/lib/zx

z88dk-install +zx --remove mylib
* removes above files

I think I may change away from a flat src directory where the mylib name is modified for each lib type to a directory tree containing these things.

A makefile equivalent to Winmake in include/_DEVELOPMENT still have to be written - I don't know if someone wants to have a like.  This makefile should be added to the library build process so that the newlib headers are built along with the libraries.  Right now I am manually maintaining them and committing as changes occur.

closed 2017-09-03 by aralbrec

aralbrec, 2017-09-03:
The last ingredient is a makefile to perform the same function as the windows batch file [Winmake.bat](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/Winmake.bat) in directory include/_DEVELOPMENT .

Winmake.bat
```
@echo off

setlocal
setlocal ENABLEDELAYEDEXPANSION

echo.
echo WINMAKE: GENERATING HEADER FILES
echo.

for /R proto %%h in (*.h) do (

   echo %%h
   
   set src=%%h
   set dst=!src:proto=sccz80!
   m4 -Dm4_SCCZ80 "%%h" > "!dst!"
   
   set src=%%h
   set dst=!src:proto=sdcc!
   m4 -Dm4_SDCC "%%h" > "!dst!"
   
   set src=%%h
   set dst=!src:proto=clang!
   m4 -Dm4_CLANG "%%h" > "!dst!"

   echo.
)
```

What this is doing is iterating over every header file in the proto tree and using m4 to make a new one for each of the sccz80, sdcc, and clang trees.  The proto tree acts like the prototype and the specialized headers for each compiler is generated from that.

This is probably better suited as a shell script but I tried to wedge it in as a makefile first and I'm not quite sure how to go about it.

I'm trying this now but it's not working:

```
# Recursive wildcard function
# https://stackoverflow.com/questions/3774568/makefile-issue-smart-way-to-scan-directory-tree-for-c-files

rwildcard=$(wildcard $1$2) $(foreach d,$(wildcard $1*),$(call rwildcard,$d/,$2))

# All headers in proto tree

HEADERS  := $(call rwildcard,proto/,*.h)

SCCZ80_HEADERS := $(addprefix sccz80/,$(HEADERS))
SDCC_HEADERS   := $(addprefix sdcc/,$(HEADERS))
CLANG_HEADERS  := $(addprefix clang/,$(HEADERS))

default: sccz80 sdcc clang

sccz80: $(SCCZ80_HEADERS)
	m4 -Dm4_SCCZ80 $(subst sccz80,proto,$@) > $@

sdcc: $(SDCC_HEADERS)
	m4 -Dm4_SCCZ80 $(subst sdcc,proto,$@) > $@

sdcc: $(CLANG_HEADERS)
	m4 -Dm4_SCCZ80 $(subst clang,proto,$@) > $@
```

Any ideas?

feilipu, 2017-09-03:
I will try this with the ChaN FatFS C code shortly.  I still need to get the diskio C interface to 8255 working. Tomorrow, probably.

Could a header file to asm code library, with C interface work this way too? I'm specifically thinking about the best way to link in the APU code I committed recently.

aralbrec, 2017-09-03:
Third party libraries are organized as follows with any subtree optional:

```
libname/
   targetname/
      include/
         classic/
            libname.h
         newlib/
            libname.h
      lib/
         classic/
            libname.lib
         newlib/
            sccz80/
               libname.lib
            sdcc_ix/
               libname.lib
            sdcc_iy/
               libname.lib
```

In this way, a single library can be built for multiple targets and the library types are separated by directory rather than by name in a flat list.

`z88dk-install`
list help

`z88dk-install +zx`
list 3rd party libraries already installed

`z88dk-install +zx -r -f libname1 libname2 ...`
remove the libname1 libname2 ... libraries from the zx target, -f for no nagging about deleting files

`z88dk-install +zx libname1 libname2 ...`
add the libname1 libname2 ... libraries to the zx target, -f for no nagging about overwriting an existing install

aralbrec, 2017-09-03:
>    I will try this with the ChaN FatFS C code shortly. I still need to get the diskio C interface to 8255 working. Tomorrow, probably.

The makefile needs to be done before it will work in linux. Windows is currently working.

>    Could a header file to asm code library, with C interface work this way too? I'm specifically thinking about the best way to link in the APU code I committed recently.

The asm code can be in the same library as the c code but I'm not sure yet what to do about an asm include. With the asm code in the library you just need to EXTERN foo to bring foo into scope and that will work for defc as well as labels.

suborb, 2017-09-03:
Sorry @aralbrec I've been Rabbiting today and didn't spot your makefile foo request.

I think as a rule you need:

    sccz80/%: proto/%
       m4 -Dm4_SCCZ80 $^ > $@

Likewise for clang and sdcc

Then the all becomes:

    all: $(SCCZ80_HEADERS) $(SDCC_HEADERS) $(CLANG_HEADERS)

But I've not tested it.

suborb, 2017-09-03:
The following works. You may want to stick an @ before the m4 commands to shut it up a bit.

```
# Recursive wildcard function
# https://stackoverflow.com/questions/3774568/makefile-issue-smart-way-to-scan-directory-tree-for-c-files

rwildcard=$(wildcard $1$2) $(foreach d,$(wildcard $1*),$(call rwildcard,$d/,$2))

# All headers in proto tree

HEADERS  := $(call rwildcard,proto/,*.h)

SCCZ80_HEADERS := $(subst proto,sccz80/,$(HEADERS))
SDCC_HEADERS   := $(subst proto,sdcc,$(HEADERS))
CLANG_HEADERS  := $(subst proto,clang,$(HEADERS))

default: dirs $(SCCZ80_HEADERS) $(SDCC_HEADERS) $(CLANG_HEADERS)

sccz80/%: proto/%
        m4 -Dm4_SCCZ80 $^ > $@

sdcc/%: proto/%
        m4 -Dm4_SDCC $^ > $@

clang/%: proto/%
        m4 -Dm4_CLANG $^ > $@

dirs:
        @mkdir -p $(dir $(SCCZ80_HEADERS))
        @mkdir -p $(dir $(SDCC_HEADERS))
        @mkdir -p $(dir $(CLANG_HEADERS))

.PHONY: dirs
```

feilipu, 2017-09-14:
I have 3 candidates for C libraries to add, but they all have warts that may make them somewhat undesirable to be included in the main code base.
- ChaN's FatFS - needs a 8255 diskio driver (just yaz180 and rc2014 have this currently), and it is one very large file that takes an age to compile. [ff.h](https://github.com/z88dk/z88dk/files/1301560/ff.h.txt)
- AVR (now z80) time functions - general for z80/z180 with an asm system_tick - neatly organised file per function but ``strftime()` is "a memory hungry monster" and takes an age to compile. [time.h](https://github.com/z88dk/z88dk/files/1301557/time.h.txt)
- FreeRTOS - general for z80/z180, with an asm system_tick - work in progress. 
[FreeRTOS.h](https://github.com/z88dk/z88dk/files/1301562/FreeRTOS.h.txt)

I'm not sure that the general z88dk user will be interested in having these libraries slowing down the compile every time, and generally making things more complicated,... yet, I'd like to have some kind of integration with z88dk to make it easy for people to use them if desired.

Could I put them in a separate repository so that they become optional downloads, and somehow link them in to the z88dk git solution?

Any thoughts on how to do this?
Is the above mechanism the preferred way, given how much they might load up z88dk?

aralbrec, 2017-09-17:
Maybe the answer is to create a new repository under https://github.com/z88dk called "z88dk-libraries" or "libraries" and have these sorts of things available directly or as git submodules.  Then if they are moved to the main repo, they can be deleted from the there.

There should be instructions for their use and instructions for building and installation.

feilipu, 2017-09-17:
With regard to this third party library process, I have created an example library, and pushed it up to a repository. For want of a better name, I called it [z88dk-time](https://github.com/feilipu/z88dk-time).

This is a fairly extensive time functions library supposedly close to fulfilling the ISO/IEC 9899 (C90) requirements.

I've been using it for some years in the [avr-libc](http://www.nongnu.org/avr-libc/user-manual/group__avr__time.html) for AVR ATmega machines, and it seems to work pretty well.

The only change for z80 is the use of the intrinsic division and modulus z88dk newlib library functions, rather than special functions provided by the avr-libc.

The compiled libraries are organised so that the `z88dk-lib` tool works.
The sccz80 compiler has trouble with some of the function calls.
I've not fixed this, so it is sdcc only at this stage.

I consider this library fairly general, as it is usable by any Z80 machine that can generate a 1 second [`system_tick`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/driver/system_time/z180/system_tick.asm) to increment the `time_t _system_time` variable. For Z180 machines, the PRT makes an ideal solution.

Happy for this to kick off a thought process. No requirement for [this repository](https://github.com/feilipu/z88dk-time) to remain in use, post resolution.

commit 7724bbe4 (2017-09-03): Fix #337: z88dk-install for third party libraries completed.