z88dk/z88dk #338: nightly build:  windows build of zxn library is incorrect

aralbrec, 2017-08-29:
I'm not sure if it's also the case for the mac osx build.

loadit.asm
```
INCLUDE "config_zxn_private.inc"

SECTION code_user

PUBLIC _main

_main:

   ; instead of calling esxdos directly we could also call through the z80 library api at
   ; https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/arch/zx/esxdos/z80
   ; the advantage would be proper handling of errno and maybe easier to read code

   ld a,'*'   ;; use current drive
   ld b,__ESXDOS_MODE_OPEN_EXIST | __ESXDOS_MODE_READ
   ld ix,name   ;; when esxdos is paged in param is in hl, when paged out param is in ix
   ld hl,name   ;; should be ix here because we call from ram but emulator issues require both
   
   rst  __ESXDOS_SYSCALL
   defb __ESXDOS_SYS_F_OPEN

   ret c

   push af
   
   ld ix,0x4000   ;; same as above
   ld hl,0x4000
   ld bc,6912
   
   rst  __ESXDOS_SYSCALL
   defb __ESXDOS_SYS_F_READ
   
   pop af
   
   rst  __ESXDOS_SYSCALL
   defb __ESXDOS_SYS_F_CLOSE
      
   jr ASMPC

name:

   defm "kv.bin",0
```

zcc +zxn -vn -m -startup=31 -clib=sdcc_iy loadit.asm -o loadit -subtype=sna -create-app

Results in:

```
Assertion failed!

Program: c:\Projects\z80\z88dk\z88dk\bin\z80asm.exe
File: modlink.c, Line 381

Expression: expr->target_name == NULL

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
Errors in source file c:\Projects\z80\z88dk\z88dk\lib\config\..\..\\libsrc\_DEVE
LOPMENT\target\zxn\zxn_crt.asm:
```

If the zxn library is manually re-built by going to z88dk/libsrc/_DEVELOPMENT and running `Winmake zxn` then the problem goes away.  Also, if the above program is compiled using `zx` as target instead of `zxn` it again works.  So there is a problem with the building of the zxn library specifically.

I did a fresh install on linux and did the z88dk build using the makefiles and it turned out fine, building the zxn library properly.

suborb, 2017-08-29:
All of th nightly libs are built on Linux and then copied into the appropriate kits.

aralbrec, 2017-08-29:
The problem is the nightly is not replacing these files when the library is built:

z88dk/libsrc/_DEVELOPMENT/target/zxn/config_zxn.h
z88dk/libsrc/_DEVELOPMENT/target/zxn/config_zxn_private.inc
z88dk/libsrc/_DEVELOPMENT/target/zxn/config_zxn_public.inc

These are generated by m4 in the Makefile:
https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/Makefile#L29

Instead the nightly is grabbing the currently committed versions of those files which are out of date and contain an undefined symbol.

The nightly build of z80asm is crashing due to this circular define `defc __IO_RRPM_MASK = __IO_RRPM_MASK` that is present because the defines are not updated.

I'm not sure why the vs2015-built windows binary behaves differently.  The vs2015 built version does not crash but instead inserts this "define" in the output as part of the unassigned sections binary.

So the nightly is probably not overwriting those files for all the targets and it should when the newlib is built.

aralbrec, 2017-08-29:
Are files marked read-only so that when libraries and binaries are built the source tree will be unaffected?  I haven't marked this config files as ignore for git because I thought they would be useful for referencing via web.

aralbrec, 2017-08-30:
the rsync did the trick

closed 2017-08-30 by aralbrec