z88dk/z88dk #339: Question: What is the use of afswap in the library?

pauloscustodio, 2017-08-29:
What is the use of z88dk/libsrc/z80_crt0s/z80_emu/afswap.asm? 
Both the Zilog and Rabbit cpus implement ex af,af' with the same opcode 0x08.

pauloscustodio, 2017-08-29:
What is the use of z88dk/libsrc/z80_crt0s/z80_emu/exxswap.asm?
Both the Zilog and Rabbit cpus implement exx with the same opcode 0xD9.

suborb, 2017-08-29:
I think it might be some 8080 compatibility code.

pauloscustodio, 2017-08-29:
Maybe I should add --cpu=8080 after the Rabit is finished. Useful?

suborb, 2017-08-29:
For use of z80asm as a general assembler, yes. However I think we've gone past supporting the 8080 directly with longs and doubles. which use both the alternate sets and an index register so z88dk itself wouldn't use that mode.

Which makes me wonder why @zx70 added those routines - it might have been for platforms that reserve the alternate set.

It's probably more useful to add the R800 instructions (mulub,muluw,tst?) since that's used by some MSX models, and the ez80 (used by one of the newer TI calculators).

zx70, 2017-08-30:
They should be zx81related but I thought a similar case could be happening also on not fully compatible z80 clones.
As an example, A' can't be touched because is holds the current character row to be displayed, and the rom code is nmi driven.  An intermediate edition of the maths and/or graphics function used lots of similar tricks, I'm not sure on the current library situation.

pauloscustodio, 2017-08-30:
I can easily add options to use these library functions instead of ex af,af' and exx.
Is this useful?

zx70, 2017-08-30:
Intriguing but I'm not sure.  I spotted some use case in genmath/*noaf.asm, it can help deciding

zx70, 2017-08-31:
Oook.. I think that, on a ZX81, "afswap" can be handy, it could be added to the "-IXIY" parameter to make more libraries usable.  On the maths library portion (and probably on the gfx too) we'll be able to remove the reworked file copies and make things cleaner.

from the zx81 forum:
A, BC, DE, HL, BC', DE', HL' (AF,BC,DE,HL are stored during interrupts / picture generation and the double set is not used at all liek BC', DE', HL'.
The register IY could be used as well in FAST mode.

..I made IY usable by reworking the video interrupts, so..   just IX and AF to be preserved, if the statement above is correct  :)

closed 2019-02-11 by suborb