z88dk/z88dk #340: Question: is there a reference emuation implementation to DAA?

pauloscustodio, 2017-08-29:
I would like to add DAA to the Rabbit compiles by calling a library routine.

Is there a reference implementation?

aralbrec, 2017-08-29:
I think your best bet is to look at an accurate open source emulator.

Fuse has this:
https://sourceforge.net/p/fuse-emulator/fuse/ci/master/tree/z80/z80.pl#l541

pauloscustodio, 2017-09-13:
Implemented DAA emulation for Rabbit in commit 905dd799d8aca8e1cc31a712db3fd3c5f8f24106 

DAA can now be used in any target, and resolves to a call __z80asm_daa on a Rabbit.

The implementation is hand-assembled based on the Fuse implementation.

The result of the H flag is not corretly emulated (it fails two of the test cases).
Ticks, which was used to test the implementation also fails the correct emulation of DDA result H flag.

The tests that fail are commented in test.pl.

closed 2017-09-13 by pauloscustodio