z88dk/z88dk #341: Produce a debugger-friendly filename/bank/memory address file

Tenebrous, 2017-08-31:
The .map and .lis files aren't currently sufficient for a debugger to point the user back at the relevant source file.

I believe a new file is required to allow debuggers to correctly locate the source.

Suggested format/example for discussion:
> File "(filename1)"
> Bank (bank name)
> (memory address) (line number) (any labels if present)
> (memory address) (line number) (any labels if present)
> ...

You could also consider including both the .c and .asm source:
> File "(filename1.c)" "(filename1.asm)"
> Bank (bank name)
> (memory address) (line number in .c) (line number in .asm) (any labels if present)
> (memory address) (line number in .c) (line number in .asm) (any labels if present)
> ...

Examples:
> File "src\filename.c"
> Bank CODE
> 0000 1 _main
> 0003 2
> 0008 5 main_loop
>
> File "src\otherfile.asm"
> Bank BANK_04
> 0006 17 loop
> 0006 18             
> 0006 19 scribble1_init
> 0006 20 
> 0008 21 
> 000B 22 
> 000B 23 
> 000D 24 scribble1_loop
> 000E 26 
> 0011 27 

> File "src\filename.c" "src\filename.asm"
> Bank CODE
> 0000 1 5 _main
> 0003 2 10
> 0008 5 12 main_loop

Tenebrous, 2017-09-01:
Via some awk I've injected line-number labels into the .asm, and extract that info from the map, and I can now automatically produce the following:
[image]

pauloscustodio, 2017-09-01:
z80asm has a "LINE nnn" directive to allow the C compiler to indicate the C source line number of each statement, defining labels "__C_LINE_nnn" at the address of each statement.

pauloscustodio, 2017-09-09:
Commit c9d173a3a428cf39c15006e8ee290ba46b3a7ffa  adds section, file name and line number information to .sym and .map files

pauloscustodio, 2017-09-11:
Please have a look at the new map file format. It does not have all the information requested, but may already help for the objective.

Having the source reference of the C code and of each assembly instruction needs more information in the object file format that does not exist currently.

Tenebrous, 2017-09-11:
@pauloscustodio Excellent, that works really well.  I have a pre-processor to add per-line labels to .asm, and a post-processor to read the .map. So now with both __CLINE__ and __ASMLINE__ labels being available in the .map, in conjunction with any use labels, I think I have everything to do source-level debugging.

Thanks!

pauloscustodio, 2017-09-11:
I can easily add the __ASMLINE__ labels in z80asm - this removes the need of the pre-processor

Tenebrous, 2017-09-11:
Agreed, that would be fantastic.  However please don't add them for the temp\xyz.asm files created by zcc when it joins other .asms together :)

Tenebrous, 2017-09-17:
@pauloscustodio  Did you have any further thoughts on the line numbering in z80asm?

pauloscustodio, 2017-09-17:
Not yet. What I had in mind was an option to pass to z80asm to force the creation of per-line labels. Then this option would be used only for asm source code and not for code generated from C.

suborb, 2017-09-20:
Have look at codegen.c and search for LINE. there's 2 ways of doing it there: emitting the line directive or a label method.

suborb, 2017-09-22:
I changed LINE to C_LINE and added a -gcline option to enable this being printed out.

Whilst I was there I removed the quotes from the filename in -glabels.

I'll need to update the ticks debugger to read the extra columns.

pauloscustodio, 2017-09-22:
Thanks @suborb, I had no time to look into it.

Tenebrous, 2017-09-22:
Excellent guys, I'll grab that and shuffle some code around in my debugger :)

suborb, 2017-09-22:
@pauloscustodio I'm done here I think so it's good to merge if you're happy

pauloscustodio, 2017-09-23:
When running ```sccz80 -mz80 -gcline test.c```, the output test.asm contains all the needed C_LINE statements. But assembly of this test.asm outputs missing library functions:

```
Error at file 'test.c' line 3: symbol 'l_gint' not defined
```

zcc, on the other hand, does not have the -gcline option, and using ```-Cc-gcline``` does not seem to make effect, the map file still refers to the temporary files, ``` zcc +z80 -m --list -Cc-gcline -Ca--debug -no-cleanup test.c test1.asm -otest.bin```, where test.c contains main(), outputs in test.map:
```
_main                           = $0187 ; addr, public, , test_c, code_compiler, C:\Users\paulo\AppData\Local\Temp\zccB3052.asm:248
```

I was expecting 
```
_main                           = $0187 ; addr, public, , test_c, code_compiler, test.c:6
```

What am I missing?

pauloscustodio, 2017-09-23:
I will commit the test script src/z80asm/t/test_bug_341.t, in case you want to reproduce the error.

aralbrec, 2017-09-23:
`zcc +z80 -m --list -Cc-gcline -Ca--debug -no-cleanup test.c test1.asm -otest.bin`

Uses zsdcc as c compile by default.  Try instead:

`zcc +z80 -m --list -clib=new -Cc-gcline -Ca--debug -no-cleanup test.c test1.asm -otest.bin`

to get sccz80 to compile.

pauloscustodio, 2017-09-23:
Help needed again.

zcc concatenates a set of pre-processed files and sends the output to z80asm. This results in debug info on the temporary file names, e.g.
```__ASM_LINE_4 = $0000 ; addr, local, , test1, , C:\Users\paulo\AppData\Local\Temp\zccCD253.asm:4```

I need zcc to output ```LINE 1, "test1.asm"``` before ```# type "test1.asm" >> "C:\Users\paulo\AppData\Local\Temp\zccCD253.asm"```

suborb, 2017-09-23:
@pauloscustodio try the attached.

[line.diff.txt](https://github.com/z88dk/z88dk/files/1326948/line.diff.txt)

pauloscustodio, 2017-09-24:
Thanks, @suborb

Tenebrous, 2017-09-24:
I've updated my debugger to use the latest .map format with these additions and it's working really well - thanks!

commit 40648879 (2017-09-24): Fix #341 Produce a debugger-friendly map file

The map file now contains references to the source lines of labels, both of C and ASM code.

For example the line
zcc +z80 -m -clib=new -Cc-gcline -Ca--debug test.c test1.asm -otest.bin

produces in the map file: (see src/z80asm/t/test_bug341.t)
__C_LINE_2                      = $0000 ; addr, local, , test, , test.c:2
__C_LINE_3                      = $0000 ; addr, local, , test, , test.c:3
__C_LINE_4                      = $017D ; addr, local, , test, code_compiler, test.c:4
__C_LINE_5                      = $018D ; addr, local, , test, code_comp [... cut]
commit a3d2a2c1 (2017-09-24): Merge pull request #406 from z88dk/feature/z80asm_debug_info

Fix #341 Produce a debugger-friendly map file
closed 2017-09-24 by pauloscustodio