z88dk/z88dk #343: Rabbit emulation of cpi and cpir is broken

suborb, 2017-09-03:
I'm enabling running the tests on +test -clib=rabbit and the following tests are failing:

strncmp
strrstr
strrev

All of these use cpi/cpir which points the finger there.

suborb, 2017-09-03:
I've tracked this down to the z80asm emulation library being compiled without the rabbit flags so ex (sp),hl resolves to 0xe3 rather than 0xed54. 0xe3 on the Rabbit is ex de',hl

pauloscustodio, 2017-09-03:
Thanks for reporting and analysing.  I'll fix it.

pauloscustodio, 2017-09-03:
I'm getting an error when trying to compile a test program:
```
>zcc +test -v clib=rabbit test.c
PROCESSING clib=rabbit.asm.m4
m4 -I "c:/Users/paulo/git/z88dk/lib/config\..\..\/lib" -I "c:/Users/paulo/git/z88dk/lib/config\..\..\/src/m4"   < "clib=rabbit.asm.m4" > "C:\Users\paulo\AppData\Local\Temp\zcc2F432"
The system cannot find the file specified.
```

What am I missing?

suborb, 2017-09-03:
-clib=rabbit

pauloscustodio, 2017-09-03:
Sorry, that was obvious... And now?
```
>zcc +test -clib=rabbit test.c
Cannot find definition for -clib=rabbit
```

suborb, 2017-09-03:
That suggests that test.cfg isn't updated.

pauloscustodio, 2017-09-04:
I've fixed the problem in my sandbox.

We need different libraries for different CPUs, as the library code may be coded differently, as you spotted in cpi. 

I've changed z80asm to use one of the z80asm-CPU-IXIY.lib libraries depending on --cpu and --IXIY. In the future I may add --reserve-af1 to allow a seamless use of ex af,af' on a zx81. The set of libraries is generated and installed in ${PREFIX}/lib, and is searched by z80asm in program-dir/../lib

@suborb, while committing I saw that you changed the src/z80asm/Makefile to not generate the z80asm.lib if CROSS is 0. Can you explain please, so that I try not to break things while merging?

suborb, 2017-09-04:
The nightly windows build is compiled on Linux. By default z80asm.lib was created as part of the build, so in this case it was trying to launch a windows binary on linux which failed (see #344). Hence setting a CROSS=1 variable when cross compiling. See https://github.com/z88dk/z88dk-build/blob/master/build2.sh for the nightly build script.

pauloscustodio, 2017-09-04:
Understood. I'll skip building the z80asm*.lib files if CROSS is 0.

You will need to change the build script after the merge to copy z80asm-*.lib instead of z80asm.lib. There is a set of 10 library files (lines 141, 182, 226 and 267 of build2.sh)

suborb, 2017-09-04:
Skip if CROSS=1

pauloscustodio, 2017-09-04:
Right!

commit 98be8010 (2017-09-04): Fix #343 - generate different libraries for different CPUs

Change z80asm to use one of the z80asm-CPU-IXIY.lib libraries depending on --cpu and --IXIY.
The set of libraries is generated and installed in ${PREFIX}/lib, and is searched automatically
by z80asm in program-dir/../lib
closed 2017-09-04 by pauloscustodio