z88dk/z88dk #344: z80asm: strange errors in the nightly build

aralbrec, 2017-09-04:
Some odd errors in the nightly while building the opcode emulation library:

```
./z80asm.exe -xz80asm.lib -d dev/z80asm_lib/z80asm_cpir.asm dev/z80asm_lib/z80asm_cpd.asm dev/z80asm_lib/z80asm_rld.asm dev/z80asm_lib/z80asm_cpdr.asm dev/z80asm_lib/z80asm_rrd.asm dev/z80asm_lib/z80asm_cpi.asm
./z80asm.exe: 1: ./z80asm.exe: MZÿÿ¸@€º´: not found
./z80asm.exe: 2: ./z80asm.exe: xÂ¬Y!: not found
./z80asm.exe: 3: ./z80asm.exe: Syntax error: "(" unexpected
```

suborb, 2017-09-04:
That's what happens when attempt to run a windows exe on Linux

suborb, 2017-09-04:
I think that one is sorted with a bit of makefile kludging. Next problem is with z88dk-install. We compile under mingw so Windows.h isn't available. Is there a reason why you've used windows specific code rather than the dirent code?

suborb, 2017-09-04:
Right case differences, windows.h works. I'm pushing through another build now

suborb, 2017-09-04:
We've got some z80asm warnings that don't seem to make a lot of sense:

```
z80asm  -xlib/sccz80/zx @target/zx/library/zx_sccz80.lst
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2671: interpreting indirect value as immediate
z80asm  -xlib/sdcc_ix/zx -D__SDCC -D__SDCC_IX @target/zx/library/zx_sdcc_ix.lst
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2671: interpreting indirect value as immediate
z80asm  --IXIY -xlib/sdcc_iy/zx -D__SDCC -D__SDCC_IY @target/zx/library/zx_sdcc_iy.lst
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2671: interpreting indirect value as immediate
m4 -DCFG_ASM_DEF target/zxn/config.m4 > target/zxn/config_zxn_private.inc
m4 -DCFG_ASM_PUB target/zxn/config.m4 > target/zxn/config_zxn_public.inc
m4 -DCFG_C_DEF target/zxn/config.m4 > target/zxn/config_zxn.h
cp target/zxn/config_zxn_private.inc config_private.inc
ZX7: Optimal LZ77/LZSS compression by Einar Saukas
File converted from 13400 to 5887 bytes! (delta 4)
ZX7: Optimal LZ77/LZSS compression by Einar Saukas
File converted from 13400 to 5886 bytes! (delta 3)
Errors in source file arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm:
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2671: interpreting indirec
t value as immediate
                   ^ ----         ex      de, hl
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2671: interpreting indirec
t value as immediate
                   ^ ----         ex      de, hl
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2671: interpreting indirect value as immediate
                   ^ ----         ex      de, hl
z80asm --cpu=z80-zxn  -xlib/sccz80/zxn @target/zxn/library/zxn_sccz80.lst
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2835: interpreting indirect value as immediate
z80asm --cpu=z80-zxn  -xlib/sdcc_ix/zxn -D__SDCC -D__SDCC_IX @target/zxn/library/zxn_sdcc_ix.lst
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2835: interpreting indirect value as immediate
z80asm --cpu=z80-zxn  --IXIY -xlib/sdcc_iy/zxn -D__SDCC -D__SDCC_IY @target/zxn/library/zxn_sdcc_iy.lst
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2835: interpreting indirect value as immediate
```

aralbrec, 2017-09-04:
> Next problem is with z88dk-install. We compile under mingw so Windows.h isn't available. Is there a reason why you've used windows specific code rather than the dirent code?

Yes no dirent on vs2015.

```
Warning at file 'arch/zx/bifrost_h/z80/BIFROSTH_ENGINE.asm' line 2671: interpreting indirec
t value as immediate
                   ^ ----         ex      de, hl
```

I'm wondering if z80asm's line counting is out.  This shows up lower in the file:

`        sub     (9-__BIFROSTH_TILE_ORDER)*2`

for which a warning might make sense?

suborb, 2017-09-04:
> Yes no dirent on vs2015.

That would do it. I think in the past I must have used something like: https://github.com/tronkko/dirent to get round that.

> for which a warning might make sense?

Yup, that seems sensible. Issue #221 has got a bunch of work in this area.

aralbrec, 2017-09-04:
> I'm wondering if z80asm's line counting is out. This shows up lower in the file:
> sub (9-__BIFROSTH_TILE_ORDER)*2

That's what it was, z80asm has the line count wrong.

> That would do it. I think in the past I must have used something like: https://github.com/tronkko/dirent to get round that.

@suborb I've put dirent.h in to get rid of the windows code in the main source but you may have to fix this at the top of z88dk-install.c for the windows build:

```
#ifdef WIN32
#include "dirent.h"
#ifndef strcasecmp
#define strcasecmp(a,b) stricmp(a,b)
#endif
#else
#include <dirent.h>
#endif
```

If mingw supplies a standard dirent.h you don't want the local one with the windows stuff in it.

suborb, 2017-09-07:
The original issue is now sorted, so closing.

closed 2017-09-07 by suborb