z88dk/z88dk #350: nightly build:  windows z88dk-install binary non-functional

aralbrec, 2017-09-07:
z88dk-install.exe as produced by the nightly build is not working on windows.  I suspect it's this bit in z88dk-install.c:

```
#ifdef WIN32
#include "dirent.h"
#ifndef strcasecmp
#define strcasecmp(a,b) stricmp(a,b)
#endif
#else
#include <dirent.h>
#endif
```

The local dirent.h is an equivalent of dirent.h for windows as mentioned elsewhere and likely does not work with the cross-compiler.  I'm guessing the cross-compiler has a proper <dirent.h> to use instead, in which case a special test in `#ifdef WIN32` must be added.  If not, I'll have to revert to windows only code again.

aralbrec, 2017-09-08:
Nightly build still doesn't work.  Even entering "z88dk-install" which should show a brief help message comes up as permission denied.

suborb, 2017-09-08:
That's weird, I've installed wine and it does work as expected:

```
% wine z88dk-install.exe

Z88DK-INSTALL v9612-c62caa4-20170908
Third Party Library Installer
...
```

I think I've got a Windows 8 tablet somewhere. I'll charge it and see what it does on there.

suborb, 2017-09-08:
I really do loathe windows. Here's an interesting "security" feature I rediscovered:

I get a UAC pop up for z88dk-install.exe. I click yes and then it runs in a new command window.  I rename it z88dk.exe and it runs as you would expect.

Anyway, I think the dirent thing is, most ironically, a -I issue. dirent.h is in the current directory so mingw picks it up rather than its own version, we need to move the header into a different directory and rename the executable.

aralbrec, 2017-09-08:
My locally compiled z88dk-install is fine, the downloaded z88dk-install gets security applied.  And as you say if I change the name, all is good.  It looks like any name with "instal" or "install" as substring gets the windows treatment.

Are you still seeing a problem with dirent?  The binary is working with the right name but aside from that the cross-compiled version is much larger ~140k versus vs2015 at ~20k so I assume that's all of mingw's dirent compatibility stuff.

suborb, 2017-09-08:
Listing the installed libraries works for me which tests the opendir path. This is using the compatibility dirent.h.

The object file is about 15k, so I suspect the larger size is probably a bunch of wrappers around msvcrt to provide some sane behaviour.

commit 1a881001 (2017-09-11): Fix #350:  z88dk-install renamed to z88dk-lib
closed 2017-09-11 by aralbrec