z88dk/z88dk #352: szcc80 - math (and / or printf) issue

feilipu, 2017-09-07:
Related to issue #351, szcc80 causes calculation error.
```cpp
#include <stdio.h>
#include <string.h>

int main (void)
{
    uint32_t lba, lba2;
    uint16_t  sz_sect;
    
    sz_sect = 512;
    lba = 6;
    lba2 = lba + 0x80000000 / (sz_sect / 2);
    printf("%lu\n", lba2);
    return 0;
}
```
With SDCC the correct result.
```bash
zcc +yaz180 -subtype=basic_dcio -v --list -m -SO3 -clib=sdcc_iy --max-allocs-per-node200000 calculation_check.c -o calculation_check -create-app

Ok
print usr(0)
8388614
 0 
Ok
```
With SZCC80 an incorrect result.
```bash
zcc +yaz180 -subtype=basic_dcio -v --list -m -SO3 -clib=new calculation_check.c -o calculation_check -create-app

Ok
print usr(0)
6
 0 
Ok
```

suborb, 2017-09-07:
The operands for the long division are crossed. So we end up calculator divisor / dividend. I think this only occurs in the case where the constant is on the lhs of the expression.

I think #116 introduced this one, I'll fix this by the end of the weekend I hope.

suborb, 2017-09-08:
As usual a rabbit hole opened. But should be good now.

closed 2017-09-08 by suborb