z88dk/z88dk #353: z80asm has the line count wrong

pauloscustodio, 2017-09-07:
See #344

pauloscustodio, 2017-09-19:
Line count wrong comes from zcc:

test.c
```
int main() {
	int a = 256;
	int b = 256;
	int c = a * b;
	return c;
}
```
```zcc +z80 test.c -otest.bin > nul``` results in:

```
test.c:5: warning 165: integer overflow in expression
```

aralbrec, 2017-09-20:
(The compiler for the following line is zsdcc rather than sccz80)

> zcc +z80 test.c -otest.bin > nul results in:
> test.c:5: warning 165: integer overflow in expression

The line number is correct because the compiler has optimized all statements away except the return, where the problem is reported.

247   0000              _main:
248   0000              ;zzz.c:5: return c;
249   0000  21 00 00    	ld	hl,0x0000
250   0003  C9          	ret

aralbrec, 2017-09-20:
I've retested the original problem code and the line number problem has gone away.

closed 2017-09-20 by aralbrec