z88dk/z88dk #354: getk() in Aquarius not working (it always returns '£')

Fabrizio-Caruso, 2017-09-07:
getk() in Aquarius not working (it always returns '£')

Moreover if I try to print the content of the locations 
0x380A 0x380E 0x380F (where the keyboard input should be stored)
I always get the same results no matter which keys I press.

zx70, 2017-09-09:
This is odd, dstar uses getk and worked.  Could it be that different rom versions existed?

Fabrizio-Caruso, 2017-09-09:
I am testing with something like:

while(1)
         {
         int i;
         printf("%c \n", getk());
         for(i=0;i<2000;++i)
             {
             }
}

I have run such examples with Virtual Aquarius.

zx70, 2017-09-12:
I tested your sample code.
You hit a limit of the fputc_cons function, not filtering out the zero value (normally it equals to NUL, so nothing should be displayed).  The easy workaround is not to print NULs.
On your test program you can verify it by replacing getk() with 0 or by keeping a key pressed long enough to see its code appear.

You may be also be interested in the ANSI library variant, it can be activated by adding:
-pragma-need=ansiterminal
       --or--
-clib=ansi

closed 2017-09-12 by zx70

Fabrizio-Caruso, 2017-09-12:
Getk on the Aquarius target is behaving as if it had no buffer: a small fraction of the key presses is detected. Moreover keeping the key pressed does not help. Clib=ANSI makes no difference in this respect.