z88dk/z88dk #355: gotoxy in MSX not using the whole width and not working for x=0

Fabrizio-Caruso, 2017-09-07:
CONIO behaves differently across the Z88DK that support it.

The MSX target has the following problems:
- gotoxy only works on 37 columns out of 40 (it uses the BASIC text area)
- gotoxy with x=0 does not work (no character is displayed at that position)
Moreover CONIO triggers scrolling, which is usually not what is wanted when displaying characters.

IMHO, CONIO should behave equally across all Z88DK targets and possibly as in DOS/Windows
and other kits (CMOC, CC65).

(0,0) should be the left-top corner 
(XSIZE-1,YSIZE-1) should be the right-bottom corner

(0,0)                   ...           (XSIZE-1,0)
                          ...   
(0,YSIZE-1)         ...           (XISE-1,YSIZE-1)

zx70, 2017-09-14:
Same answer given already for #370:  Due to the crappy results with the non-ansi based gotoxy() implementation I think it is time to get rid of them.   An implementation for the 40 columns text mode of the MSX/SVI family will follow soon.

closed 2017-09-14 by zx70

Fabrizio-Caruso, 2017-09-14:
Please provide some conio support for text mode 1 (32 columns) which has colors.
Mode 0 is monochromatic and not suited for games.

zx70, 2017-09-14:
Good point, thankyou.   I also remember some difference in the "set text mode" options between MSX and SVI, hope they won't cause troubles.