z88dk/z88dk #357: szcc80 compile issue with FatFS

feilipu, 2017-09-10:
Trying to compile this [Version 13 Patch 2](https://github.com/z88dk/z88dk/files/1288178/ff13p2.zip) of ChaN's FatFS, there is an issue with szcc80.
```bash
sccz80:"ff.h" L:266 Error:#45:Missing token, expecting ) got c
sccz80:"ff.h" L:266 Warning:#14:Expected ','
sccz80:"ff.h" L:266 Error:#39:Illegal symbol name: 
Compilation aborted
```
```cpp
FRESULT f_forward (FIL* fp, UINT(*func)(const BYTE*,UINT), UINT btf, UINT* bf);	/* Forward data to the stream */
```
This code compiles correctly with zsdcc.

suborb, 2017-09-10:
That's to be expected with sccz80. Function pointers can't be defined with arguments.

Sorting out the decl parsing is on my list of things that I want to do, it's a fairly big job so might take me a while to actually get around to doing.

feilipu, 2017-09-10:
No problem or urgency. I assumed it was an inherent small C issue.
Just flagging it as a use case example. Perhaps useful for testing purposes later.

suborb, 2018-01-12:
I think this one should actually be solved with the decl rewrite

closed 2018-01-12 by suborb