z88dk/z88dk #359: Are the rcmx_cpi & friends needed in the library?

pauloscustodio, 2017-09-10:
z80asm now includes the emulation libraries for CPI, CPIR, CPD and CPDR at the link phase if assembling for a Rabbit.

The source of these libraries is in src/z80asm/dev/z80asm_lib and they are deployed to the lib/ directory during install

I've found and fixed a bug in the output of the H and N flags in the CPIR and CPDR routines in commit c2bcbbfe389cca9c3d89d9605cf1bcb06c9c4756 

Should I update the copies in libsrc/z80_crt0s/z80_emu, or remove them and keep them only in the z80asm source tree?

aralbrec, 2017-09-11:
z80asm will link these replacements last so it will be possible for the user or library to override them?

suborb, 2017-09-11:
I think I adjusted all references to them to the z80asm version, but forgot to remove the rcmx versions. So they can go.

pauloscustodio, 2017-09-11:
Yes, z80asm will search the z80asm-{cpu}-{ixiy}.lib last, allowing the user to override the __z80asm__xxx functions.

commit 7bdd24dc (2017-09-11): Fix #359 - remove functions that now are part of z80asm*.lib
closed 2017-09-11 by pauloscustodio