z88dk/z88dk #361: How to compile/load programs for the VZ200 target?

Fabrizio-Caruso, 2017-09-11:
Hi

In https://www.z88dk.org/wiki/doku.php?id=platform:vz200
I could not understand what the current loading procedure is...
The wiki page only mentions the old procedure.

I am using the only emulator I could find for this system, namely:  VZEM.
If I compile with
zcc +vz  -subtype=basic -vn -D__VZ__ -lndos -Cz--audio -create-app -o  vz200_experimental ..
(and without -subtype)

I get files I cannot use with the emulator.
If I use the .wav file, the emulator complains about the sampling rate.

zx70, 2017-09-13:
The old emulators require the simpler VZ files (barely binary images with a small leading stub).
The very first build examples in the that documentation page just suggest to rename the output file without involving appmake.  You can either rename a.bin to a.vz or add the "-o a.vz" option.

Fabrizio-Caruso, 2017-09-14:
Thanks! This solves my issue!

closed 2017-09-14 by zx70