z88dk/z88dk #363: appmake +zxn: memory bank naming

aralbrec, 2017-09-11:
So I think I have a decent method for dealing with bankswitching and memory banks in general.

For the zx next target in particular there are:  40 [104] * 16k banks paged at 0xc000.  Each of these 40 [104] 16k pages can be divided into two 8k pages - 80 [208] in total.  These 8k pages can be placed in any 8k segment of the z80's 64k space via special MMU instructions (not added to the instruction list yet).  In addition there is 256k of divmmc memory that is divided into 8k pages and can be paged into the 0x2000-0x3fff region.

The numbers above are for an unexpanded machine and the numbers in square brackets are for a machine with 1MB expansion included.

The tools, in particular `appmake +glue`, recognize section names with substring "BANK_NN" (NN = hex number) as belonging into the 64k bank #NN.  Sections without that substring go into the main 64k bank.  `appmake +glue` scans z80asm's map file for section org and reads the output binaries to put together a single binary for each 64k bank.  Sections that overlap in a 64k bank get errors or warnings.

Right now "NN" is a hex number.  I'm thinking of changing that to decimal.  Hex is fine when the number of banks is small but when the number of banks gets up to 208, that's a lot of conversion in the head especially since bank numbers are thought of in decimal.  This is going to affect the sega master system, the zx next, and anyone using `appmake +glue` now.  The zx target is not affected because the bank numbers only go up to 07.

The other issue is how we're going to divide a single 16k bank NN into two 8k banks so that code / data can be generated into the 16k bank as a whole or into each 8k bank making up that whole.

I'm leaning toward this:

```
SECTION BANK_00    ;; bank 00 as example
org 0xc000  ;; 16k banks always paged in at 0xc000, modifiable by pragma
SECTION BANK_00_0   ;; the first 8k of bank_00
org 0xc000   ;; org is modifiable by pragma, important because this 8k can appear anywhere
SECTION BANK_00_1  ;; the second 8k of bank_00
org 0xe000   ;; org is modifiable by pragma, important because this 8k can appear anywhere
```

(Are there better names for the first and second halves of the 16k bank?)

Because the 8k segments can appear anywhere in the 64k memory space, their org can be set to anything by the user.  Then `appmake +zxn` will treat the BANK_NN banks in a special manner.  It will mask off the top bits of the org address of each of the sections so that they lie in 0-16k.  And then it will merge all BANK_NN sections into one 16k section before generating the output.  The user can define more BANK_NN sections separately that will also merge into this 16k because the bank assignment is done by name "BANK_NN".

The divmmc memory must be handled differently with the 256k space divided into 32 8k sections.  To keep it generally applicable to all targets, I am thinking of calling these `SECTION ALTBANK_NN` so that the BANK_NN and ALTBANK_NN are two independent ways to name 64k spaces.

Is it sounding sane or any suggestions about naming, etc?

pauloscustodio, 2017-09-11:
Sounds good to me.

I suggest for the two halves BANK_xx_L and BANK_xx_H - it is maybe less confusing than just two numbers. 
I agree with the numbering in decimal.

How will the system handle loading and relocation? If the code is assembled for org 0xC000, it will not run at a different address without relocation.

Each of these BANK_xx is an individual section, or a set of sections with the same root (e.g. BANK_xx_code, BANK_xx_data, etc)?

Would section groups help? They are in my plans for z80asm.

aralbrec, 2017-09-11:
> I suggest for the two halves BANK_xx_L and BANK_xx_H - it is maybe less confusing than just two numbers.

I like that more.

> How will the system handle loading and relocation? If the code is assembled for org 0xC000, it will not run at a different address without relocation.

If it's code, it has to be banked into the right 8k segment to run.  The user has the option of setting org to any 16-bit address for z80asm's consumption but the appmake program will strip off the extra address bits to get an offset into the 16k bank for placement purposes.  If the 8k bank is paged into the right 8k segment, then that particular code will run fine.  There isn't any runtime relocation.  Data, on the other hand, can always be accessed as an offset from the current segment address.

For loading, the zx next integrates esxdos which is a posix disk api for the built-in sd card.  So the idea will be to get appmake to pack all the banks into a single file, up to 1MB+ in size and then the crt will open a file on itself and load up all the memory banks before starting.

To run the program, I am looking at making a special .run dot command.  Dot commands are special programs that are stored in the BIN directory and can be run from basic.  .run will search for a program according to a PATH variable and then load it up depending on the file type.  The usual file types will be supported .tap, .sna, etc and also this new format for 1MB games.

We'll still make sure that other program types can be generated.  The appmake step described above is going to output raw binaries for each bank.  Then the next appmake step can turn that into a tap, sna or 1MB file.  Extra binaries can always be stored directly on the sd card and can be loaded from disk by the program on demand via esxdos.

> Each of these BANK_xx is an individual section, or a set of sections with the same root (e.g. BANK_xx_code, BANK_xx_data, etc)?

It's a set of sections.  If the section name contains "BANK_NN" as substring, it's understood to be in NN's 64k bank.  So you can define code, data, etc subsections but for the zx next in particular, that's not applicable since it's all ram.

The default memory map has the c code and library code going into the main bank.  That's the programming model most systems are using - you have a fixed bank containing the main code and then a bank that is paged in as needed to access data or level-specific code.  But you can change this by modifying the library's memory map so that stuff is assigned elsewhere.  And you can get the compiler to place compiled c code in specific sections athough you have to be careful that any subroutines the c compiler calls are accessible while that code runs, probably by having that stuff in a fixed bank.

> Would section groups help? They are in my plans for z80asm.

The memory map being set up now is effectively doing section groups but in an ad hoc manner.   I still have to take time to think about that proposal.  One very clear need now is a way to specify sections by wildcard so I can say "all sections with name code_* go here unless a more specific rule is someplace else".  Right now, the memory map for newlib contains, I don't know, 50? defined sections and each one has to be named to be in the memory map.  When a 3rd party library is added, it's forced to use existing section names otherwise its code does not get placed into the output binary and you can't create a new name because that requires the user to add it to the memory map explicitly.  It would be good to say my library is in section "code_graphics" and then the existing memory map would place by default using section/group "code_*" while still giving the end user the ability to place that library's code in whatever memory bank he wants by being more specific.

aralbrec, 2017-09-15:
Code to handle multiple memory banks has now been added to appmake.  `appmake +glue` has already been modified to use it and `appmake +zxn`, `appmake +zx` and `appmake +sms` will be modified soon to use it as well.

There is a concept of "main bank" and multiple user-defined "bank spaces".   The "main bank" is a 64k memory bank that contains all sections without special names.

Each "bank space" is identified by a substring in the section name.  A bank space can contain up to 256 64k memory banks identified by number.  The common example is the bank space "BANK" and targets can assign to independent 64k spaces by creating sections "BANK_0", "BANK_1", ... "BANK_255".  So, eg, the zx spectrum 128 has eight 16k banks that would be defined as "BANK_0" through "BANK_7" contained in bank space "BANK".  Likewise, the sega master system has 16k banks "BANK_0" through "BANK_31" contained in bank space "BANK".  It's important to note that the bank space identifier is a substring of the section name so "BANK_0_code" and "BANK_0_H" are both part of the same 64k bank "BANK_0".

The reason why there can be multiple bank spaces is that targets may have memory in different pools that are not treated the same.  The need is coming out of the zx next target.  For this target, two bank spaces will be defined:  "BANK" and "DIVMMC".  "BANK" corresponds to 16k memory pages that can be banked in at 0xc000 whereas "DIVMMC" corresponds to a different pool of 8k pages that can only be paged in at 0x2000.  On the next, the "BANK" pages are also special in that they can be divided into two 8k pages (we are calling them "BANK_nnn_L" and "BANK_nnn_H") which can then be paged into any 8k segment of the z80's 64k space.  Appmake will be able to handle this using the scheme described in the first post.

The new functions added will automatically scan the generated map file and look at the output binaries to generate a data structure describing the memory banking.  A particular target can define bank spaces and then call these functions, operate on the bank spaces if needed and then use more functions to generate ouput binaries describing each 64k bank.

Sections with substring "_align_nnn" (nnn is a power of 2) are also check for alignment during the process.

aralbrec, 2017-09-15:
Example use:
https://drive.google.com/file/d/0B6XhJJ33xpOWYjEwRWNlMUlYdjA/view?usp=sharing

```
zcc +zx -vn --no-crt -clib=sdcc_iy zmap.asm test.asm -o test -m
appmake +glue -b test -f 0 --clean
```

The output will be:

```
test__.bin (main bank)
test__001.bin (BANK 01)
test__002.bin (BANK 02)
```

aralbrec, 2017-09-20:
The zxn memory model is now implemented with commit https://github.com/z88dk/z88dk/commit/a28e5332e72758d95b1609783b9214c5836cae38 .

Two banks spaces are defined:   "BANK" and "DIV".

"BANK" is the usual set of 16k pages paged in at address 0xc000 as inherited from the 128k spectrum.  In the basic zxn machine there are 40 such pages which we are calling "BANK_0" through "BANK_39" in decimal.  In the expanded memory machine there are 104 such pages extended this through "BANK_103".  Things are complicated, however, by the fact each 16k pages can be split into two 8k halves and these 8k portions can be mapped into any 8k slot in the z80's 64k address space.  To accommodate that we split each 16k BANK into low (bottom 8k) and high (upper 8k) portions.  So in addition to the 16k banks defined above, there are also banks like "BANK_0_L" and "BANK_0_H".  You can org your data and code anywhere in the 64k space and assign it to one of these 16k banks.  The org address will be properlychanged into an offset into the 16k bank so that all binaries belonging to the bank will be properly merged into 16k.  The actual org address will be used in code generation and you will be able to map 8k sections anywhere and have it properly run if the org address is correct.

The "DIV" bank space is for holding code and data destined for divmmc memory.  The zxn has 32 8k banks of divmmc memory which we are called "DIV_0" through "DIV_31".  Each of these banks have fixed org at 0x2000 and size 8k.

The banking system in z88dk is 100% flexible.  You can define your own sections with arbitrary org that belong to any of these memory banks.  If you define a section with name "MYSTUFF_BANK_20_L", that stuff will be placed in 16k bank 20 in the lower 8k.  Everything is taken care of automatically.

tap and esxdos dot command generation do not use the banking model as they don't need it.

sna generation uses the model and properly generates banks 0-8 in 128k snapshots as described above.  Seriously consider adding "--clean" to your appmake invoke so that appmake deletes files it uses in its sna generation.  That way if you have more output binaries that do not fit into snas, only they will be left and you can find other means to get them into zxn memory.

Two more appmake types are coming:

bin:  Do all the bank consolidation as described above and only leave behind binaries representing the actual zxn memory bank contents.

zxn: A new output type that will automatically load all zxn memory banks and start your program.  zxn programs can megabytes in size and will be started by a special dot command that will act as the one-stop command to start all programs on the next.

aralbrec, 2017-10-26:
#477 proposes extending this banking scheme, so closing.

closed 2017-10-26 by aralbrec