z88dk/z88dk #364: _sc_cursor_pos already defined error in M5 and SC3000

Fabrizio-Caruso, 2017-09-11:
While trying to make my "universal" game work with the M5 and SC3000 I am getting the following error
"
symbol '_sc_cursor_pos' already defined in module 'C__Users_Brizio_Retro_ASCII_CHASE_merged_display_macros
"
I get no similar conio-related error with other targets.

In my code I include conio.h only in one header (display_macros.h) file which is protected from re-definitions by an #ifndef directive.

I do not understand why this symbol (only found in M5 and SC3000 conio.h) is causing a problem (and no the other symbols in conio.h).

suborb, 2017-09-12:
See: https://github.com/z88dk/z88dk/commit/6287fb44e35f685ca5e577c22854413ada5fafa2#diff-92717702464f42f245313a5988b52a78 

I'll sort this out in the next couple of days.

zx70, 2017-09-13:
Thanks for pointing it out, Dom.

closed 2017-09-13 by suborb

Fabrizio-Caruso, 2017-09-14:
Problem while compiling for the SORD M5:
PROCESSING CRT
z80asm   -s  --cpu=z80 -I"c:/z88dk//lib" -I"c:/z88dk/\lib"  "C:\Users\fcaruso\AppData\Local\Temp\zcc
638412.asm"
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc638412.asm' line 115: syntax error
1 errors occurred during assembly
Errors in source file c:/z88dk/\lib\m5_crt0.asm:
Error at file 'C:\Users\fcaruso\AppData\Local\Temp\zcc638412.asm' line 115: syntax error
                   ^ ---- _sc_cursor_pos        defw    0x70a6

suborb, 2017-09-14:
Sorry, I thought I'd tested the m5 as well as sc3000.

It's missing a colon after pos on the highlighted line - if you just temporarily do it, I'll commit a change this evening.

Fabrizio-Caruso, 2017-09-14:
No worries. I can wait for your commit and use tomorrow's snapshot.

commit 6f00a9e6 (2017-09-13): Fix #364: Don't define sc_cursor_pos each time conio.h is included