z88dk/z88dk #369: tail jump optimisation issue - question

feilipu, 2017-09-13:
I'm trying to make a piece of code atomic, using the `cpu_push_di()` and `cpu_pop_ei()` functions referred to in #367.

But, the peephole optimiser is moving the call to `call _z180_pop_ei` to after the `pop xi` and converting it to a `jp _z180_pop_ei` which will mess up the stack, I believe.

Although `_z180_pop_ei` takes no parameters, it disturbs the stack. Is there a way of signalling this to the C interface?

The C code:
```c
void
set_system_time(time_t timestamp)
{
    cpu_push_di();
	_system_time = timestamp;
	cpu_pop_ei();
}
```
and the resulting compiler output.
```asm
486   0000              	SECTION code_compiler
487   0000              ;./set_system_time.c:45: set_system_time(time_t timestamp)
488   0000              ;	---------------------------------
489   0000              ; Function set_system_time
490   0000              ; ---------------------------------
491   0000              _set_system_time:
492   0000  CD 00 00    	call	___sdcc_enter_ix
493   0003              ;./set_system_time.c:47: cpu_push_di();
494   0003  CD 00 00    	call	_z180_push_di
495   0006              ;./set_system_time.c:48: _system_time = timestamp;
496   0006  DD 6E 04    	ld	l,(ix+4)
497   0009  DD 66 05    	ld	h,(ix+5)
498   000C  22 00 00    	ld	(__system_time),hl
499   000F  DD 6E 06    	ld	l,(ix+6)
500   0012  DD 66 07    	ld	h,(ix+7)
501   0015  22 02 00    	ld	(__system_time + 2),hl
502   0018              ;./set_system_time.c:49: cpu_pop_ei();
503   0018  DD E1       	pop	ix
504   001A  C3 00 00    	jp	_z180_pop_ei
505   001D              	SECTION IGNORE
```

suborb, 2017-09-13:
Both sccz80 and sdcc support the __critical function annotation (and critical blocks). I think this does what you want to do - save and restore the interrupt enabled state.

feilipu, 2017-09-13:
Ok. Yes,thanks. Much better method.

Was looking in the libraries for this outcome.
Should have read the compiler instructions.

feilipu, 2017-09-13:
Ok. Yes,thanks. Much better method.

Was looking in the libraries for this outcome.
Should have read the compiler instructions.

Closed #367.

feilipu, 2017-09-13:
Actually, the bad optimisation is still happening with `__critical{ }`
```c
#include "time.h"
extern time_t _system_time;
void
set_system_time(time_t timestamp)
{
  __critical
  {
    _system_time = timestamp;
  }
}
```

```asm390   0000              	SECTION code_compiler
391   0000              ;./set_system_time.c:43: set_system_time(time_t timestamp)
392   0000              ;	---------------------------------
393   0000              ; Function set_system_time
394   0000              ; ---------------------------------
395   0000              _set_system_time:
396   0000  CD 00 00    	call	___sdcc_enter_ix
397   0003              ;./set_system_time.c:48: }
398   0003  CD 00 00    	call	____sdcc_cpu_push_di
399   0006              ;./set_system_time.c:47: _system_time = timestamp;
400   0006  DD 6E 04    	ld	l,(ix+4)
401   0009  DD 66 05    	ld	h,(ix+5)
402   000C  22 00 00    	ld	(__system_time),hl
403   000F  DD 6E 06    	ld	l,(ix+6)
404   0012  DD 66 07    	ld	h,(ix+7)
405   0015  22 02 00    	ld	(__system_time + 2),hl
406   0018  DD E1       	pop	ix
407   001A  C3 00 00    	jp	____sdcc_cpu_pop_ei_jp
408   001D              	SECTION IGNORE
```
But the correct result is obtained for below code
(except that this results in code in the `__critical` section that is not required to be critical).
```c
#include "time.h"
extern time_t _system_time;
void
set_system_time(time_t timestamp) __critical
{
  _system_time = timestamp;
}
```

aralbrec, 2017-09-13:
I think the first instance in the original post was due to the custom function cpu_pop_ei().

sdcc is doing tail code optimization on the library calls that restore interrupt state and we've been fixing that in post-processing which you can see in `jp ____sdcc_cpu_pop_ei_jp` where a different entry point is used instead of `jp ____sdcc_cpu_pop_ei` so that the exit can happen via jump.

But you are right that the `pop ix` done before the jump will foul it up.  I think it's difficult to add a special case to the tail call optimizations but maybe another band-aid will work - I'll look into both anyway.

aralbrec, 2017-09-13:
Band-aid solution applied.  It should be more reliable than the name band-aid suggests.  There are a number of issues accumulating for sdcc that will need us to go in and actually start working on the compiler rather than around it.

feilipu, 2017-09-14:
Hmm. I think it may be better to leave the band-aid off.
```c
void
set_system_time(time_t timestamp) __critical
{
    _system_time = timestamp;
}
```
now gives this...
```asm
390   0000              	SECTION code_compiler
391   0000              ;./set_system_time.c:43: set_system_time(time_t timestamp) __critical
392   0000              ;	---------------------------------
393   0000              ; Function set_system_time
394   0000              ; ---------------------------------
395   0000              _set_system_time:
396   0000  CD 00 00    	call	____sdcc_cpu_push_di
397   0003  CD 00 00    	call	___sdcc_enter_ix
398   0006              ;./set_system_time.c:45: _system_time = timestamp;
399   0006  DD 6E 06    	ld	l,(ix+6)
400   0009  DD 66 07    	ld	h,(ix+7)
401   000C  22 00 00    	ld	(__system_time),hl
402   000F  DD 6E 08    	ld	l,(ix+8)
403   0012  DD 66 09    	ld	h,(ix+9)
404   0015  22 02 00    	ld	(__system_time + 2),hl
405   0018  CD 00 00    	call	____sdcc_cpu_pop_ei
406   001B  DD E1       	pop	ix
407   001D  C9          	ret
408   001E              	SECTION IGNORE
```
I guess it can't know that the different usage of `__critical` means that the optimisation is actually good (correct).

Perhaps (until this can be fixed with sdcc), it is best remove the band-aid, and just note in the wiki references for usage that a `__critical` section can't be the last thing in a function?

closed 2017-09-16 by aralbrec

aralbrec, 2017-09-16:
Let's hope this puts it to rest.

commit 87c411f1 (2017-09-13): Fix #369: correct tail call optimization by zsdcc on library's critical exit function
commit 57ac6e35 (2017-09-16): Fix #369: tail call optimization on special crtitical section close function repaired