z88dk/z88dk #370: SVI conio/display issues and possible solution

Fabrizio-Caruso, 2017-09-13:
The SVI target shows the cursor while displaying characters on the screen.

Moreover the SVI target shares some problems with MSX and other targets:
1. only the columns used in BASIC are used by conio (problem in MSX as well)
2. if gotoxy is used with x=0, then the first characters is displayed on the right end of the screen (some problem in MSX and CPC)

POSSIBLE SOLUTION:
All these problems can be sorted out with a simple approach: just write straight on the video ram:
- no cursor is displayed
- all area is used correctly
Remark:
If video ram works as in MSX (as I suppose), the base address has to be adjusted depending on the video mode used.

zx70, 2017-09-14:
Due to the crappy results with the non-ansi based gotoxy() implementation I think it is time to get rid of them.   An implementation for the 40 columns text mode of the MSX/SVI family will follow soon:  getting a flexible resolution implementation in graphics mode similar to the Spectrum's one is more complex than expected because of the indirect addressing of the video memory page.

Fabrizio-Caruso, 2017-09-14:
Flexible resolution in graphics mode is not what people need when doing color character graphics.
I am trying to use color character graphics: so I need text mode with fixed size.
Fixed size is important in order to have color attributes on each single character.

This is a totally different use case.
There are already library for monochromatic graphics with sprites.

Fabrizio-Caruso, 2017-09-14:
@zx70 , 40 column text mode is monochromatic in MSX.
Could you please also support MSX mode 1 (color 32 column text mode)?

zx70, 2017-09-23:
Things are evolving,  but few extra problems appeared meanwhile on the msx target. 
The minor bugs are fixed already but there is an odd problem on the msx making in example the graphics data in dstar to be corrupted.   I'm investigating.

zx70, 2017-10-02:
As said #427 is in the way, so I'm using also the sci target to debug.. accessing the BPD is not as easy as hoped

zx70, 2017-10-03:
The ANSI-VT version for MSX and SVI now works, hope the nightly build will be fine.
It can be enabled, eg by adding:
-pragma-define:ansicolumns=51 -pragma-need=ansiterminal
It is fully compatible with the Spectrum version, but the 8 bit font specific optimization is (at the moment) not present.  Scroll should be fixed to save memory, and Newton, SEGA and others will be added soon (code port is straightforward for the platforms having the same VDP).

zx70, 2017-10-07:
Changing the gotoxy origin to (0,0)

The msdos standard by Microsoft and Borland for conio is (1,1) but cc65 and many custom implementations on many platforms (including msdos) had (0,0).  I'm leaving the text size detection as it is,  it will return the rows and columns count.

zx70, 2017-10-09:
I restored the old vt-ansi option in text mode as.. an option for zx80, zx81 (in gfx81.lib to stay aside the existing graphics ones) and Lambda (to be enabled soon in appmake, extending the zx81 dependencies)

closed 2017-10-12 by zx70