z88dk/z88dk #374: newlib +yaz180:  add copt rules file to allow inlining of asm functions

aralbrec, 2017-09-13:
From #373.
Add mechanism to yaz180 to inline asm instructions without affecting optimizer.

aralbrec, 2017-09-13:
I can see this becoming a thing where each target may get additional post-processing steps.  We probably should have a better place to store the various peephole rules for copt, sccz80 and sdcc.

aralbrec, 2017-09-13:
For __BREAK, I've changed the yaz180.h proto header to this:

```
// Halt the YAZ180 with single step hardware.

#define __BREAK  __BREAK_HELPER()
__OPROTO(`a,d,e,h,l,iy,iyh',`a,d,e,h,l,iyl,iyh',void,,__BREAK_HELPER,void)
```

The define to `__BREAK_HELPER()` allows `__BREAK;` to be used as is inside c code.

`__BREAK_HELPER` is being defined as a function that preserves all registers except bc.  This way inlining the asm won't affect any registers zsdcc is using.

In a new file yaz180_rule.1, a copt rule is added to replace the function call the compilers are generating with inlined asm:

```
	call	___BREAK_HELPER
=
	IFNDEF __IO_BREAK
		EXTERN	__IO_BREAK
	ENDIF
	ld	bc,__IO_BREAK
	out	(c),c

	call	__BREAK_HELPER
=
	IFNDEF __IO_BREAK
		EXTERN	__IO_BREAK
	ENDIF
	ld	bc,__IO_BREAK
	out	(c),c
```

zsdcc adds another leading underscore to the name, sccz80 does not.  I'm checking if `__IO_BREAK` is defined in case someone does something bizarre like `include "config_private.inc"` somehow as that will then include a file defining that label; I think an EXTERN will cause an error out of z80asm in that case.  The EXTERN will work because the crt exports those labels as PUBLIC.

suborb, 2017-09-13:
> I can see this becoming a thing where each target may get additional post-processing steps. We probably should have a better place to store the various peephole rules for copt, sccz80 and sdcc.

Yes, that was my thinking behind: https://github.com/z88dk/z88dk/tree/feature/more_build_changes/lib/arch/z88 Something didn't feel quite right about what I was doing though so I halted work on it.

feilipu, 2017-09-14:
I'm afraid that I've got a case where this doesn't quite work.

Where there is a simple assignment of function, the `___BREAK_HELPER` doesn't work.
Probably this won't occur very often (as it creates a useless program), but still.
```c
#include <stdlib.h>
#include <arch/yaz180.h>

void main(void)
{
    __BREAK;
    return;
}
```
The list looks like this
```
346   0000              	SECTION code_compiler
347   0000              ;./atest.c:4: void main(void)
348   0000              ;	---------------------------------
349   0000              ; Function main
350   0000              ; ---------------------------------
351   0000              ;./atest.c:6: __BREAK;
352   0000              ;./atest.c:8: return;
353   0000              	defc _main	=	___BREAK_HELPER
354   0000              	SECTION IGNORE
```
```bash
z80asm --cpu=z180 -b -d  -o"break_app" -m -s -L.  -L"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/yaz180" --list  -im -iyaz180    "@/tmp/tmpXXFAOtpi.lst"
Error at file '/tmp/tmpXX8g7pVZ.asm' line 353: symbol '___BREAK_HELPER' not defined
1 errors occurred during assembly
Errors in source file /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/yaz180/yaz180_crt.asm:
Error at file '/tmp/tmpXX8g7pVZ.asm' line 353: symbol '___BREAK_HELPER' not defined
                   ^ ---- 	defc _main	=	___BREAK_HELPER
```
Otherwise, it works as required.
Thanks :+1:

feilipu, 2017-09-15:
Could something like this work?
At least to remove the `ret` optimisation?

```
#define __BREAK;             \
#pragma save /* save the current optimization settings */  \
#pragma no-peep-return /* Do not let the peep hole optimizer do return optimizations */ \
	call	___BREAK_HELPER; \
#pragma restore /* turn the optimizations back on */ \
#endif
```

aralbrec, 2017-09-15:
I'll take a look at your solution tomorrow (it's getting too late here!).  I'm a little nervous about depending on whitespace to sort out problems but I think it may be possible to make exceptions for certain names.

feilipu, 2017-09-15:
Yes, white space makes me nervous too (not a python fan). Hence I mentioned it.
The [yaz180_rules.1](https://github.com/feilipu/z88dk/blob/9acc443f7daa249b3805b525c9aac8e47e764be5/lib/yaz180_rules.1) copt seems to need to match exactly with the output the compiler and initial copt produces.

The `call	___BREAK_HELPER` comes straight from the compiler with a tab character.
The `jp  ___BREAK_HELPER` comes from copt with two spaces.
So `yaz180_rules.1` has to match these inputs exactly, or it breaks.
I tested it.
Seems fragile to me.

aralbrec, 2017-09-16:
I've fixed the space brittleness issue.  In reality, sdcc's peephole optmizer is very fragile, depending on exact formatting of instructions and spacing for it to work.

closed 2017-09-16 by aralbrec

feilipu, 2017-09-16:
It has made you a s**t load of work, just to insert two lines of asm.
Sounds like technical overhang to me. :sweat_smile:

Anyway, that solution looks solid.
The `defc` case is a moot one anyway.

(Unrelated, but worth celebrating... just fixed a longstanding PLD issue with my implementation of a soft [flash programmer](http://www.ecstaticlyrics.com/electronics/Z80/EEPROM_programmer/), so the last outstanding yaz180 hardware concern is a [I2C](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_UnitTest/z180i2c.asm) proof of life. :hand: )

commit b34bafef (2017-09-13): Fix #374: add post-processing step with copt for yaz180 target
commit 61676d09 (2017-09-16): Fix #374: yaz180 resolve issues with inlined BREAK macro