z88dk/z88dk #375: Loading a Jupiter Ace binary (or .tap?)

Fabrizio-Caruso, 2017-09-14:
I have tried to follow the instructions on how to compile and use the binary for the Jupiter Ace target.

It is not clear to me what I should do:
0 0 bload a.bin
will do nothing by itself.
Then I need to open a file but the EightyOne emulator does not let me choose .bin files.
If I select a .tap file then it loops forever 
slowing a  z88dk code file.

zx70, 2017-09-14:
You must use the tap or the wav file (-create-app option), EitghyOne should be a good choice, make sure your virtual recorder is playing the tape correctly (from the top of it).

Fabrizio-Caruso, 2017-09-14:
What should I type then in Forth?
Should I follow the instructions on what to type (i..e, type 0 0 bload a.bin)? and the choose the .wav file instead of the suggested .bin?

Fabrizio-Caruso, 2017-09-14:
Problem solved! 
My issue came from the fact that I was using a path BEFORE the name a.bin.
The emulator searches for a name containing the path and so, it never was not able to find the program.

closed 2017-09-14 by Fabrizio-Caruso