z88dk/z88dk #376: Add banking support to ticks

suborb, 2017-09-14:
Ticks is turning into quite a useful test tool, it would be useful for it to support banking so that we can test out different strategies for handling multiple bank programs efficiently.

- [ ] Update all mem[x] references in ticks source code to call functions
- [ ] Define a banking switching strategy for z80/z180
- [ ] Look into XPC support for Rabbit
- [ ] Update debugger/disassembler to allow bank references to be used
- [ ] Implement modular banking strategies
- [ ] ticks needs to load code into banks

Then onto code generation. sccz80 already (theoretically) supports calling across banks (via a trampoline) using `__SHARED__` and `-shareoffset` which allows C functions to access their arguments with an additional offset. Whether that's useful here remains to be seen.

feilipu, 2017-09-14:
On the issue of banking strategy for z80/z180, this [bpBIOS Section 3.1](https://github.com/z88dk/z88dk/files/1302641/Banked_and_Portable-BIOS.pdf) document has the best clarity on the banked CPM versions that I've found. I've got building a bpBIOS implementation on my roadmap for 2018.

When I started building my yaz180 machine (naively, without any background in z80 architecture), I thought to keep the lowest logical page constant, because the RST interrupts couldn't be moved, so I filled the first 2kB physical addresses with Flash for booting a system, and the second 2kB with RAM for system buffers, vectors, trampolines, etc, and system stack usage (assumed logical addressing too). The further intention was that the rest of the 4kB logical pages could be either Flash or RAM (in Banked or Common 1) depending on what the user desired. Only the lowest page is dual (Flash / RAM) use on natural 4kB increments.

After reading what CP/M did with banking, perhaps the above discussion in moot. But, supporting bpBIOS CP/M is a fairly high priority, IMHO.

suborb, 2018-04-17:
Update on what's been happening:

#698 implemented ZX Next style banking
#641 added far calling techniques to both sdcc and sccz80

suborb, 2021-05-22:
Not much demand for this, so closing.

closed 2021-05-22 by suborb