z88dk/z88dk #378: Update toolchain to compile and link with GLib

pauloscustodio, 2017-09-14:
I know this is not consensual.

As my plans for a complete rewrite of z80asm in C++, to use the C++STL, are more or less in the low-priority range, and I will have to do fundamental changes in the linker to allow the change requests on sections, banked memory and supporting the ZX Next, I think going to GLib is a mid-way that allows the usage of more powerful platform-independent libraries, reducing the code foot-print, while being able to leave parts of the code untouched.

For example, it solves the resent issues with srtcasecmp and dirent by providing it's own platform-independent routines.

I have setup a branch feature/z80asm_glib and will do my changes there. We will need to find a way to test the build toolchain based on that branch before merging.

Any thoughts?

suborb, 2017-09-14:
Previously we had glib as part of the build chain, I'm not sure if it's still there. The Mac version will certainly need updating.

However, the nightly build machine is a container so I can either:

a) Give you access to it
b) Give you access to a clone of it
c) Send you an lxc image of it
d) Sort it out myself

The choice is yours :)

pauloscustodio, 2017-09-14:
I'll go for option b) first, are revert to d) if I get stuck. OK?

suborb, 2017-09-14:
Sure. If you email me an ssh public key I'll sort it out over the weekend.

pauloscustodio, 2020-09-25:
I gave up...

closed 2020-09-25 by pauloscustodio