z88dk/z88dk #380: Classic libsrc changes

suborb, 2017-09-15:
Classic libsrc has grown a bit and we've got extra libraries where we probably don't need them and duplicated code with newlib. So lets do some rationalisation.

Reduce duplication:
* [ ] Switch over to using newlib string functions
* [ ] Use newlib stdlib functions
* [ ] Remove stdio_new (it's confusing being there)
* [ ] Pull in errno.h from newlib

Fold into core libraries:
* [ ] Fold balloc 
* [ ] Fold algorithm
* [ ] Fold adt 
* [ ] Fold debug 
* [ ] Fold rect

Speed build up:
* [ ] Sort out the graphics build mire (repeated cleans etc)

Despite pulling in newlib, classic will still support compiling with sdcc with a single library and mixing of compiled code (with the floating point, char caveats)

suborb, 2017-09-17:
Some notes (work continuing on Phase 2 in feature/classic_reorg_phase2 branch)

newlib adt depends on malloc (not u_malloc/u_free). Without switching classic allocation, we'll need to provide asm_realloc, asm_free to bridge to classic heap.

Additionally, the API is named differently. For compatibility we should introduce defc wrappers.

Balloc is automatically initialised with newlib. For compatibility we should provide the old method for classic as well as defc wrappers.

zx70, 2017-10-28:
I am on the msx lib for other reason,  hope not to collide

suborb, 2017-10-28:
I ended up getting distracted by sccz80 rather than sorting out the msx, psg directories, so you shouldn't be clashing!

suborb, 2017-11-09:
e8f154c5c191605a56c961ee06308bb598ca6314 did stdlib (wrong issue number)

closed 2020-08-14 by suborb