z88dk/z88dk #384: how to trap an existing function?

zx70, 2017-09-16:
What is the cleanest way for a programmer to trap an existing function (or portion)?
I'd like to suggest to disable the ansi scrollup functionality by simply trap the ANSI_scroll entry and replace it with a do_nothing one.

suborb, 2017-09-16:
Use the same trick as is used for fputc_cons

aralbrec, 2017-09-16:
The library functions are linked last so if the user provides a different version it will be used instead.

zx70, 2017-09-17:
Thanks, I'll first try to prevent the scroll event to happen too soon, in case of trobles I have your hints.
A further possibility is to add a flag in the ansi engine and permit programs to enable/disable the scroll at runtime, but it sounds a bit useless.

closed 2017-09-17 by zx70