z88dk/z88dk #386: create new z88dk/z88dk-libraries to hold third library code for use with z88dk

aralbrec, 2017-09-17:
Some discussion in #337 : https://github.com/z88dk/z88dk/issues/337#issuecomment-329352853

The specific examples given are the ChaN's FatFS, AVR time.h, and Free RTOS.  We may not want to integrate these directly for reasons of compile time (each time the z80 library is built) / licensing / awaiting customization for z88dk but libraries like these could be useful for users.

Maybe what we can do is create a new repo to hold code like this so that it is available from a convenient location with all the necessary instructions for use and building using z88dk.  We can use the new z88dk-lib utlity to install them for a target after building.

Thoughts?

feilipu, 2017-09-17:
With regard to this third party library process, I have created an example library, and pushed it up to a repository. For want of a better name, I called it [z88dk-time](https://github.com/feilipu/z88dk-time).

This is a fairly extensive time functions library supposedly close to fulfilling the ISO/IEC 9899 (C90) requirements.

I've been using it for some years in the [avr-libc](http://www.nongnu.org/avr-libc/user-manual/group__avr__time.html) for AVR ATmega machines, and it seems to work pretty well.

The only change for z80 is the use of the intrinsic division and modulus z88dk newlib library functions, rather than special functions provided by the avr-libc.

The compiled libraries are organised so that the `z88dk-lib` tool works.
The sccz80 compiler has trouble with some of the function calls.
I've not fixed this, so it is sdcc only at this stage.

I consider this library fairly general, as it is usable by any Z80 machine that can generate a 1 second [`system_tick`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/driver/system_time/z180/system_tick.asm) to increment the `time_t _system_time` variable. For Z180 machines, the PRT makes an ideal solution.

Happy for this to kick off a thought process. No requirement for [this repository](https://github.com/feilipu/z88dk-time) to remain in use, post resolution.

feilipu, 2017-09-18:
Added the C[haN FatFs code](https://github.com/feilipu/z88dk-libraries/tree/master/ff). Still clunky, because it requires multiple headers, but I might integrate them into `ff.h` to save trouble.

I had a go at breaking the public functions out into their own files.
The private functions I've left in one file. It is a can of spaghetti to untangle it all.
Messy work. I hope I've not broken anything.

feilipu, 2020-01-22:
The method to link 3rd party libraries works well enough, with some [example libraries here](https://github.com/feilipu/z88dk-libraries). One limitation is that only one header file can be included per library, but that's not too hard to overcome.

The [z88dk external programs repository](https://github.com/z88dk/z88dk-ext) already exists.

Does there still need to be a specific libraries repository?

feilipu, 2020-03-26:
No, probably no need for this centralised 3rd party repository.
If the need arises, re-open this issue.

closed 2020-03-26 by feilipu