z88dk/z88dk #389: z88dk-lib: third party library code should be able to have multiple headers and lib files

aralbrec, 2017-09-17:
At the moment, z88dk-lib only allows installation of 3rd party libraries composed of one header and one lib file.  A more general solution would allow such libraries to contain multiple headers and multiple library files.

This introduces the problem of managing the set of files.  When enumerating or removing the library, how do you know which files belong to it?  At the moment, that is done by library name which is used to find one header and one lib.

The easiest solution is probably to allocate a single directory for each of headers and lib files to the library by name and then their contents would belong to the library.  Such libraries would have to add the additional directory to includes in source and to link paths on the compile line.

zx70, 2017-11-20:
There's one thing I noticed recently,  not totally pertaining but I prefer not no open an issue just for it.
To override a library you can link yet another library AFTER the ones loaded already,  but if you wish to specify the same fn twice in a lst file, (eg to override an included lst) it must be BEFORE.

suborb, 2017-11-20:
I think user specified libraries are prepended to the link line so the logic in z80asm is the same.

zx70, 2017-11-20:
I see, I thought it was different,  tnx for telling it

feilipu, 2020-02-21:
I'd like to build a z88dk-FreeRTOS library, but there are quite a few (multiple)[ header files](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/master/include), needed for each of the feature groups.

I guess there's a "for all files in `/include/`" loop that could be written? But I'm not sure of how to do this neatly. Anyone able to help, please?