z88dk/z88dk #391: zcc: add a -map=format option to generate symbol files for consumption by specific tools?

aralbrec, 2017-09-17:
There is exactly one place in zcc.c that copies the map file to the destination directory.  It occurred to me we could perform a translation at that point to massage the map file into a format needed by a specific tool.  I'm thinking an additional option to zcc like "-map=emulicious", "-map=cspect", etc could be used to get zcc to generate an additional symbols file for consumption by these tools.

See also #387, #388, #341.

Tenebrous, 2017-09-17:
It would probably be better to output a .map format with all info and let the tool convert/process it, rather than catering for all the various tools out there.

aralbrec, 2017-09-17:
That won't happen though.  Things like emulicious and cspect are closed source and define their own proprietary symbols format, usually something connected to a specific tool (wla-dx (I think) and snasm in these cases).  It's unlikely they will accommodate different symbols formats for the same reason you mention - there are dozens of tools out there and they don't want to write a dozen converters.

z88dk does regularly accommodate dozens of external entities though - `appmake` generates output files for dozens of systems and emulators.  A symbols generator wouldn't be out of place although maybe it should be a standalone tool like `appmake` is rather than being integrated into `zcc`.  I'm already looking at writing two of these conversion tools, which would be nicer to put together in one place rather than maintaining many such individual tools somewhere.

Tenebrous, 2017-09-18:
That makes a lot more sense - basically a .map file converter.  Clearly it's better for consumers of the files to go with existing formats but I agree, that's never going to happen throughout.

aralbrec, 2017-10-22:
I think a map file converter is the way to go.  Something that can read in map files generated by various tools (z88dk initially) and can then generate map files in another format for another tool.  The primary aim is to be able to use z88dk map files in third part debuggers.

#387 and #388 are concerned with this for specific tools cspect and emulicious so I'll close them and relocate that here.

Emulicious has recently allowed z88dk map files as input (surprise!) but it doesn't seem to be fully functional at the moment.

aralbrec, 2017-11-05:
Chris Kirby has written an independent tool that converts z88dk map files to cspect format.

For future reference:

```
 https://github.com/Ckirby101/Z88CSPECT

Used like this, juts add to make file:
Z88CSPECT build\racing.map cspect.map
CSpect.exe -map=cspect.map -zxnext -mmc=.\ -s14 build\racing.sna

Compiled version
https://www.dropbox.com/s/w90ax12yuwwga9u/Z88CSPECT.rar?dl=0
```