z88dk/z88dk #394: Latest snapshot breaks ZX81 and ZX80 (at least with my game)

Fabrizio-Caruso, 2017-09-18:
With ZX81 I now get the following errors:
from file C:\Users\Brizio\Retro\ASCII-CHASE\merged\main.c, line 30:
#include "character.h"
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 24: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 25: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 27: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 29: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 35: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 37: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 46: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 47: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 53: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 59: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 65: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 71: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 77: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 83: symbol 'INVRS' not defined
14 errors occurred during assembly
Errors in source file c:/z88dk/\lib\zx81_crt0.asm:
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 24: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 25: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 27: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 29: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 35: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 37: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 46: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 47: symbol 'BOLD' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 53: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 59: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 65: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 71: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 77: symbol 'INVRS' not defined
Error at file 'stdio/ansi/zx81/f_ansi_attr.asm' line 83: symbol 'INVRS' not defined

With ZX80 I now get:
cpp: line 44, Warning: Redefining defined variable "gotoxy"
                #define gotoxy(x,y) zx_setcursorpos(y,x)
from file display_macros.h, line 64:
        #include "patch/z88dk_conio_implementation.h"
from file C:\Users\Brizio\Retro\ASCII-CHASE\merged\display_macros.c, line 23:
#include "display_macros.h"
cpp: line 44, Warning: Redefining defined variable "gotoxy"
                #define gotoxy(x,y) zx_setcursorpos(y,x)
from file display_macros.h, line 64:
        #include "patch/z88dk_conio_implementation.h"
from file character.h, line 26:
#include "display_macros.h"
from file enemy.h, line 28:
#include "character.h"
from file C:\Users\Brizio\Retro\ASCII-CHASE\merged\enemy.c, line 26:
#include "enemy.h"
cpp: line 44, Warning: Redefining defined variable "gotoxy"
                #define gotoxy(x,y) zx_setcursorpos(y,x)
from file display_macros.h, line 64:
        #include "patch/z88dk_conio_implementation.h"
from file character.h, line 26:
#include "display_macros.h"
from file level.h, line 29:
#include "character.h"
from file C:\Users\Brizio\Retro\ASCII-CHASE\merged\invincible_enemy.c, line 27:
#include "level.h"
cpp: line 44, Warning: Redefining defined variable "gotoxy"
                #define gotoxy(x,y) zx_setcursorpos(y,x)
from file display_macros.h, line 64:
        #include "patch/z88dk_conio_implementation.h"
from file character.h, line 26:
#include "display_macros.h"
from file C:\Users\Brizio\Retro\ASCII-CHASE\merged\level.c, line 25:
#include "character.h"
cpp: line 44, Warning: Redefining defined variable "gotoxy"
[... 84 lines cut ...]
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 965: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 967: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1007: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1009: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1030: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1032: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1047: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1049: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1093: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1095: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1117: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition
Error at file 'd:\WINDOW~1\zcc39EF11.asm' line 1119: symbol 'vtrendition' not defined
                   ^ ----       call    vtrendition

suborb, 2017-09-19:
This is related to the ansi library changes that have been happening. What version were you using beforehand?

Fabrizio-Caruso, 2017-09-19:
It worked with a very recent version including the fix to getk() in Aquarius by ZX70.
I could compile and correctly run 44 different versions of my game, half of which compiled with one of the two Z88DK compilers.

Fabrizio-Caruso, 2017-09-19:
I could check the solution of #368 5 days ago when I compiled LOTS of targets.
Morever I kept on improving the ZX81 for a couple of days.
So it was probably broken 2 or 3 days ago.

suborb, 2017-09-19:
Thanks, I think it was #377 that broke it.

The +zx81 is calling vtrendition() via CLEAR_SCREEN(). vtrendition calls f_ansi_attr, which needs BOLD, INVRS from the unimplemented (on the zx81) f_ansi_CHAR file.

And I suspect the same is true for MSX as well given that the ANSI engine isn't enabled there either.

It looks like @zx70 may be half way through sorting out conio for ports without an ansi driver.

zx70, 2017-10-01:
I think you are calling clrscr, which was not optimal but worked.
Supposing that clrscr is not required to reset color attributes, I  made it simpler, which should work for you.  Please remember that now conio is evolving and other similar issues can happen again (eg changing the origin text pos.)

closed 2017-10-07 by zx70