z88dk/z88dk #395: Latest snapshot breaks CPC: blank screen at startup

Fabrizio-Caruso, 2017-09-18:
The code works but nothing is shown on the screen.

I  have tested it with both color and no color version of my game for the CPC.

zx70, 2017-10-07:
It it resolved now?

zx70, 2017-10-13:
I'm keeping the issue open because the snakes.c game is not visible.. colour problems.?

zx70, 2017-10-21:
First small portion of cpcrslib is now merged in the current classic library.  Using the standard library edition is still possible,  the only conflict is in cpc.h, which had very few functions useful in conjunction with cpcrslib.  The high resolution graphics is now working at a full 640x200 monochrome resolution with an almost full coverage of the function list (it was 256x200 sometimes, 320x200 for other rom based stuff with limited coverage of the whole library).  Cp/m variant and greylib will require a serious reassessment,  but the new stuff worths the changes.

zx70, 2017-11-14:
Fonts and non-tile-based sprites are now in.   In the latest commit I'm also correcting a bug on the font color settings, which was also compromising the code stability.

zx70, 2017-11-23:
640x200 monochrome mode is now fully operational and respectably fast, I think *all* the graphics demos will be in working order by tomorrow morning.   
The CPCRSLIB is not totally included yet, the tile mode graphics have been improved in the years and there are two different, slightly incompatible versions around.  The code structure for 'tiles' could be kicked approaching it like we did with the ANSI VT options, I'm taking some time to think at it.
By the way, the current CPCRSLIB imported portion is now in a brand new, shiny shape.

zx70, 2017-12-01:
lots of open points already,  the original claim is solved and the library is now reasonably extended and cleaned..     I still need to check the status of the cp/m version of the libraries, so I keep the issue open

zx70, 2019-01-18:
I recently removed the 'startup=2' option  from the default cpc configuration. I couldn't succeed in running the graphics programs otherwise.    please advise if you find problems after this

closed 2019-02-06 by zx70