z88dk/z88dk #399: appmake +zx +zxn +sms +glue:  major internal changes for these targets

aralbrec, 2017-09-20:
A new bankswitched memory model has been incorporated that affects appmake for these targets.  If output files are not being properly generated please note them here.

feilipu, 2017-09-22:
Yes. I get an issue.
Do I need to specific a target or a bank model now?

```bash
PROCESSING CRT
z80asm --filler=0xFF -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/z180" --list   -s  --cpu=z180 -I"/home/phillip/Z80/z88dk/lib/config/../..//lib" -I"/home/phillip/Z80/NASCOM_BASIC_4.7/yaz180_NascomBasic56k"  "/tmp/tmpXXBcsyl6.asm"

z80asm --cpu=z180 -b -d  -o"int56k" -m -L.  -L"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/z180" --list  -iz180    "@/tmp/tmpXXhXJ9Pr.lst"
cat "/tmp/tmpXXBcsyl6.map" >> "int56k.map"
cat "/tmp/tmpXXBcsyl6.lis" >> "int56k.lis"
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/yaz180_NascomBasic56k$ appmake +glue -b int56k --ihex --pad --filler 0xFF --recsize 24 --clean
Segmentation fault (core dumped)
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/yaz180_NascomBasic56k$ 

```

aralbrec, 2017-09-22:
No it should continue to work as before.

```
appmake +glue -b test --ihex --pad --filler 0xFF --recsize 24 --clean
Creating bank space BANK
Creating test__.bin (org 0x8000 = 32768)
Creating test__BANK_001.bin (org 0xc000 = 49152)
Creating test__BANK_002.bin (org 0xc000 = 49152)
```

This is what I get out of a test compile using the same appmake invoke.   The "Creating bank space.." message comes out very early in +glue before much has happened.  (The default is to create a BANK bank space).

Is it possible to get a copy of what you're compiling with compile line?  Actually does a simple hello world work?

feilipu, 2017-09-22:
I'm compiling this... https://github.com/feilipu/NASCOM_BASIC_4.7/tree/master/sbc_NascomBasic56k_HexLoadr
I've made a few small mods which I just pushed up now.
The command lines are here.
```bash
zcc +z80 --no-crt -v -m --list -Ca--filler=0xFF @nascom56k.lst -o int56k
appmake +glue -b int56k --ihex --pad --filler 0xFF --recsize 24 --clean
```

`make install-clean; git pull; ./build.sh;` after the first segfault, so about 60 minutes ago.

I'll play with a hello world in the interim.

feilipu, 2017-09-22:
OK. This is not quite [hello_world](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_UnitTest/z180ide.asm), but it is simpler for me to get a quick answer with.

The `-create-app` options works. Separating the process, doesn't.

```bash
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/yaz180_UnitTest$ zcc +yaz180 -subtype=basic_dcio -SO3 -m --list z180ide.asm -o _ide -lm -create-app
.
.
.
z80asm --cpu=z180 -b -d  -o"_ide" -m -L.  -L"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC -D__SDCC_IY -I"/home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/yaz180" --list  -im -iyaz180    "/tmp/tmpXXSADEti.o" "/tmp/tmpXXF6lPah.o"
cat "/tmp/tmpXXSADEti.map" >> "_ide.map"
cat "/tmp/tmpXXSADEti.lis" >> "_ide.lis"
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/yaz180_UnitTest$ appmake +glue -b _ide --ihex--clean
Segmentation fault (core dumped)
phillip@SilverQuad:~/Z80/NASCOM_BASIC_4.7/yaz180_UnitTest$ 
```

aralbrec, 2017-09-22:
Ok I will give this one a try.  I'm getting compile errors on the other one ("z80intr.asm" should be a lst file I think "z80intr.lst" and then "nascom56k.lst" should have "@z80intr.lst").  I'm also missing "yaz180_config.h"

feilipu, 2017-09-22:
Yes, sorry. [Clone the head repo](https://github.com/feilipu/NASCOM_BASIC_4.7). That should give you the files you need.
I don't know if the sym-links work automatically. but the "yaz180_config.h" is in the same directory.

feilipu, 2017-09-22:
I've also just tried it in C with hello_world. Still getting a segfault, irrespective of the options or inputs.

1. My compile is broken somehow. I can trash the z88dk repo and clone.
2. Something else?

aralbrec, 2017-09-22:
I got the ide one to compile after adding the headers:

```
zcc +yaz180 -subtype=basic_dcio -SO3 -m --list z180ide.asm -o _ide -lm -create-app -vn

appmake +glue -b _ide --ihex --clean
Creating bank space BANK
Creating _ide__.bin (org 0x0000 = 0)
```

So that's confusing because it works for me but also because the org is wrong.

It looks like the "_ide.bin" file that -create-app is making is confusing appmake +glue.  If I remove create-app so ide.bin is not made:

```
appmake +glue -b ide --ihex --clean
Creating bank space BANK
Creating ide__.bin (org 0x8200 = 33280)
```

which is where it should be.

feilipu, 2017-09-22:
I will trash my clone, and see if it is my problem.
May take a few minutes to download again.

aralbrec, 2017-09-22:
Ok

feilipu, 2017-09-22:
Still have the segfault issue, with a new clone, and new `./build.sh`.
I'm now trying some of the test programmes, sudoku, umchess.

feilipu, 2017-09-22:
Should just I'm using an up-to-date Ubuntu 16.04, with standard gcc, etc.

aralbrec, 2017-09-22:
```
appmake +glue -b umchess --ihex --pad --filler 0xFF --recsize 24 --clean
Creating bank space BANK
Creating umchess__.bin (org 0x8000 = 32768)
```

It's working on windows :-/.  I will have to wait until tomorrow to try a linux compile.   The nightly build for windows also works - its binaries are cross compiled with gcc.

There's not much that can go wrong before printing the "Creating bank space.." message although it could be the output buffer is not flushed prior to the crash.

https://github.com/z88dk/z88dk/blob/master/src/appmake/glue.c#L83

feilipu, 2017-09-22:
I've just compiled a version from here https://github.com/z88dk/z88dk/tree/20ae76af95523651ec9a68a8fbd39237c08133b6
which works for me.

If you like I can step through and find the breakage later today.

aralbrec, 2017-09-22:
@feilipu   I think I found it - strtok was being applied to the constant string "BANK" which can't be done because strtok writes into the string.  Try compiling the appmake binary again.

feilipu, 2017-09-22:
Fixed!

aralbrec, 2017-09-23:
Noted in #222 : appmake +zx is demanding a map file for generating taps when it shouldn't.

closed 2017-10-26 by aralbrec