z88dk/z88dk #409: library make failing on mingw

zx70, 2017-09-25:
Recent changes make my (very personal and bugged) mingw installation fail:

--- Building Z88 Library ---

C:/MinGW/bin/make gfxdeps
make[1]: Entering directory 'c:/z88dk/libsrc'
make[1]: Leaving directory 'c:/z88dk/libsrc'
C:/MinGW/bin/make -C ctype
make[1]: Entering directory 'c:/z88dk/libsrc/ctype'
process_begin: CreateProcess(NULL, z88dk-lstmanip -i c:/z88dk/libsrc//_DEVELOPMENT//ctype/ctype_sccz80.lst -p c:/z88dk/libsrc//_DEVELOPMENT/ -t, ...) failed.
make[1]: Makefile:14: pipe: No error
mkdir: missing operand
Try `mkdir --help' for more information.
make[1]: *** [Makefile:72: dirs] Error 1
make[1]: Leaving directory 'c:/z88dk/libsrc/ctype'
make: *** [Makefile:198: z88_clib.lib] Error 2
bash-3.1$

suborb, 2017-09-25:
Dumb question: z88dk-lstmanip is compiled and in the path?

zx70, 2017-09-26:
You are right, sorry   😊

commit e2dab45d (2017-10-06): Fix #409: Fix #409: library make failing on mingw

The same solution as in the last commit is applied, but without hard-coding the MinGW root path in lstmanip.c.

Pass a new ```-r /``` command line option to lstmanip, this is translated to ```c:/MinGW/msys/1.0``` in MinGW.
commit 4dd4538f (2017-10-06): Fix #409: library make failing on mingw

The prefix-removing was failing on non-MinGW. Fixed.
commit 46aca48b (2017-10-06): Fix #409: library make failing on mingw

Problems found and fixed:

1) MinGW transforms the input command line arguments, inserting the full path name of the mingw home
before the given command line argument, i.e. the argument /home/paulo/git/z88dk-msys/libsrc/_DEVELOPMENT/
to lstmanip is received in argv[] as c:/MinGW/msys/1.0/home/paulo/git/z88dk-msys/libsrc/_DEVELOPMENT/

The result is that the $(subts...) in the makefiles that replace the library root NEWLIB_DIRECTORY by obj/z80/
fail to find the library root and the result paths are wrong.

Solved by a very hughly hack in lstmanip - filte [... cut]
commit ce0120e1 (2017-10-06): Merge pull request #432 from z88dk/fix_409

Fix #409: library make failing on mingw
closed 2017-09-26 by zx70