z88dk/z88dk #410: classic lib build:  adt-newlib is in the way

zx70, 2017-09-26:
C:/MinGW/bin/make -C stack all
make[2]: Entering directory 'c:/z88dk/libsrc/adt/stack'
make[2]: Leaving directory 'c:/z88dk/libsrc/adt/stack'
make[1]: Leaving directory 'c:/z88dk/libsrc/adt'
C:/MinGW/bin/make -C adt-newlib
make[1]: Entering directory 'c:/z88dk/libsrc/adt-newlib'
process_begin: CreateProcess(c:\MinGw\msys\1.0\bin\mkdir.exe, mkdir -p obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/c/sccz80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_array/z80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b_vector/c/sccz80/ obj/z80/adt/b   (...)    z80/, ...) failed.
make (e=206): The filename or extension is too long.
make[1]: *** [Makefile:72: dirs] Error 206
make[1]: Leaving directory 'c:/z88dk/libsrc/adt-newlib'
make: *** [Makefile:681: msx_clib.lib] Error 2
bash-3.1$
bash-3.1$ cd adt-newlib/
bash-3.1$ cat adt.lst
@adt-newlib/adt-newlib.lst

I'd also suggest to move the rs232 stuff outside the single target sections.

suborb, 2017-09-26:
Right, I'll need to uniq the directories. I think the command line is too long for windows!

suborb, 2017-09-26:
If you're experimenting, try changing the dirs: target to: `@mkdir -p $(sort $(dir $(OBJECTS)))`

zx70, 2017-09-26:
Takes long, but works !

suborb, 2017-09-26:
Yeah, it's a tricky call. Do we build object files separately and potentially get a speed up for the second build, or do we do as newlib and let z80asm reassemble them each time. 

The second related question is whether zcc or z80asm should be invoked to create the object files. 

But, at least it works!

zx70, 2017-09-26:
Sure, well done !

pauloscustodio, 2017-09-26:
> Yeah, it's a tricky call. Do we build object files separately and potentially get a speed up for the second
build, or do we do as newlib and let z80asm reassemble them each time.

If you pass -d to z80asm and leave the extension out in the list file, then z80asm only assembles the file if the object file does not exist or is out-of-date in relation to the source file.

aralbrec, 2017-09-27:
There may be target dependent code in the asm files so -d may not work as intended everywhere.

suborb, 2017-09-29:
Indeed. If z80asm is being used (I think I do in some places), the .asm is compiled and then the object is moved to the appropriate place, thus we get the correct object file. I suspect I could use -o instead of the mv, but I did have a problem with it at one point.

closed 2017-09-29 by suborb

zx70, 2017-09-30:
The slowness problem of the unix shims (mingw, cygwin...) on windows is mostly consequence of the way the process spawning is achieved.  Also the console support could probably have a better optimization, not mentioning the frequent security hardening extras.  Trying to keep good performances when the bottleneck is elsewhere is something I'd try to do only when the situation will be highly critical.