z88dk/z88dk #411: Failed to build rcmx000 support an examples

trlsmax, 2017-09-26:
Hi
I build the tool from souce and try to build rcmx000 support and examples, but failed.
```
[max@max-laptop rcmx000]$ make
gcc -o mk_coldboot_hfile mk_coldboot_hfile.c
ln -s ../../lib/rcmx000_boot.asm .
z80asm -DDEFINED_NOFLASH  rcmx000_boot.asm
z80asm -DDEFINED_NOFLASH  -a -m -orcmx000_boot.bin rcmx000_boot.o
Error: illegal option '-a'
1 errors occurred during assembly
make: *** [Makefile:26: rcmx000_boot.bin] Error 1
```

```
z80asm --cpu=r2k -b -d  -o"twinkle2000.bin" -L.   -L"/usr/local/share/z88dk//lib/clibs" -I"/usr/local/share/z88dk//lib" -ircmx000_clib -indos -i"rcmx000_crt0"     "/tmp/tmpXXfDAd1x.o" "/tmp/tmpXXiJiBxP.o"
Error at file '__printf_number64.asm' line 99: symbol 'asm1_ulltoa' not defined
Error at file '/home/max/Downloads/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
2 errors occurred during assembly
Errors in source file /usr/local/share/z88dk//lib/rcmx000_crt0.asm:
Error at file '__printf_number64.asm' line 99: symbol 'asm1_ulltoa' not defined
Error at file '/home/max/Downloads/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
                   ^ ----    cpir
make: *** [Makefile:21: twinkle2000.bin] Error 1
```

how to build those support and examples ?

suborb, 2017-09-26:
That's embarrassing, looks like we don't build the Rabbit examples nightly. I'll fix these problems up this evening so should be good for you tomorrow.

The Rabbit support tools were contributed many years ago, unfortunately none of us have a Rabbit board so we can't verify the bootloader etc.

pauloscustodio, 2017-09-26:
I will have a look. It seems to be related to recent changes in z80asm and the fact that these examples are not built by the top-level make.

suborb, 2017-09-27:
The Rabbit examples are now successfully building in both travis and the nightly.

I forgot to fix the boot tool so I'll do that tonight - I think it's looking at .map files which have recently changed.

trlsmax, 2017-09-28:
Thank you,
boot tool compile well, but examples still failed.

```
[max@max-laptop rcmx000]$ make clean
rm -f *.bin ; rm -f *.LOD ; rm -f tmp_* ; rm -f zcc_opt.def ; \
        rm -f *~ ; rm -f #* ; rm -f *#
[max@max-laptop rcmx000]$ make
zcc +rcmx000 -oexample.bin example.c

PROCESSING example.c
zcpp -iquote"." -DZ80 -DRCMX000 -D__RCMX000__  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/usr/local/share/z88dk//include"    "example.c" "/tmp/tmpXXSyRF9F.i2"
zpragma -sccz80 < "/tmp/tmpXXSyRF9F.i2" > "/tmp/tmpXXSyRF9F.i"
sccz80  -ext=opt -mr2k -standard-escape-chars   "/tmp/tmpXXSyRF9F.i"
copt "/usr/local/share/z88dk//lib/z80rules.9" < "/tmp/tmpXXSyRF9F.opt" > "/tmp/tmpXXSyRF9F.op1"
copt "/usr/local/share/z88dk//lib/z80rules.2" < "/tmp/tmpXXSyRF9F.op1" > "/tmp/tmpXXSyRF9F.opt"
copt "/usr/local/share/z88dk//lib/z80rules.1" < "/tmp/tmpXXSyRF9F.opt" > "/tmp/tmpXXSyRF9F.op1"
copt /usr/local/share/z88dk//lib/rabbit_rules.1 < "/tmp/tmpXXSyRF9F.op1" > "/tmp/tmpXXSyRF9F.asm"
cat "/tmp/tmpXXSyRF9F.asm" >> "/tmp/tmpXXSyRF9F.tmp"
cat "/tmp/tmpXXSyRF9F.tmp" >> "/tmp/tmpXXSyRF9F.asm"
z80asm --cpu=r2k  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" "/tmp/tmpXXSyRF9F.asm"

PROCESSING /usr/local/share/z88dk//lib/rcmx000_crt0.asm
cat "/usr/local/share/z88dk//lib/rcmx000_crt0.asm" >> "/tmp/tmpXXJWwIeL.asm"
cat "/tmp/tmpXXJWwIeL.asm" >> "/tmp/tmpXXJWwIeL.tmp"
cat "/tmp/tmpXXJWwIeL.tmp" >> "/tmp/tmpXXJWwIeL.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm --cpu=r2k  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" -I"/usr/local/share/z88dk//lib"  "/tmp/tmpXXJWwIeL.asm"
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 20: symbol '__start_prog' not defined
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 148: symbol '__prefix' not defined
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 167: symbol '__postfix' not defined
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 177: symbol '__patch_baudrate' not defined
4 errors occurred during assembly
Errors in source file /usr/local/share/z88dk//lib/rcmx000_crt0.asm:
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 20: symbol '__start_prog' not defined
                   ^ ----       ld (__start_prog+3),a
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 148: symbol '__prefix' not defined
                   ^ ----       ld hl,__prefix
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 167: symbol '__postfix' not defined
                   ^ ----       ld de,__postfix
Error at file '/usr/local/share/z88dk//lib/rcmx000_flash.asm' line 177: symbol '__patch_baudrate' not defined
                   ^ ----       ld a,(__patch_baudrate)       ; This is saved when we receive it from host
make: *** [Makefile:17: example.bin] Error 1
```

aralbrec, 2017-09-28:
One the latest commits changed case of those variables but missed a few places.  I've just committed a change that should fix this.

trlsmax, 2017-09-28:
It still failed

```
z80asm --cpu=r2k -b -d  -o"twinkle2000.bin" -L.   -L"/usr/local/share/z88dk//lib/clibs" -I"/usr/local/share/z88dk//lib" -ircmx000_clib -indos -i"rcmx000_crt0"     "/tmp/tmpXXfZreL8.o" "/tmp/tmpXXIdqQFu.o"
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
1 errors occurred during assembly
Errors in source file /usr/local/share/z88dk//lib/rcmx000_crt0.asm:
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
                   ^ ----    cpir
make: *** [Makefile:21: twinkle2000.bin] Error 1
```

aralbrec, 2017-09-28:
@paulo Is z80asm supposed to link the compatibility instructions automatically?

pauloscustodio, 2017-09-28:
Yes, the compatibility library is loaded automatically from one of the predefined locations, but if the library does not exist it does not issue an error message (maybe it should? - the point was that these functions might not be needed at all).

The standard locations are ${ZCCCFG}/.. (i.e. lib) and directory(z80asm.exe)/../lib, and these are populated by the top-level make.

I think that maybe the compatibility libraries were not built, they should be at z88dk/lib:
```
z80asm-r2k-.lib
z80asm-r2k-ixiy.lib
z80asm-r3k-.lib
z80asm-r3k-ixiy.lib
z80asm-z180-.lib
z80asm-z180-ixiy.lib
z80asm-z80-.lib
z80asm-z80-ixiy.lib
z80asm-z80_zxn-.lib
z80asm-z80_zxn-ixiy.lib
```

If you add the verbose option to z80asm, it will tell that the library was loaded:
```
> zcc +rcmx000 -Ca-v -otwinkle2000.bin twinkle2000.c
z80asm --cpu=r2k -v ...
Reading library 'c:/Users/paulo/git/z88dk/lib/config/../z80asm-r2k-.lib'
Predefined constant: __CPU_R2K__ = $0001
...
```

aralbrec, 2017-09-28:
@trlsmax I have successfully built that example - I think your install is possibly not up to date.  Try downloading a binary package or rebuilding everything again and see if the problem is solved?

The cpir error I think is coming from an old z80asm.  The assembler is supposed to replace z80 instructions not present on the rabbit processors with calls to compatibility functions.

The next error that can occur is this:

```
Error at file '/home/build/z88dk/build/z88dk/libsrc//_DEVELOPMENT//string/z80/as
m_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
1 errors occurred during assembly
```

which will happen if the r2k compatibility function libraries have not been built as mentioned by paulo.  They should be in z88dk/lib.

trlsmax, 2017-09-29:
@aralbrec I'm not sure what went wrong, but the error still happen. I use the following command to update and compile the example:

```
git pull
make clean
./build.sh
sudo make install
cd examples/rcmx000
make
```

And , I use manjaro KDE and gcc 7.2.0, is this would be the cause ?

aralbrec, 2017-09-29:
Would you try running `zcc +rcmx000 -Ca-v -otwinkle2000.bin twinkle2000.c` from z88dk/examples/rcmx000 and let us know what errors come up?

trlsmax, 2017-09-29:
```
[max@max-laptop rcmx000]$ pwd
/home/max/prj/z88dk/examples/rcmx000
[max@max-laptop rcmx000]$ zcc +rcmx000 -Ca-v -otwinkle2000.bin twinkle2000.c

PROCESSING twinkle2000.c
zcpp -iquote"." -DZ80 -DRCMX000 -D__RCMX000__  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/usr/local/share/z88dk//include"    "twinkle2000.c" "/tmp/tmpXX9beTl3.i2"
zpragma -sccz80 < "/tmp/tmpXX9beTl3.i2" > "/tmp/tmpXX9beTl3.i"
sccz80  -ext=opt -mr2k -standard-escape-chars   "/tmp/tmpXX9beTl3.i"
copt "/usr/local/share/z88dk//lib/z80rules.9" < "/tmp/tmpXX9beTl3.opt" > "/tmp/tmpXX9beTl3.op1"
copt "/usr/local/share/z88dk//lib/z80rules.2" < "/tmp/tmpXX9beTl3.op1" > "/tmp/tmpXX9beTl3.opt"
copt "/usr/local/share/z88dk//lib/z80rules.1" < "/tmp/tmpXX9beTl3.opt" > "/tmp/tmpXX9beTl3.op1"
copt /usr/local/share/z88dk//lib/rabbit_rules.1 < "/tmp/tmpXX9beTl3.op1" > "/tmp/tmpXX9beTl3.asm"
cat "/tmp/tmpXX9beTl3.asm" >> "/tmp/tmpXX9beTl3.tmp"
cat "/tmp/tmpXX9beTl3.tmp" >> "/tmp/tmpXX9beTl3.asm"
z80asm --cpu=r2k -v  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" "/tmp/tmpXX9beTl3.asm"
Predefined constant: __CPU_R2K__ = $0001
Assembling '/tmp/tmpXX9beTl3.asm' to '/tmp/tmpXX9beTl3.o'
Reading '/tmp/tmpXX9beTl3.asm'
Reading 'z80_crt0.hdr'
Module 'twinkle2000' size: 118 bytes

PROCESSING /usr/local/share/z88dk//lib/rcmx000_crt0.asm
cat "/usr/local/share/z88dk//lib/rcmx000_crt0.asm" >> "/tmp/tmpXXxY62D5.asm"
cat "/tmp/tmpXXxY62D5.asm" >> "/tmp/tmpXXxY62D5.tmp"
cat "/tmp/tmpXXxY62D5.tmp" >> "/tmp/tmpXXxY62D5.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm --cpu=r2k -v  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" -I"/usr/local/share/z88dk//lib"  "/tmp/tmpXXxY62D5.asm"
Predefined constant: __CPU_R2K__ = $0001
Assembling '/tmp/tmpXXxY62D5.asm' to '/tmp/tmpXXxY62D5.o'
Reading '/tmp/tmpXXxY62D5.asm'
Reading 'zcc_opt.def'
Reading 'rcmx000_boot.asm'
Reading 'rcmx000_flash.asm'
Reading 'rcmx000_flutil.asm'
Reading 'crt0_runtime_selection.asm'
Reading 'crt0_section.asm'
Module 'rcmx000_crt0' size: 1139 bytes

z80asm --cpu=r2k -b -d  -o"twinkle2000.bin" -L.   -L"/usr/local/share/z88dk//lib/clibs" -I"/usr/local/share/z88dk//lib" -ircmx000_clib -indos -i"rcmx000_crt0"     "/tmp/tmpXXxY62D5.o" "/tmp/tmpXX9beTl3.o"
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
1 errors occurred during assembly
Errors in source file /usr/local/share/z88dk//lib/rcmx000_crt0.asm:
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
                   ^ ----    cpir
[max@max-laptop rcmx000]$ 
```

pauloscustodio, 2017-09-29:
@trlsmax, can you please run ```which z80asm``` and check the directory of z80asm/../lib for a bunch of z80asm*.lib files?
It seems the z80asm did not find the libraries, either because there is another z80asm in the path or because the make install failed to copy them to the right place.

trlsmax, 2017-09-29:
Here it is :

```
[max@max-laptop bin]$ which z80asm
/usr/local/bin/z80asm
[max@max-laptop bin]$ ls ../lib
z80asm-r2k-ixiy.lib  z80asm-r3k-.lib       z80asm-z80-ixiy.lib      z80asm-z80_zxn-.lib
z80asm-r2k-.lib      z80asm-z180-ixiy.lib  z80asm-z80-.lib
z80asm-r3k-ixiy.lib  z80asm-z180-.lib      z80asm-z80_zxn-ixiy.lib
[max@max-laptop bin]$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/android-sdk/platform-tools:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/max/bin:/usr/local/diamond/3.9_x64/bin/lin64/
```

pauloscustodio, 2017-09-29:
Can you try running ```zcc +rcmx000 -Ca-v -Cl-v -otwinkle2000.bin twinkle2000.c``` from z88dk/examples/rcmx000 to see also the link pass messages?

trlsmax, 2017-09-29:
```
[max@max-laptop rcmx000]$ pwd
/home/max/prj/z88dk/examples/rcmx000
[max@max-laptop rcmx000]$ zcc +rcmx000 -Ca-v -Cl-v -otwinkle2000.bin twinkle2000.c

PROCESSING twinkle2000.c
zcpp -iquote"." -DZ80 -DRCMX000 -D__RCMX000__  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/usr/local/share/z88dk//include"    "twinkle2000.c" "/tmp/tmpXXa0Sdgk.i2"
zpragma -sccz80 < "/tmp/tmpXXa0Sdgk.i2" > "/tmp/tmpXXa0Sdgk.i"
sccz80  -ext=opt -mr2k -standard-escape-chars   "/tmp/tmpXXa0Sdgk.i"
copt "/usr/local/share/z88dk//lib/z80rules.9" < "/tmp/tmpXXa0Sdgk.opt" > "/tmp/tmpXXa0Sdgk.op1"
copt "/usr/local/share/z88dk//lib/z80rules.2" < "/tmp/tmpXXa0Sdgk.op1" > "/tmp/tmpXXa0Sdgk.opt"
copt "/usr/local/share/z88dk//lib/z80rules.1" < "/tmp/tmpXXa0Sdgk.opt" > "/tmp/tmpXXa0Sdgk.op1"
copt /usr/local/share/z88dk//lib/rabbit_rules.1 < "/tmp/tmpXXa0Sdgk.op1" > "/tmp/tmpXXa0Sdgk.asm"
cat "/tmp/tmpXXa0Sdgk.asm" >> "/tmp/tmpXXa0Sdgk.tmp"
cat "/tmp/tmpXXa0Sdgk.tmp" >> "/tmp/tmpXXa0Sdgk.asm"
z80asm --cpu=r2k -v  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" "/tmp/tmpXXa0Sdgk.asm"
Predefined constant: __CPU_R2K__ = $0001
Assembling '/tmp/tmpXXa0Sdgk.asm' to '/tmp/tmpXXa0Sdgk.o'
Reading '/tmp/tmpXXa0Sdgk.asm'
Reading 'z80_crt0.hdr'
Module 'twinkle2000' size: 118 bytes

PROCESSING /usr/local/share/z88dk//lib/rcmx000_crt0.asm
cat "/usr/local/share/z88dk//lib/rcmx000_crt0.asm" >> "/tmp/tmpXX7ZIRpZ.asm"
cat "/tmp/tmpXX7ZIRpZ.asm" >> "/tmp/tmpXX7ZIRpZ.tmp"
cat "/tmp/tmpXX7ZIRpZ.tmp" >> "/tmp/tmpXX7ZIRpZ.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm --cpu=r2k -v  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" -I"/usr/local/share/z88dk//lib"  "/tmp/tmpXX7ZIRpZ.asm"
Predefined constant: __CPU_R2K__ = $0001
Assembling '/tmp/tmpXX7ZIRpZ.asm' to '/tmp/tmpXX7ZIRpZ.o'
Reading '/tmp/tmpXX7ZIRpZ.asm'
Reading 'zcc_opt.def'
Reading 'rcmx000_boot.asm'
Reading 'rcmx000_flash.asm'
Reading 'rcmx000_flutil.asm'
Reading 'crt0_runtime_selection.asm'
Reading 'crt0_section.asm'
Module 'rcmx000_crt0' size: 1139 bytes

z80asm --cpu=r2k -b -d  -o"twinkle2000.bin" -L.   -v -L"/usr/local/share/z88dk//lib/clibs" -I"/usr/local/share/z88dk//lib" -ircmx000_clib -indos -i"rcmx000_crt0"     "/tmp/tmpXX7ZIRpZ.o" "/tmp/tmpXXa0Sdgk.o"
Reading library '/usr/local/share/z88dk//lib/clibs/rcmx000_clib.lib'
Reading library '/usr/local/share/z88dk//lib/clibs/ndos.lib'
Reading library '/usr/local/share/z88dk//lib/clibs/rcmx000_crt0.lib'
Predefined constant: __CPU_R2K__ = $0001
Linking library module 'closeall'
Linking library module 'fabandon'
Linking library module 'fclose'
Linking library module 'printf'
Linking library module 'fputc_callee'
Linking library module 'fchkstd'
Linking library module 'puts_cons_native'
Linking library module 'asm_printf'
Linking library module '__printf_doprint'
Linking library module '__printf_format_table'
Linking library module '__printf_get_flags_impl'
Linking library module '__printf_handle_X'
Linking library module '__printf_handle_c'
Linking library module '__printf_handle_d'
Linking library module '__printf_handle_n'
Linking library module '__printf_handle_o'
Linking library module '__printf_handle_s'
Linking library module '__printf_handle_u'
Linking library module '__printf_handle_x'
Linking library module '__printf_handle_B'
Linking library module '__printf_number'
Linking library module '__printf_print_aligned'
[... 101 lines cut ...]
Section 'smc_user' size: 0 bytes ($0B91 to $0B90)
Section 'data_clib' size: 53 bytes ($0B91 to $0BC5)
Section 'data_crt' size: 0 bytes ($0BC6 to $0BC5)
Section 'data_compiler' size: 0 bytes ($0BC6 to $0BC5)
Section 'data_user' size: 0 bytes ($0BC6 to $0BC5)
Section 'data_alloc_balloc' size: 0 bytes ($0BC6 to $0BC5)
Section 'DATA_END' size: 0 bytes ($0BC6 to $0BC5)
Section 'BSS' size: 0 bytes ($0BC6 to $0BC5)
Section 'bss_fp' size: 0 bytes ($0BC6 to $0BC5)
Section 'bss_error' size: 2 bytes ($0BC6 to $0BC7)
Section 'bss_crt' size: 65 bytes ($0BC8 to $0C08)
Section 'bss_fardata' size: 0 bytes ($0C09 to $0C08)
Section 'bss_compiler' size: 0 bytes ($0C09 to $0C08)
Section 'bss_clib' size: 0 bytes ($0C09 to $0C08)
Section 'bss_string' size: 0 bytes ($0C09 to $0C08)
Section 'bss_alloc_balloc' size: 0 bytes ($0C09 to $0C08)
Section 'bss_user' size: 0 bytes ($0C09 to $0C08)
Section 'BSS_END' size: 0 bytes ($0C09 to $0C08)
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
1 errors occurred during assembly
Errors in source file /usr/local/share/z88dk//lib/rcmx000_crt0.asm:
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
                   ^ ----    cpir
[max@max-laptop rcmx000]$ 
```

pauloscustodio, 2017-09-29:
I've pushed a new z80asm with some debug messages to try to understand why it is not loading the support library. Please pull, rebuild and post again the result of ```zcc +rcmx000 -Ca-v -Cl-v -otwinkle2000.bin twinkle2000.c```.

suborb, 2017-09-29:
@pauloscustodio I'm wondering if this is a problem with make install - Max's install is in /usr/local/

pauloscustodio, 2017-09-29:
Yes. I've added debug messages to find out where exactly z80asm is searching. 

For the future,  what is the most appropriate place to z80asm*.lib? I thought it would be bin/../lib, but apparently I was wrong.

suborb, 2017-09-29:
For z88dk searching in ${ZCCCFG}/.. works. For more general usage I'd say it should search any -L directory for them as well.

trlsmax, 2017-09-29:
Just pull the new commit and compile it again :

```
[max@max-laptop rcmx000]$ pwd
/home/max/prj/z88dk/examples/rcmx000
[max@max-laptop rcmx000]$ which z80asm
/usr/local/bin/z80asm
[max@max-laptop rcmx000]$ zcc +rcmx000 -Ca-v -Cl-v -otwinkle2000.bin twinkle2000.c

PROCESSING twinkle2000.c
zcpp -iquote"." -DZ80 -DRCMX000 -D__RCMX000__  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/usr/local/share/z88dk//include"    "twinkle2000.c" "/tmp/tmpXXb0CAXz.i2"
zpragma -sccz80 < "/tmp/tmpXXb0CAXz.i2" > "/tmp/tmpXXb0CAXz.i"
sccz80  -ext=opt -mr2k -standard-escape-chars   "/tmp/tmpXXb0CAXz.i"
copt "/usr/local/share/z88dk//lib/z80rules.9" < "/tmp/tmpXXb0CAXz.opt" > "/tmp/tmpXXb0CAXz.op1"
copt "/usr/local/share/z88dk//lib/z80rules.2" < "/tmp/tmpXXb0CAXz.op1" > "/tmp/tmpXXb0CAXz.opt"
copt "/usr/local/share/z88dk//lib/z80rules.1" < "/tmp/tmpXXb0CAXz.opt" > "/tmp/tmpXXb0CAXz.op1"
copt /usr/local/share/z88dk//lib/rabbit_rules.1 < "/tmp/tmpXXb0CAXz.op1" > "/tmp/tmpXXb0CAXz.asm"
cat "/tmp/tmpXXb0CAXz.asm" >> "/tmp/tmpXXb0CAXz.tmp"
cat "/tmp/tmpXXb0CAXz.tmp" >> "/tmp/tmpXXb0CAXz.asm"
z80asm --cpu=r2k -v  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" "/tmp/tmpXXb0CAXz.asm"
Searching support library 'z80asm-r2k-.lib'
Searching support library './z80asm-r2k-.lib'
Searching support library './../lib/z80asm-r2k-.lib'
Searching support library '/../z80asm-r2k-.lib'
Support library 'z80asm-r2k-.lib' not found
Found library '(null)', trying to load it
Predefined constant: __CPU_R2K__ = $0001
Assembling '/tmp/tmpXXb0CAXz.asm' to '/tmp/tmpXXb0CAXz.o'
Reading '/tmp/tmpXXb0CAXz.asm'
Reading 'z80_crt0.hdr'
Module 'twinkle2000' size: 118 bytes

PROCESSING /usr/local/share/z88dk//lib/rcmx000_crt0.asm
cat "/usr/local/share/z88dk//lib/rcmx000_crt0.asm" >> "/tmp/tmpXX25hkGU.asm"
cat "/tmp/tmpXX25hkGU.asm" >> "/tmp/tmpXX25hkGU.tmp"
cat "/tmp/tmpXX25hkGU.tmp" >> "/tmp/tmpXX25hkGU.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm --cpu=r2k -v  -s  --cpu=r2k -I"/usr/local/share/z88dk//lib" -I"/usr/local/share/z88dk//lib"  "/tmp/tmpXX25hkGU.asm"
Searching support library 'z80asm-r2k-.lib'
Searching support library './z80asm-r2k-.lib'
Searching support library './../lib/z80asm-r2k-.lib'
Searching support library '/../z80asm-r2k-.lib'
Support library 'z80asm-r2k-.lib' not found
Found library '(null)', trying to load it
Predefined constant: __CPU_R2K__ = $0001
Assembling '/tmp/tmpXX25hkGU.asm' to '/tmp/tmpXX25hkGU.o'
Reading '/tmp/tmpXX25hkGU.asm'
Reading 'zcc_opt.def'
Reading 'rcmx000_boot.asm'
Reading 'rcmx000_flash.asm'
Reading 'rcmx000_flutil.asm'
Reading 'crt0_runtime_selection.asm'
Reading 'crt0_section.asm'
Module 'rcmx000_crt0' size: 1139 bytes

z80asm --cpu=r2k -b -d  -o"twinkle2000.bin" -L.   -v -L"/usr/local/share/z88dk//lib/clibs" -I"/usr/local/share/z88dk//lib" -ircmx000_clib -indos -i"rcmx000_crt0"     "/tmp/tmpXX25hkGU.o" "/tmp/tmpXXb0CAXz.o"
Reading library '/usr/local/share/z88dk//lib/clibs/rcmx000_clib.lib'
Reading library '/usr/local/share/z88dk//lib/clibs/ndos.lib'
Reading library '/usr/local/share/z88dk//lib/clibs/rcmx000_crt0.lib'
Searching support library 'z80asm-r2k-.lib'
Searching support library './z80asm-r2k-.lib'
Searching support library './../lib/z80asm-r2k-.lib'
Searching support library '/../z80asm-r2k-.lib'
Support library 'z80asm-r2k-.lib' not found
Found library '(null)', trying to load it
Predefined constant: __CPU_R2K__ = $0001
[... 123 lines cut ...]
Section 'smc_user' size: 0 bytes ($0B91 to $0B90)
Section 'data_clib' size: 53 bytes ($0B91 to $0BC5)
Section 'data_crt' size: 0 bytes ($0BC6 to $0BC5)
Section 'data_compiler' size: 0 bytes ($0BC6 to $0BC5)
Section 'data_user' size: 0 bytes ($0BC6 to $0BC5)
Section 'data_alloc_balloc' size: 0 bytes ($0BC6 to $0BC5)
Section 'DATA_END' size: 0 bytes ($0BC6 to $0BC5)
Section 'BSS' size: 0 bytes ($0BC6 to $0BC5)
Section 'bss_fp' size: 0 bytes ($0BC6 to $0BC5)
Section 'bss_error' size: 2 bytes ($0BC6 to $0BC7)
Section 'bss_crt' size: 65 bytes ($0BC8 to $0C08)
Section 'bss_fardata' size: 0 bytes ($0C09 to $0C08)
Section 'bss_compiler' size: 0 bytes ($0C09 to $0C08)
Section 'bss_clib' size: 0 bytes ($0C09 to $0C08)
Section 'bss_string' size: 0 bytes ($0C09 to $0C08)
Section 'bss_alloc_balloc' size: 0 bytes ($0C09 to $0C08)
Section 'bss_user' size: 0 bytes ($0C09 to $0C08)
Section 'BSS_END' size: 0 bytes ($0C09 to $0C08)
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
1 errors occurred during assembly
Errors in source file /usr/local/share/z88dk//lib/rcmx000_crt0.asm:
Error at file '/home/max/prj/z88dk/libsrc//_DEVELOPMENT//string/z80/asm_strlen.asm' line 33: symbol '__z80asm__cpir' not defined
                   ^ ----    cpir
[max@max-laptop rcmx000]$ 
```

suborb, 2017-09-29:
For a quick fix set the environment variable ZCCCFG to /usr/local/share/z88dk/lib/config

@pauloscustodio as an addendum to my comment above, we should probably search PREFIX/lib for the library as well.

@trlsmax as you've probably guessed, we've not done a system install since these libraries were added. Since we work out of a local source tree the paths are anchored using the ZCCCFG variable. In a system install, ZCCCFG internally defaults to the value mentioned above, however z80asm isn't doing that for the support libraries.

pauloscustodio, 2017-09-29:
Problem found. Thanks for reporting.

There are two problems:
1) argv[0] in Linux does not contain the full executable path as in Windows, so dirname(argv[0])/../lib does not point to the bin/../lib directory
2) The z80asm support libraries should be located in /lib/clibs/

For a workaround until I fix this in z80asm, please set ZCCCFG to /usr/local/share/z88dk//lib/config

pauloscustodio, 2017-09-29:
@suborb, thanks. I'll search also the -L path and PREFIX/lib.

Should I store the libraries in PREFIX/lib/z80asm instead of PREFIX/lib or PREFIX/clibs?

aralbrec, 2017-09-29:
@subirb It might be a good idea to disable the longlong stuff by default as that pulls in a wad of code that is rarely used.

```
Linking library module '__printf_handle_ll'
Linking library module '__printf_format_table64'
Linking library module '__printf_handle_llB'
Linking library module '__printf_number64'
Linking library module '__printf_handle_llX'
Linking library module '__printf_handle_lld'
Linking library module '__printf_handle_llo'
Linking library module '__printf_handle_llu'
Linking library module '__printf_handle_llx'
```

suborb, 2017-09-29:
This is a bit of edge case where no formats have been detected so everything is pulled in. Here the printf format table isn't defined by the crt, but instead comes from the library. The same thing happens with sdcc without a pragma printf.

Probably easiest to resolve it by adding a `#pragma printf`.

closed 2017-09-29 by pauloscustodio

trlsmax, 2017-09-30:
Just confirm it works, Thank you.

commit c583883b (2017-09-29): Fix #411: z80asm now searches PREFIX and -L library path for its support libraries

Also: more verbose text telling where libraries are being searched.